Marus/cortex-debug #823: SEGGER RTT Graphing Not Working

DepthDeluxe, 2023-02-09:
I'm trying to get up and running with graphing support for SEGGER RTT via cortex-debug and while I can configure and launch the graph, it doesn't plot anything.  I'm using this guide https://github.com/Marus/cortex-debug/wiki/SWO-Output#output-graphing-graphing.

For my demo, I'm writing an incrementing 32-bit counter value out of RTT channel 4 every 100ms.

```C
static void count_fn(void *a, void *b, void *c)
{
  ARG_UNUSED(a);
  ARG_UNUSED(b);
  ARG_UNUSED(c);
  k_msleep(1000);

  uint32_t value;
  while (true) {
    k_msleep(100);
    SEGGER_RTT_Write(PLOT_RTT_CHANNEL, &value, sizeof(value));
    ++value;
  }
}
```

I manually confirmed via the debugger that the thread is running and I checked that the correct data is being written out RTT channel 4.  Attached is a snippet of the `binary` decoder configured for this port.

```
[2023-02-09T20:56:15.718Z, +010161ms] ....  0x00000001 -     16777216 -     16777216
[2023-02-09T20:56:15.820Z, +010263ms] ....  0x00000002 -     33554432 -     33554432
[2023-02-09T20:56:15.922Z, +010365ms] ....  0x00000003 -     50331648 -     50331648
[2023-02-09T20:56:16.021Z, +010464ms] ....  0x00000004 -     67108864 -     67108864
...
[2023-02-09T20:57:59.501Z, +113942ms] ....  0x0000000a -    167772160 -    167772160
[2023-02-09T20:57:59.602Z, +114043ms] ....  0x0000000b -    184549376 -    184549376
[2023-02-09T20:57:59.707Z, +114148ms] ....  0x0000000c -    201326592 -    201326592
```

When I select the "graph" decoder, I don't see any data coming out of the graph.  Here is the relevant part of my launch.json

**Graph**
[image]

**launch.json**
```
        {
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "MIMXRT685S_M33",
            "request": "attach",
            "rtos": "/Applications/SEGGER/JLink/GDBServer/RTOSPlugin_Zephyr",
            "executable": "build/zephyr/zephyr.elf",
            "gdbPath": "${workspaceRoot}/.dependencies/gcc-arm-zephyr-eabi-10.3.0-0.13.2/bin/arm-zephyr-eabi-gdb",
            "serverArgs": [
                "-select",
                // JLINK_MINI
                "USB=821006741"
            ],
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "label": "",
                        "port": 4,
                        "type": "binary",
                        "encoding": "unsigned",
                        "graphId": "1",
                        "scale": 1,
                    }
                ],
            },
            "graphConfig": [
                {
                    "label": "Audio Signal (CM33)",
                    "timespan": 0.5,
                    "type": "realtime",
                    "annotate": true,
                    "minimum": -1000,
                    "maximum": 1000,
                    "plots": [
                        {
                            "graphId": "1",
                            "label": "Data",
                            "color": "#53753c"
                        },
                    ]
                }
            ]
        },
```

I am running
Cortex-Debug 1.6.10
VSCode 1.75.1
OSX 11.7.2

haneefdm, 2023-02-09:
Shouldn't your rttConfig.decoders "type" be a "graph"? You have it as "binary" so the output is probably going to the TERMINAL window instead.

Please see the SWO example carefully.

DepthDeluxe, 2023-02-09:
@haneefdm, apologies I copied the config from when I modified the config to print the binary values out to generate the issue.  When I change the decoder type to "graph", I also get an empty graph.

DepthDeluxe, 2023-02-09:
I should also add that I'm using a SEGGER JLink programmer and have dev tools 7.84a installed on my local machine.

haneefdm, 2023-02-09:
You are using channel 4. That is a hit and miss. SEGGER gives us a few milliseconds to change channel from default 0 to 4.

Did you try multiple starts? If you are seeing results from the binary output, then you should also be seeing it from the graph. Hint you can have both (multiple) graph and binary decoders for the same channel. I would add two decoders and see if the binary one is working while the graph one is not.

Also, I notice that your values are way outside the plotting range for min/max

haneefdm, 2023-03-01:
Pinging again to see if any progress has been made. Nothing we see actionable on our end.

closed 2023-03-02 by haneefdm

joncmse, 2024-06-17:
I'd like to resurrect this thread as I am having the same issue. I was very excited to see this feature existed in Cortex Debug, as it could save me from having to write it in Python!

My situation is similar, but I am writing 32-bit unsigned integers at 8khz like so:
```
    uint32_t plotValue = 123;
    SEGGER_RTT_Write(0, &plotValue, 4);
```

My launch.json looks like this:

```
    {
        "name": "JLink RTT Plot",
        "cwd": "${workspaceFolder}/builds/dbg-stm/bin",
        "executable":"${workspaceFolder}/builds/dbg-stm/bin/aux_inverter_app/aux_inverter_app.elf",
        "device": "STM32F407VG",
        "request": "launch",
        "type": "cortex-debug",
        "showDevDebugOutput": "none", // Try "raw" for more info
        "servertype": "jlink",
        "rttConfig": 
        {
            "enabled": true, 
            "address": "auto",
            "decoders": [
            {
                "type": "graph",
                "port": 0,      
                "encoding": "unsigned",
                "graphId": "1", 
                "scale": 1
            }
            ]
        },
        "graphConfig": [
        {
            "label": "Graph 1",
            "timespan": 5,
            "type": "realtime",
            "annotate": false,
            "maximum": 255,
            "minimum": 0,
            "plots": [
            {
                "graphId": "1", 
                "label": "data 1",
                "color": "#53753c"
            }
            ]
        }
        ]
    }
```
Plot looks like this:
[image]

Cortex Debug v1.12.1
VSCode v1.90.0
Ubuntu v22.04.4
JLink Pro v5.1A