Marus/cortex-debug #824: Debugging in BE8 causes wrong instruction to be executed

nonarkitten, 2023-02-10:
Okay, this one is a head scratcher and took a while to figure out precisely what's going on.

My project depends heavily on reading/writing through the GPMC and DRAM in big-endian mode. For performance reasons, I'm not about to wrap each and every read/write operation with a byte swap, especially when that's expressly what BE8 mode is for. Anyway, the code compiles fine and when I load and run it, all is well. I can also stop it fine, no problem so far.

The chaos happens when I step or use breakpoints. The debugger appears to MUTATE the instruction into what it would be if the bytes were reversed -- often into something nonsensical. These sometimes actually work simply because the broken opcode will generate an exception, but often it produces a valid opcode and the code will run right over it into la-la land. Other times they get mutated into branches and jump into random parts of code.

Given that I'm dealing with three layers (JLINK, GDB Server and cortex-debug) I cannot say which part is causing the problem. If I had to guess, it's GDB Server.

A simple solution might be to disable the use of software breakpoints, but I cannot find any way to do that. I've never had to debug GDB itself, and honestly don't even know where to start, but this needs to get fixed and from Google, I can see that BE8 has been a bit of an unanswered problem so far.

haneefdm, 2023-02-10:
You forgot the most important layer: GDB :-) Normally, it is GDB who will do that. It knows if the memory is writable and use SW breakpoints to modify you code at that point by putting a trap or illegal instruction causing an exception. but if/when that exception is hit,, it will restore the instruction, single step, etc. before continuing. Complicated song and dance. If gdb is doing this wrong, then it is a bug in GDB. We have used GDB that way all of my life and it never interferes with actual running of the program.

We are just a front-end. These low-level things happen in either GDB or the GDB-server.

Your executable and the gdb-server inform GDB to let it know which memory regions are writable/readable/etc. You can do an `info mem` and/or `maintenance info sections` in the Debug Console and get a list

You can do some amount of management in gdb to change the memory attributes

https://sourceware.org/gdb/onlinedocs/gdb/Memory-Region-Attributes.html

But, if the gdb-server is doing it, I have no clue and I would contact them.

Hope this helps.

closed 2023-02-10 by haneefdm

haneefdm, 2023-02-10:
Oh, you said "big-endian" mode. I would not be very surprised if GDB was misconfigured for your device. Or it truly is a gdb-bug if gdb is the one handling SW breakpoints. Not many big-endian devices out there.

I bet we have some bugs too in that regard -- luckily, we rarely work at a byte level -- other than SVD peripheral registers. For us, most values are simply strings as given to us by GDB. We do not try to interpret them.

nonarkitten, 2023-02-10:
Okay, at lest I can eliminate this as a possibility then, thanks. Yes, it's most likely an issue with GDB itself. There's sadly no "proper" way of configuring it and it would seem developers in the GCC team have largely just given up. And for the record, every ARMv7-A CPU is bi-endian.

haneefdm, 2023-02-10:
> And for the record, every ARMv7-A CPU is bi-endian.

Do you mean `big-endian`? Did not know that about ARMv7-A. I know that most ARM processors can choose their endian-ness and the default is generally little so is the preference for vendors building the chips. It is very rare for me to run across a big-endian ARM implementation/use, but I have seen it once and now twice. But your experience seems to be the opposite of mine.

https://www.quora.com/Is-ARM-big-endian-or-little-endian

Btw, you can tell gdb to use a particular endianness in case gdb was built with the wrong configuration for your processor.

https://sourceware.org/gdb/download/onlinedocs/gdb/Byte-Order.html

haneefdm, 2023-02-10:
If gdb chose the wrong endian-ness, you should force the gdb endian-ness using either the preLaunchCommands or a .gdbinit file

nonarkitten, 2023-02-10:
Bi-endian (no 'g') is "both-endians". You can use the SETEND to switch. So drop that as the first op after your reset vector and you're basically in big-endian mode. ARM in the infinite wisdom didn't REALLY make it proper big-endian since opcodes remain in little-endian format. So GDB sort of loses it's mind.

JLinkGDBServerCLExe is run with -endian big
I use 'set endian big' just in case 'auto' is confused
Code is compiled with -mbig-endian and -mbe8
Code is linked with -Wl,--be8 -Wl,--format=elf32-bigarm
The startup.s has SETEND BE

And more importantly, it runs fine if I just click "go". It doesn't melt down until I start to step or set breakpoints as I suspect GDB is mutating the opcodes from little-endian into big-endian.

I guess I'm downloading GCC source...

haneefdm, 2023-02-11:
> You can use the SETEND to switch
Yes, I know

> I guess I'm downloading GCC source...

You are braver than I am, orders of magnitude :-)  Perhaps there is an easier way. Do you have any other gdb based debugger that works right?

Good luck. Let us know