Marus/cortex-debug #826: embOS detected but unable to collect RTOS information

NOhs, 2023-02-14:
Hello there,

Thanks again for this great plugin! I couldn't do my work without it. Currently, I have embOS running on a little evaluation board and I can use embOS-View to see the different tasks and their stacks etc. If I try to use the XRTOS window of cortex-debug, I get the message (while pausing at a break point):

> **RTOS Views: Session Name: "armclang", embOS detected.**
>
> Unable to collect full RTOS information. Could not read any task from "OS_Global.pTask". Perhaps program is busy or did not stop long enough 

If I use the default variable watch feature of cortex-debug, I CAN access `OS_Global.pTask`. So I am a bit confused as to what is going on. Do I need a special configuration for cortex-debug to handle embOS? I currently only have:

```json
"rtos": "embOS",
```

in my `launch.json`.

haneefdm, 2023-02-14:
Does the Refresh button help at all? Curiously, I do not see the refresh button in the screenshots. There is also a command you can use from the Command Palette

This extension does not use that setting (or the entire launch.json). We scan for all RTOSes automatically. Someone with embOS experience can comment on what you are seeing.

Perhaps you can enable the gdb traces by setting `"showDevDebugOutput": "raw"` and post the output of Debug Console here. You can tell what is happening with `OS_Global.pTask`

haneefdm, 2023-02-14:
Unfortunately, this issue you filed is in the wrong repo. You should file this at 

https://github.com/mcu-debug/rtos-views

and, there is no way to transfer the issue.

NOhs, 2023-02-14:
Ok, I will create a new issue in the other repo. This one can be closed.

PhilippHaefele, 2023-02-14:
@NOhs and as you do create the new issue, please state some values like versions (embOS, RTOS Views, ...) and the used hardware (mainly the used micro-controller). I will need them to tackle this issue. Thanks

PhilippHaefele, 2023-02-26:
@haneefdm as we've fixed the issue in the meanwhile, we can close this one. Thanks

closed 2023-02-26 by haneefdm