Marus/cortex-debug #827: Run a RAM app from JLink

basprins, 2023-02-15:
Dear Haneef,
First of all my apologies to open an issue on your library. But I don't know where else to go, so I hope you are willing to help me.

I am using cortex debug for quite a while, serves me really well. We are currently at the point where our product is going to production. We need to create "tools" for the production organization to build our product. For that, we need to burn a fuse on the chip, before any software will be able to launch.

Since it seems to be impossible to burn that fuse through JLink directly, the only other option we have is to run a RAM app, which burns the fuse, and from that moment forward we can use our regular apps where they can execute the tests with.

But, I can't seem to figure out how I should deploy a RAM app from JLink directly. But... I CAN run a RAM app with Cortex-Debug. So I know it is possible.

Thus, the question. This is what I used for quite a while to run some RAM app using Cortex-Debug

```
    {
      "name": "bt-fuse-sel-1024",
      "type": "cortex-debug",
      "request": "launch",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/bt-fuse-sel-1024/build/bt-fuse-sel-1024.axf",
      "serverpath": "/usr/bin/JLinkGDBServerCLExe",
      "windows": {
        "serverpath": "C:\\Program Files (x86)\\SEGGER\\JLink\\JLinkGDBServerCL.exe",
      },
      "osx": {
        "serverpath": "/Applications/SEGGER/JLink_V764e/JLinkGDBServerCLExe",
      },
      "servertype": "jlink",
      "device": "MIMXRT1024xxx5A",
      "overrideLaunchCommands": [
        "-interpreter-exec console \"monitor halt\"",
        "-interpreter-exec console \"monitor reset\"",
        "-target-download",
        "-enable-pretty-printing",
    ],
      "interface": "swd",
      "preLaunchTask": "bt-fuse-sel-1024 - build",
      "runToEntryPoint": "main",
      "showDevDebugOutput": "raw"
    },  
```

I didn't invent this myself, I've stolen the trick from this forum too.

This work, this launches the RAM app which burns the fuse, works perfectly.

Now I need to achieve the exact same thing, but then using JLink only. So I wrote a script (well, I wrote many by now, but none of them work, and they all boil down to the following)

```
reset
loadfile bt-fuse-sel-1024.bin 0x20000000 noreset
halt
setpc 0x20000000
go
```

As I said, I wrote many, I tried this with and without `noreset`, I tried this with the .bin file, but also with the .axf file. I tried `setpc 0x20000004` and many other location (because I know the reset handler sits at `0x20000004`). All to no avail. 

So my question, if you look at this, do you see what I am doing wrong / what I am missing? Because you must do something similar inside your library to instruct JLink to do whatever it's doing, which in fact successfully launches my app from RAM.

Hope you can help me out. Although I perfectly understand if you won't :)

haneefdm, 2023-02-15:
@basprins This is way beyond my pay grade.

One idea. Have you considered running a script/progam using VSCode's built-in `preLaunchTask` to automate some of this stuff? You can also disable cortex-debug from programming/resetting the device at all and rely on your script to do it all.

Maybe someone can come in and help?

basprins, 2023-02-16:
@haneefdm 
Figured it out eventually.... And of course, it's a tremendously stupid mistake on my end.

After some time some brain activity took place in my head, and I checked the memory after loading the bin file.

```
J-Link>mem32 0x20000000,40
20000000 = 20010000 200002DD 20000361 200045D1 
20000010 = 20000365 20000367 20000369 00000000 
20000020 = 00000000 00000000 00000000 2000036B 
20000030 = 2000036D 00000000 2000036F 20000371 
20000040 = 20004161 20004169 20004171 20004179 
20000050 = 20004181 20004189 20004191 20004199
```

Remember my script, I am setting the PC to 0x20000000. What a joke. When I set the PC to the actual address of the reset handler, things work a lot better :).

```
halt
loadfile bt-fuse-sel-1024.bin 0x20000000 noreset
setpc 0x200002DD
go
```

And magic happens.. problem solved. 

Sorry for the bother! And thanks a lot the attempt to still help out.

closed 2023-02-16 by basprins

haneefdm, 2023-02-16:
Nice, you found the issue. If you adjust your PC, don't you also have to adjust your SP/LR to some reasonable values? Nut you are the expert.

Have fun

basprins, 2023-02-17:
Nah, they get set in the startup code.
stack pointer is the first word in the vector table.
The load register will be filled by the processor when it jumps to a functions. So that would be when it jumps from the reset handler to the fietst functions i think (since reset handler itself is naked)