Marus/cortex-debug #828: Better look of the Reset buttom

Eplankton, 2023-02-18:
Well....It appears that the current icon of `reset` doesn't render well on my screen, and it looks just blur, I suggest some svg to replace it.
They're lightning-like, for example:
[image]
[image]
[image]

haneefdm, 2023-02-18:
Thank you for taking the time. Not sure what theme you are using but in general I thought the existing one looked okay and maybe not the best for light themes. Btw, our Reset icon is already an SVG. I do want to stick with the conventions and guidelines from Microsoft/VSCode. I want it to resemble something like a power/restart/reset button. Here is my attempt at it in various themes. I started with the SVG that VSCode had for the Restart button. When using SVG, the light/dark modes don't work well unless you supply an icon for each. It takes some work to create one icon that works for all themes.

Abyss
[image]

Dark+ (default)
[image]

Light+ (default)
[image]

haneefdm, 2023-02-18:
Btw, you can experiment with SVGs yourself with various themes by replacing the file

~/.vscode/extensions/marus25.cortex-debug-X.Y.Z/images/reset.svg

or you can perhaps edit 

~/.vscode/extensions/marus25.cortex-debug-X.Y.Z/package.json

haneefdm, 2023-02-18:
Here are the two SVG files, if someone wants to tweak them.

[image]

[image]

And, here is the MS one for the built-in restart
[image]

And, my package.json looks like this now
```JSON
            {
                "command": "cortex-debug.resetDevice",
                "title": "Reset device",
                "icon": {
                    "dark": "images/reset-dark.svg",
                    "light": "images/reset-light.svg"
                }
            },
```

selectstriker2, 2023-02-19:
I think the reset icon looks just fine, and a I'm not sure that a lightning bolt makes sense in that context. It is interesting to know that the icons are tweakable though.

Eplankton, 2023-02-19:
> 这是两个 SVG 文件，如果有人想调整它们的话。
> 
> [image]
> 
> [image]
> 
> 而且，这是用于内置重启的 MS [image]
> 
> 而且，我的 package.json 现在看起来像这样
> 
> ```json
>             {
>                 "command": "cortex-debug.resetDevice",
>                 "title": "Reset device",
>                 "icon": {
>                     "dark": "images/reset-dark.svg",
>                     "light": "images/reset-light.svg"
>                 }
>             },
> ```

Thanks, I shall replace it by myself, and it just looks awesome.

PhilippHaefele, 2023-02-19:
@haneefdm Just for clarification. Are you planning to replace the icon by the one(s) you showed above?

I like them more than the current one as it is more clear that it is a kind of reset

Eplankton, 2023-02-19:
> @haneefdm Just for clarification. Are you planning to replace the icon by the one(s) you showed above?
> 
> I like them more than the current one as it is more clear that it is a kind of reset

Oh, sorry...I don't mean that. I just replace the icon on my OWN pc since the developer has told me where the file is, nothing else.

haneefdm, 2023-02-19:
@PhilippHaefele Yes, I will replace the Reset button in the next release. As you can see, they work much better for various themes and looks more like a Reset. I found the SVG for the MS Restart, now that MS published all of their icons.

I am going to close this

closed 2023-02-19 by haneefdm