Marus/cortex-debug #829: Better error message when GDB fails to start

sigurdo, 2023-02-25:
Hey!

I just spent quite some time today to figure out why I wasn't able to debug from my Arch WSL distro while everything worked perfectly in normal Windows 11. The problem seemed to be that Arch Linux does not have libncurses v5, but only v6 by default, and this is fine, I should have to deal with such issues when using Arch, but it would have been a lot easier to find out what the issue was if the error message from GDB was printed somewhere in the debug log in vscode. Instead, in my case it only displayed this:

[image]

After a while I figured out it would make sense to try to launch GDB from the command line myself and I got the error below, and I just felt like it would have made so much more sense if this was also printed in vscode. When I solved this, it worked perfectly in vscode as well.

```
arm-none-eabi-gdb: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```

haneefdm, 2023-02-26:
Wouldn't that be nice. Except, we never see anything in stdout or stderr. Just gdb abruptly quitting or worse hanging in various ways that we have no idea. Seems to change with every linux release and gdb release. We even wait for 5 seconds to see what went wrong and print this, and this used to work because it failed in yet another way we were not expecting -- I made a small change that can hopefully help but hard to test for these things.

```javascript
            this.log('log', 'Error: Unable to start GDB even after 5 seconds or it couldn\'t even start ' +
                'Make sure you can start gdb from the command-line and run any command like "echo hello".\n');
            this.log('log', '    If you cannot, it is most likely because "libncurses" or "python" is not installed. Some GDBs require these\n');
```

Linux has nothing but a PITA. if it is not libncurses, it is python (a specific version) or something else. Our problem is that no errors are ever reported to us.

This is also why we print every command exactly the way we invoke it.

Please don't hesitate to debug/edit our source code and suggest any improvements.

closed 2023-02-26 by haneefdm

sigurdo, 2023-02-26:
I have no choice other than to believe what you say, and it is really a shame. I saw that code, but that error message is not printed for me. I totally believe you when you say it is a PITA, the code actually does a lot of stuff to try to give reasonable feedback, but still it somehow doesn't work in my case.

I have tried to look at the code myself, and did another attempt at trying to understand what is going on, but I almost getting more confused. The timeout after 5 seconds never occurs. You might say it is because it is cleared, but AFAICS it isn't and I even tried making my own timeouts after 1000ms and 1ms, but neither of those occur either. I have a strong feeling that something is going profoundly wrong with the extension's source code, but I have no idea what, and totally respect that you don't bother reproducing this potentially bizarre case to fix it. I might come back to it some day myself, but most likely not. Just keep the issue closed.

sigurdo, 2023-02-26:
You can look at the stuff I tried [here](https://github.com/Marus/cortex-debug/compare/master...sigurdo:cortex-debug-print-gdb-libncurses5-error:master) if you're interested.

haneefdm, 2023-02-26:
Having tried to deal with this and spent weeks trying to figure out why it breaks with every Linux/gdb release, and you saying **I don't bother** is very disrespectful and hurtful. I can't count how many VMs I created over time. Yes, this is the thanks I get. Awesome.

If you want to make a PR please do so. Test the hell out of it though.

Why don't you file an issue with people who build/release gdb without full dependencies? Why is it everyone else's problem? This is technically, not a bug in our SW.

Problem is that the error message does not come from gdb itself. It is coming from the system's link loader and it is not on gdb's own stdout/stderr, because gdb hasn't even started yet. This is what I think. But it used to behave very differently last time I tested and put those damn messages. Now, I see a new way it is failing. And, thank you for giving me the details on your OS, gdb distribution, etc. Super helpful.

sigurdo, 2023-02-26:
Oh, I think I misunderstood something then, I thought you talked about having dealt with other problems  of similar nature and agreed that they were not worth fixing. I was by no means sarcastic and did totally not mean to disrespect you. I also did not imagine you had already spent that much effort on the problem.

I know my work is shitty because it is, and I have no experience with developing vscode extensions🙃.

I don't think I got your point in the last paragraph. Was it sarcastic or not? I find sarcasm really hard to get in pure text.