Marus/cortex-debug #830: Failed to launch OpenOCD GDB Server: Timeout on versions newer than 0.4.0

DarkCowMoo, 2023-02-26:
Hi,

I am trying to figure out why I keep receving a timeout error when working on a version newer than 0.4.0 (had to search manually which version works properly).

`Cortex-Debug: VSCode debugger extension version 1.6.10 git(0458417). 
Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /usr/bin/objdump-multiarch --syms -C -h -w fcos_v3.14.0.elf
Reading symbols from /usr/bin/nm-multiarch --defined-only -S -l -C -p /cos.elf
Launching GDB: /usr/bin/gdb-multiarch -q --interpreter=mi2 -ex "target remote localhost:50000" -ex load -ex "file fcos.elf"
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002"  -f marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl -f NUCLEOF746ZG/openocd.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 74 ms
warning: Architecture rejected target-supplied description
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.

Truncated register 16 in remote 'g' packet
No executable file specified.
Use the "file" or "exec-file" command.
Remote debugging using localhost:50000
Finished reading symbols from nm: Time: 1130 ms
Failed to launch OpenOCD GDB Server: Timeout.

With debug on -

-> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
Remote debugging using localhost:50000
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/shauli/.vscode/extensions/marus25.cortex-debug-1.6.10/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/shauli/.vscode/extensions/marus25.cortex-debug-1.6.10/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "/home/shauli/Git/flightcontroller_mark3/obj/main/fcos_v3.14.0-a1_WOLVERINE_V2_P2_XTEND_FC_XM00471_X00.elf"
Finished reading symbols from nm: Time: 1176 ms
-> 8^done
Failed to launch OpenOCD GDB Server: Timeout.

With 0.4.0 everything simply works -

Launching server: "openocd" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-f" "NUCLEOF746ZG/openocd.cfg"
Launching GDB: "/usr/bin/gdb-multiarch" "-q" "--interpreter=mi2" "-ex" "target remote localhost:50000" "-ex" "load"
1-gdb-set target-async on
2-interpreter-exec console "source /.vscode/extensions/marus25.cortex-debug-0.4.0/support/gdbsupport.init"
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
5-target-select extended-remote localhost:50000
6-interpreter-exec console "monitor reset halt"
7-target-download
8-interpreter-exec console "monitor reset halt"
9-enable-pretty-printing
10-interpreter-exec console "set remotetimeout 10000"

Thanks,
Shauli

haneefdm, 2023-02-26:
Few things
1. Can you attach your launch.json? I am not sure what you are doing and why. Why is this done? `-ex "target remote localhost:50000" -ex load -ex "file fcos.elf"`?
2. Between 0.4.0 and whichever version (you did not specify) you are trying, why are gdb options different? It appears your so-called newer version does not know how to talk to GDB. Remember, we don't talk to openocd at all. It is private communication between gdb and openocd.
3. You are not showing the output of OpenOCD, which makes this super difficult.

Why not run the two versions of OpenOCD from the command-line and see what the differences are? If you are using ST devices, we HIGHLY recommend using openocd from their Cubd IDE install. Not some random stuff from the internet. It also needs to be configured the way ST does it with their IDE (you can see all of that in Eclipse).

DarkCowMoo, 2023-02-26:
Hi @haneefdm ,
Thank you for the quick response.
My bad, I am running the latest version available on VSCode (1.4.0).

launch.json -
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/obj/main/fcos.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "interface": "jtag",
            "gdbPath": "/usr/bin/gdb-multiarch",
            "svdFile": "${workspaceFolder}/dev/svd/STM32F7x6.svd",
            "preLaunchTask": "build",
            "serverTimeout": 60000,
            "serverLaunchTimeout": 20000,
            "serverReadyTimeout": 20000,
            "attachTimeout": 20000,
            // "showDevDebugOutput": "raw",
            "debuggerArgs": [
                "-ex", "target remote localhost:50000",
                "-ex", "load",
            ],
            
            "configFiles": [
                "${workspaceFolder}/obj/main/NUCLEOF746ZG/openocd.cfg"
            ],

            "postLaunchCommands": [
                // "monitor sleep 200",
                "set remotetimeout 10000",
            ],

            // First break at function main()
            "runToEntryPoint": "main",
        }
    ]
}

OpenOCD output -

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v38 API v2 SWIM v27 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 0.000000
Error: target voltage may be too low for reliable debugging
Warn : Silicon bug: single stepping will enter pending exception handler!
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10016449
Info : flash size = 1024kbytes
Info : dropped 'gdb' connection

Regarding your question -
I did not change the version on OpenOCD- I simply ran version 1.4.0 and a downgraded version of 0.4.0.

The first GDB command is a bit different because I added the '-file' flag, thinking this might be the issue.
Thanks,
Shauli

haneefdm, 2023-02-26:
I am super confused. 1.4.0 and 0.4.0 of what? Those version do not jar a memory at all.

Our extension is
```
`Cortex-Debug: VSCode debugger extension version 1.6.10 git(https://github.com/Marus/cortex-debug/commit/0458417b2c01e50e2a367521d9c0504d8c4b6fbc).
```
OpenOCD seems to be an obsolete one
```
Open On-Chip Debugger 0.10.0
```

None of these in your launch.json make sense to us -- not sure what you are experimenting with.
```
"serverTimeout": 60000,
"serverLaunchTimeout": 20000,
"serverReadyTimeout": 20000,
"attachTimeout": 20000,
```

And, you gdb build also does not make sense. It even told you that it doesn't understand what arch it is dealing with. The generic gdb that comes with linux does not understand ARM embedded processors. GDB can be build a million different ways and some are specific to certain target architectures. `multiarch` simply means more than one arch. Doesn't mean ALL
```
"gdbPath": "/usr/bin/gdb-multiarch",
...
warning: Architecture rejected target-supplied description
warning: No executable has been specified and target does not support
determining executable automatically. Try using the "file" command.
```

We have wrong versions of pretty much everything. If you cannot make these tools via command-line, there is no hope of making them work in VSCode.

haneefdm, 2023-02-26:
OpenOCD seems to be a pretty old one. 0.11.x is the minimum you should be using. 0.10.x is from 2017.

Again, I stress that you should either use the versions of GNU tools shipped by ST Cube IDE or you have to know everything about why you want to use custom/ramdom tools.

I am going to close this for now. You can run everything on the command-line and prove that if things work or not.

closed 2023-02-26 by haneefdm