Marus/cortex-debug #831: Unable to read variables at 2nd breakpoint stop

tdjastrzebski, 2023-03-01:
I experience this problem probably since always but now it really stops me from debugging so I report it.
The first time execution stops at a breakpoint everything works as expected, variable values properly show up in RUN AND DEBUG 

[image]

At the second stop variables do not show up in RUN AND DEBUG panel, but on hoover they still do.

[image]
And, finally, at the third stop variables do not show at all. On hoover only little empty box is displayed.  The issue is probably related to #808 

DEBUG CONSOLE window below 

2nd stop
[image]

3rd stop and hoover
[image]

Environment: OpenOCD 0.12, GCC 12.2.1, Cortex-Debug the latest 1.6.1.
With OpenOCD 0.11 and GCC 10.3 results are the same.

tdjastrzebski, 2023-03-02:
Cortex-Debug 1.6.11-pre4 seems to fix the hoover issue but variable values in  RUN AND DEBUG panel still do not show up during the 2nd and subsequent breakpoint stops.

haneefdm, 2023-03-02:
Can I see your launch.json and the output from the Debug Console?

tdjastrzebski, 2023-03-02:
@haneefdm Here it is, console output attached.
```JSON
{
    "name": "Debug (OpenOCD)",
    "type":"cortex-debug",
    "request": "launch",
    "servertype":"openocd",
    "showDevDebugOutput": "both",
    "cwd": "${workspaceRoot}",
    "executable": "BUILD/debug/${workspaceRootFolderName}.elf",
    "svdFile": "STM32L4x2.svd",
    "runToEntryPoint": "main",
    "device": "STM32L432KCUx",
    "searchDir": ["${env:OpenOCD}/scripts"],
    "serverArgs": [
        "-c hla_serial 066EFF535488524867153759"
    ],
    "configFiles": [
        "/interface/stlink.cfg",
        "/target/stm32l4x.cfg",
    ],
    "interface": "swd",
    "swoConfig": {
        "enabled": true,
        "cpuFrequency": 180000000,
        "swoFrequency": 2000000,
        "source": "probe",
        "decoders": [
            { "type": "console", "label": "ITM", "port": 0 }
        ]
    },
}
```
[console ouput.txt](https://github.com/Marus/cortex-debug/files/10873908/console.ouput.txt)

haneefdm, 2023-03-02:
Is there anything in the STACK window? If you don't see a stack, then you will not see any variables either.

You can also set "showDevDebugOutput": "vscode" and attach that console output? We are pretty sure we sent VSCode the stack information and it then has to request variables -- which it kinda seems like it did.

tdjastrzebski, 2023-03-02:
CALL STACK info is always displayed correctly, DEBUG output with "showDevDebugOutput": "vscode" atached.
[console ouput 2.txt](https://github.com/Marus/cortex-debug/files/10875893/console.ouput.2.txt)
[image]

haneefdm, 2023-03-03:
Aha. I know exactly why that is happening. Bug in our SW and perhaps even a change in how more recent versions of gdb work. A fix is on the way

haneefdm, 2023-03-03:
Do you mind trying the latest 1.7.x pre-release?

https://github.com/Marus/cortex-debug/releases

Eager to hear feedback

haneefdm, 2023-03-03:
If you enable pre-releases within VSCode, this release will also be available over there. Note that this release has a LOT of changes -- mostly related to this issue. New features as well like `Live Watch`. A LOT of code had to be re-written/re-factored

tdjastrzebski, 2023-03-04:
@haneefdm Thank you for your help. It seems that in 1.7.0-pre1 the problem still persist. Console output attached.
[console output 3.txt](https://github.com/Marus/cortex-debug/files/10888036/console.output.3.txt)

haneefdm, 2023-03-04:
Could you try this release please.

https://github.com/Marus/cortex-debug/releases/tag/v1.7.1-pre1

This was a huge change in our code base. I had been testing it for a while and it was already supposed to be in 1.7.0 but I missed it.

tdjastrzebski, 2023-03-04:
As far as I can tell, now it works as expected! Thank you for your effort, it is much appreciated.

haneefdm, 2023-03-04:
Thanks for reporting back.

Do, globals, statics in the 'VARIABLES' Panel, and expressions in the 'WATCH' windows also work properly? I know, it is hard to tell...

Take your time, no hurry.

haneefdm, 2023-03-10:
I am going to close this issue...

closed 2023-03-10 by haneefdm

tdjastrzebski, 2023-03-11:
Everything seems to works as expected, thanks again, it is probably one of the most important issue fix.