Marus/cortex-debug #832: Issue decoding templatized class when combining cortex-debug with iar-vsc-build

jwbmwv, 2023-03-01:
I am not certain which piece of the puzzle is responsible for the error I am getting and may simply be asking for help identifying the guilty piece. I am using a combination of C/C++, cortex-debug, iar-build extension and jlnk.

I have a C++ class that is defined via a template (ByteArray) that inherits from another templatized class (Array). When I hit a breakpoint and try to look at the decoded variable, I get a popup that says:
"Could not expand variable: Variable object not found (from var-list-children --all-values var_txByteArray_256.Array<unsigned char>)"

If I use the iar-debug extension instead of cortex-debug on the same image, I can decode the class.. Is is slow and I miss having the built-in RTT support of cortex-debug. 

Any help getting me pointed in the right direction would be appreciated.

haneefdm, 2023-03-01:
Post your relevant launch.json, Also post output of DebugConsole with debug enabled (see instructions in Debug Console). Maybe I or others can help.

Also, snippets of your C++ code can help. As you can tell, we are working blind. Compiler we don't use and anything you can provide to help us understand the issue better is super helpful. Even some small screenshots.

jwbmwv, 2023-03-10:
Sorry it took so long, but This zip file should have everything you've asked for. I could provide a complete, compiled project for a nucleo board (I used a nucleo-H723ZG, but have a few others) if it would be helpful. I typically use a jlink, but also see the problem using the built-in stlink,

Thanks
[jwbmwv_files_1.zip](https://github.com/Marus/cortex-debug/files/10945123/jwbmwv_files_1.zip)

haneefdm, 2023-03-10:
Before I do anything, could you please try out our pre-release 1.9.x It is on the marketplace and you can download it from within VSCode.

jwbmwv, 2023-03-10:
I was using the latest pre-release from this morning when I grabbed that data.; 1.9.2

haneefdm, 2023-03-10:
I can take this source code and put it into my own project I suppose.

Is this IAR specific? I don't have an IAR license. Can you also reproduce with GCC?

jwbmwv, 2023-03-10:
I've only experienced this with IAR, but will give a quick test with GCC now.
...
And GCC (the version included with the latest STM32CubeIDE) also has the problem using an st-link

haneefdm, 2023-03-10:
Yes. Just duplicated it with GCC. I can see. Don't understand yet.

In this case, st-link, vs jlink or openocd doesn't matter. It is all frontend stuff.

haneefdm, 2023-03-10:
Found the problem and fixed it. The colons were throwing off the gdb parser, so I had to quote the whole thing. Our mistake

[image]

I will be working on a new pre-release shortly and this change will be in there. Wish everyone could supply me with details so I can get to the bottom of issues.

jwbmwv, 2023-03-10:
That's great. Thanks

haneefdm, 2023-03-10:
I will do a pre-release over the weekend

haneefdm, 2023-03-10:
You can see a credit to you in the changelog

https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md

I am going to close this for now, comment and I can re-open. Trying to reduce my backlog

closed 2023-03-10 by haneefdm