Marus/cortex-debug #833: External GDB service fails to start correctly

hongshui3000, 2023-03-03:
I want the extension to help me start an external gdb service and listen to the localhost:20000 gdb port for debugging, but it seems that the extension does not start my external gdb service, and it directly listens to localhost:20000
my launch config
```
            "servertype": "external",
            "serverpath": "C:/Program Files (x86)/Tensilica/Xtensa OCD Daemon 14.06/xt-ocd.exe",
            "serverArgs": ["-c ${workspaceFolder}/.vscode/xt-ocd-config.xml"],
```
my log
[log.txt](https://github.com/Marus/cortex-debug/files/10877620/log.txt)

my try
I execute xt-ocd.exe -c ${workspaceFolder}/.vscode/xt-ocd-config.xml in the terminal to start the service and then use the extended connection gdb to debug and it can work normally.It seems that I have to manually start the service to work

haneefdm, 2023-03-03:
Why would we help in starting an external server? "external" means you start it with whatever means you got, on whatever ports you got, and whatever computer you got. You just tell us via `gdbTarget` how to connect. `serverpath` and `serverArgs` have no meaning.

"external" may also mean limited functionality since we have no clue what its capabilities are. Things like swo/rtt will not work.

closed 2023-03-03 by haneefdm

hongshui3000, 2023-03-03:
> Why would we help in starting an external server? "external" means you start it with whatever means you got, on whatever ports you got, and whatever computer you got. You just tell us via [...]

I can start the external service myself, but I can't automatically know when to enter debug or exit debug.

The external service is used because, it's a debug service from the processor vendor I'm using, their service supports jlink, and I can't use jlink without using their debug service. jlink official does not support my core

haneefdm, 2023-03-03:
Once the server is started, you can enter and exit debug sessions anytime you want. gdb connects and disconnects when you order it.

However, some servers are designed to run only one session and some run for ever. That is up to you and your server. We cannot control that.

"external" literally means "external". We have zero interaction with controlling it other than gdb trying to connect/disconnect

hongshui3000, 2023-03-03:
Can I control the start of cortex debug and monitor the stop of cortex debug, if I write another extension

hongshui3000, 2023-03-03:
Is there any way to make cortex debug start a specific launch configuration?

hongshui3000, 2023-03-03:
Seems like ‘cppdbg’ will help me start external services.
```
     "miDebuggerServerAddress": "localhost:20000",
     "debugServerPath": "~/Tensilica/xocd-14.04/xt-ocd",
      "debugServerArgs": "-c ${workspaceFolder}/.vscode/xt-ocd-config.xml",
```