Marus/cortex-debug #836: Pico Debug literally does nothing and will not launch openocd

mytechnotalent, 2023-03-04:
running the Pico Debug within VS Code literally does nothing.  I have openocd installed and can do everything manually on the terminal however this extension does not work
```
--config Debug --target badge1 --
[build] ninja: no work to do.
[build] Build finished with exit code 0
```

launch.json
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "Pico Debug (Cortex-Debug)",
      "cwd": "${workspaceFolder}",
      "executable": "${command:cmake.launchTargetPath}",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "gdbPath": "arm-none-eabi-gdb",
      "device": "RP2040",
      "configFiles": [
        "interface/cmsis-dap.cfg",
        "target/rp2040.cfg"
      ],
      "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
      "runToEntryPoint": "main",
      "openOCDLaunchCommands": [
        "adapter speed 1000"
      ]
    },
    {
      "name": "Pico Debug (Cortex-Debug with external OpenOCD)",
      "cwd": "${workspaceFolder}",
      "executable": "${command:cmake.launchTargetPath}",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "external",
      "gdbTarget": "localhost:3333",
      "gdbPath": "arm-none-eabi-gdb",
      "device": "RP2040",
      "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
      "runToEntryPoint": "main"
    },
    {
      "name": "Pico Debug (C++ Debugger)",
      "type": "cppdbg",
      "request": "launch",
      "cwd": "${workspaceFolder}",
      "program": "${command:cmake.launchTargetPath}",
      "MIMode": "gdb",
      "miDebuggerPath": "arm-none-eabi-gdb",
      "miDebuggerServerAddress": "localhost:3333",
      "debugServerPath": "openocd",
      "debugServerArgs": "-f interface/cmsis-dap.cfg -f target/rp2040.cfg -c \"adapter speed 1000\"",
      "serverStarted": "Listening on port .* for gdb connections",
      "filterStderr": true,
      "stopAtEntry": true,
      "hardwareBreakpoints": {
        "require": true,
        "limit": 4
      },
      "preLaunchTask": "Flash",
      "svdPath": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd"
    }
  ]
}
```

haneefdm, 2023-03-04:
What do you see in the `Debug COnsole`? It has instructions to enable verbose output. Also, what is the output from openocd? I don't see any information in your bug report about what/how your manual stuff worked and how or what our extension did.

mytechnotalent, 2023-03-04:
@haneefdm that was literally the totality of the info in the `Debug Console`.  OpenOCD performs as expected however VSCode literally does nothing when attempting to use the debug tool.

```
--config Debug --target badge1 --
[build] ninja: no work to do.
[build] Build finished with exit code 0
```

haneefdm, 2023-03-05:
That (ninja) has NOTHING to do with our debugger. That seems to be your build output. I am saying "Debug Console" where output goes and instructions as well. It looks something like this

[image]

If you are new to VSCode, I suggest reading this

https://code.visualstudio.com/docs/editor/debugging

haneefdm, 2023-03-05:
A screenshot like I sent can be helpful because I am beyond confused

mytechnotalent, 2023-03-05:
I can't reproduce on my other Windows machine.  The output from the `DEBUG CONSOLE` literally is the following.
```
Cortex-Debug: VSCode debugger extension version 1.6.10 git(0458417). Usage info: https://github.com/Marus/cortex-debug#usage
```

This appears after attempting to press the play button on the Pico Debug (Cortex -Debug) UI.

I will close the issue.

closed 2023-03-05 by mytechnotalent