Marus/cortex-debug #837: Trying to use STLink debugger

jagauthier, 2023-03-05:
Quick background: I am trying to development on a multicore STM32H745.  In the past, I've used VScode, OpenOCD and did not use the 2nd (M4) processor.   Now, I am working on a new project and I want to use both and debug both simultaneously. I could not figure out how to do it with OpenOCD, so I opened the project with STM32CubeIDE and saw what they were doing with the STLink GDB server (couldn't get OpenOCD to work in their environment either). So, I started up an stlink server and using their IDE I was able to launch into  debug mode and debug both processors at the same time.  So, I took their STLink GDB commands and I am trying to reproduce them with my launch.json and I can't seem to get it right.

Their commands for CM7:
`C:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.400.202209281104\tools\bin\ST-LINK_gdbserver.exe -p 61234 -l 1 -d -t -s -cp C:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.500.202209151145\tools\bin -m 0 -k`

For the CM4:
`C:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.400.202209281104\tools\bin\ST-LINK_gdbserver.exe -p 61236-l 1 -d -t -s -cp C:\ST\STM32CubeIDE_1.6.1\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.500.202209151145\tools\bin -m 3 -k`

My launch for the CM7:
```
   {
      "name": "Cortex Debug M7 ST-Link",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/build/EEMon_CM7.elf",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "stlink",
      "stlinkPath": "ST-LINK_gdbserver.exe",
      "stm32cubeprogrammer": "",
      "device": "STM32H745ZIT6",
      "runToEntryPoint": "main",
      "preLaunchTask": "Build",
      "svdFile": "${workspaceRoot}/STM32H745_CM7.svd",
      "showDevDebugOutput": "raw",
      "serialNumber": "0044001F3137510E39383538",
    },
```

For the CM4:
```
      {
        "name": "Cortex Debug M4 - ST-Link",
        "cwd": "${workspaceRoot}",
        "executable": "${workspaceRoot}/build/EEMon_CM4.elf",
        "request": "attach",
        "type": "cortex-debug",
        "servertype": "stlink",
        "stlinkPath": "ST-LINK_gdbserver.exe",
        "stm32cubeprogrammer": "",
        "device": "STM32H745ZIT6",
        "runToEntryPoint": "main",
        "preLaunchTask": "Build",
        "svdFile": "${workspaceRoot}/STM32H745_CM4.svd",
        "showDevDebugOutput": "raw",
        "serialNumber": "0044001F3137510E39383538",
        "numberOfProcessors": 2,
        "targetProcessor": 1
      }   
```

The generated command looks like this:
CM7
`ST-LINK_gdbserver.exe -p 50000 -cp "C:\\ST\\STM32CubeIDE_1.10.1\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.500.202209151145\\tools\\bin" --shared --swd --serial-number 0044001F3137510E39383538 --halt`
CM4
`ST-LINK_gdbserver.exe -p 50000 -cp "C:\\ST\\STM32CubeIDE_1.10.1\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.500.202209151145\\tools\\bin" --shared --swd --serial-number 0044001F3137510E39383538 --attach --halt`

So, I need some help identifying the launch options to configure the differences between theirs and mine.
Those are:
```
   -p <port number>,  --port-number <port number>
     TCP port number for GDB client
   -m <apID>,  --apid <apID>
     Select core on multi-core devices
   -k,  --initialize-reset
     Initialize the device under reset condition
```

And then lastly. how to remove --halt.
With those commands being alike, I am hoping that I can then debug both cores using my two workspaces in vscode.

Thank you!

haneefdm, 2023-03-05:
We have to halt the program for either launch or attach even if briefly for VSCode to work. Multi-core debug works with OpenOCD, JLink or STLink, but it is tricky. Here is an example...

https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging#st-link-notes

jagauthier, 2023-03-05:
Thank you. I was able to work through those examples and put the appropriate information into each workspace's launch,json and I was able to debug both processors at the same time from two different instances.  
It's far from perfect, but it's definitely an improvement so I can hopefully get done what I am trying to do!

closed 2023-03-05 by jagauthier

haneefdm, 2023-03-05:
Well, we did the best we could think of. Given so many external tools that we have no control over including VSCode. And, every gdb-server is a bit different. And, every device is different from each vendor on how they start the processors.

I am glad you got something working.

haneefdm, 2023-03-05:
Also, you can edit the Wiki (another user contributed that section) if you find better ways of doing things.

jagauthier, 2023-03-05:
"Far from perfect" was not meant to be a jab.  I'm using VSCode and cortex-debug as a choice because STM32CubeIDE is "far from perfect".  All I have to do it write enough code on one processor and then leave it be and I can focus my efforts on the other one.  If it becomes too frustrating, I can switch the CubeIDE for that portion and them move back to VSCode for the heavy lifting.  I appreciate the response.