Marus/cortex-debug #838: Remove Cortex Registers view

PhilippHaefele, 2023-03-06:
As with V1.2.0 registers moved to the variable section the `Cortex Registers` view should be removed. 

From release notes:
> **Registers**: `Registers` are now available in the `Variables` panel. You can now change the values of registers. But more importantly, unlike the `Registers` panel, this will track your current thread/frame in the `Call Stack` window. This means that registers are now shown in the context of the current frame. What we had before in the `Registers` panel was information we had at the time a halt/breakpoint was hit and potentially incorrect when you refreshed -- this was because there was no API in VSCode for extensions to track the Call Stack window. **`The old Registers panel will be DEPRECATED`**

We had a lot of deprecation time (over 1 year now).

Therefore i suggest to remove it in the upcoming release (pre-release 1.9.0).

haneefdm, 2023-03-06:
Hehe. I am scared of doing it. Because people think it is actually accurate and I am sure there will be complaints. I was hoping VSCode would fix the issue where we could track the current program/thread/frame. The issue on VSCode/MS side is still open.

haneefdm, 2023-03-06:
Yes, I will hide it in the next release.

PhilippHaefele, 2023-03-06:
> Yes, I will hide it in the next release.

Thanks

One thing we need to consider when removing the view, is that it currently houses the `Enable Hex mode` / `Enable Natural mode` button. There's a command for that, but maybe we can move this button somewhere else (variables view?)

haneefdm, 2023-03-06:
Yup. Doesn't work. That panel/window is created much before cortex-debug is enabled and it appears that extensions cannot add to the title bars. Strange but true.

Registers Panel is now gone.

closed 2023-03-06 by haneefdm

haneefdm, 2023-03-09:
> Back then, I didn't know how to add buttons to one of VSCode built-in panels. Maybe it is possible now.

I finally figured out how to add a button to the VARIABLES title bar. Very obtuse way. Will be removing `Enable/Disable hex mode` from the context menu as it is really global to the entire window.

[image]

I wanted to add at the beginning but I don't know how to

PhilippHaefele, 2023-03-10:
> I finally figured out how to add a button to the VARIABLES title bar. Very obtuse way. Will be removing `Enable/Disable hex mode` from the context menu as it is really global to the entire window. [...]

Just tested it with the latest pre-relase and i really like it.