Marus/cortex-debug #841: Announcement: Peripheral (SVD) window now available in a separate extension

haneefdm, 2023-03-07:
https://marketplace.visualstudio.com/items?itemName=mcu-debug.peripheral-viewer

You don't have to do anything special if you were using Cortex-Debug launch.json for viewing Peripherals before. We also recommend using the latest pre-release (1.9.X) of Cortex-Debug or if you are using releases prior to 1.8.X you may see two peripheral views in VSCode

PhilippHaefele, 2023-03-13:
As of this change following issues need to be retested / closed / transferred.

1. https://github.com/Marus/cortex-debug/issues/509
2. https://github.com/Marus/cortex-debug/issues/736
3. https://github.com/Marus/cortex-debug/issues/800
4. https://github.com/Marus/cortex-debug/issues/804
5. https://github.com/Marus/cortex-debug/issues/805
6. https://github.com/Marus/cortex-debug/issues/806

@haneefdm Are you willing to ping all these issues as you started discussions in most of them already?

haneefdm, 2023-03-13:
Yes, it is on my list. I think most of them are now fixed but some I had to carry manually, fixes from this repo to the other one and I may have missed a couple.

haneefdm, 2023-03-13:
So, now, we are left two issues and they are dups. I cannot move these issues because I am not the owner of both repos.

closed 2025-01-17 by haneefdm