Marus/cortex-debug #842: Failed to launch OpenOCD GDB Server: Timeout.

matt001k, 2023-03-09:
The GDB server takes a while to run and it timeouts all of the time.
There is no way to increase the timeout length. This should be configurable.

The following is my debug console output:
```
Cortex-Debug: VSCode debugger extension version 1.8.0 git(6bc81e5). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from c:/users/matthewkrause/projects/nuvoton/nuvoton_firmware_tools/compiler/bin/arm-none-eabi-objdump.exe --syms -C -h -w C:/Users/matthewkrause/Projects/firmware/Main-M2354/build/nuvoton_secure.elf
Reading symbols from c:/users/matthewkrause/projects/nuvoton/nuvoton_firmware_tools/compiler/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users/matthewkrause/Projects/firmware/Main-M2354/build/nuvoton_secure.elf
Launching GDB: "C:\\Users\\matthewkrause\\Projects\\Nuvoton\\nuvoton_firmware_tools\\Compiler\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2
1-gdb-version
Launching gdb-server: "C:\\Users\\matthewkrause\\Projects\\Nuvoton\\nuvoton_firmware_tools\\OpenOCD\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Users\\matthewkrause\\Projects\\firmware\\Main-M2354\\util" -f "c:/Users/matthewkrause/.vscode/extensions/marus25.cortex-debug-1.8.0/support/openocd-helpers.tcl" -f nulink.cfg -f numicroM23.cfg
    Please check TERMINAL tab (gdb-server) for output from C:\Users\matthewkrause\Projects\Nuvoton\nuvoton_firmware_tools\OpenOCD\bin\openocd.exe
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.16)) 11.2.90.20220202-git\n"
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://bugs.linaro.org/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
Finished reading symbols from objdump: Time: 3740 ms
Finished reading symbols from nm: Time: 3606 ms
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/matthewkrause/.vscode/extensions/marus25.cortex-debug-1.8.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/matthewkrause/.vscode/extensions/marus25.cortex-debug-1.8.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "C:/Users/matthewkrause/Projects/firmware/Main-M2354/build/nuvoton_secure.elf"
-> 8^done
Failed to launch OpenOCD GDB Server: Timeout.
```
The following is my terminal output for openOCD:
```
Open On-Chip Debugger 0.10.0-dev-00476-gd1de37d6-dirty (2021-12-08-10:03)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link2
[2023-03-09T21:54:19.438Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

I expect the following to be ran from OpenOCD before the GDB client begins to run:

```
Open On-Chip Debugger 0.10.0-dev-00476-gd1de37d6-dirty (2021-12-08-10:03)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link2
Info : NULINK firmware_version(7215), product_id(0xxxxxxxx)
Info : IDCODE: 0xxxxxxxxx
Info : NuMicro.cpu: hardware has 4 breakpoints, 4 watchpoints
```

haneefdm, 2023-03-09:
20 seconds is an eternity. How long does it take if you run the same exact command line we are running? I also noticed that no gdb-ports were open. That is what we are waiting for. You will see a message from openocd indicating it is ready for business. You will see something like
```
Info : starting gdb server for psoc6.cpu.cm0 on 50000
Info : Listening on port 50000 for gdb connections
```
If you are not seeing that, then there is no way to talk to openocd. It is hung and waiting for hours/days does not matter.

You maybe using a super old version of openocd.

haneefdm, 2023-03-09:
In the Debug Console, you can see how we ran openocd. Try to run the exact same command and see what is wrong with that. Even when you ran openocd, it did not work, although you say a bit more output. We are not doing any magic here, it is all in the Debug Console

matt001k, 2023-03-10:
Looks fine to me...
See the below console output for two command line windows running expected commands from cortex-debug:
[image]
[image]

I was able to flash the device no problem and set a breakpoint at main and run to main.
I can script all of this debugging through vim to have debugging there, but many of my colleagues prefer .vscode and cortex-debug has worked fine for me in the past with j-link.
OpenOCD has always had problems.
What are the next steps to get this to work properly after seeing this information.

haneefdm, 2023-03-10:
Can't help you with OpenOCD, especially with a vendor-provided one. We are looking for that message and until we see that message, we know we cannot connect. Your version is from Jan 2017.

I would try the latest just in case it works. You may have to copy the openocd scripts director from 10 to latest.

I will have to close this as there is nothing on our end to fix.

haneefdm, 2023-03-10:
You also did not attach your launch.json so I have no idea what is in there

matt001k, 2023-03-10:
```
{
    "version": "0.2.0",
    "configurations": [
      {
        "type": "cortex-debug",
        "request": "launch",
        "name": "Flash Debug Cortex M23 Secure",
        "cwd": "${workspaceFolder}",
        "servertype": "openocd",
        "serverpath": "${env:NUVOTON_TOOLS}\\OpenOCD\\bin\\openocd.exe",
        "executable": "./build/nuvoton_secure.elf",
        "postLaunchCommands": [
          "monitor flash write_image erase ./build/nuvoton_secure.bin 0x00000000",
          "monitor flash write_image erase ./build/nuvoton_nonsecure.bin 0x00040000",
          "monitor reset init",
          "monitor reset run"
        ],
        "gdbPath": "${env:NUVOTON_TOOLS}\\Compiler\\bin\\arm-none-eabi-gdb.exe",
        "configFiles": [ "nulink.cfg", "numicroM23.cfg" ],
        "searchDir": [ 
            "${workspaceRoot}\\util"
        ],
        "showDevDebugOutput": "raw",
      }     
    ]
}
```
So we can't add a configurable regex to look for a different print statement?
I mean this seems pretty frail if you need to see a specific string or else it will never work.

haneefdm, 2023-03-10:
Yes, you can have a configurable regex... see

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

>> Also many chip manufacturers create one off releases of OpenOCD, how is this fair to not be backwards compatible?

How is fair to us if OpenOCD does not behave like it is supposed to. We don't have infinite resources to go back and check everything from 5 years ago.

closed 2023-03-10 by haneefdm

flamboyante, 2024-12-15:
@matt001k 
hello!  Has this issue been resolved? I've recently encountered the same situation when I'm using OpenOCD V10 provided by the chip manufacturer , performing cortex-debug in VSCode.                                                                                                            

Thanks for your reply.

matt001k, 2024-12-17:
@flamboyante I am not sure, this issue was part of the reason I switched to Neovim and debug from the command line now😄

flamboyante, 2024-12-24:
@matt001k ,Sorry for the late reply. I have resolved this issue, the cause was that the chip manufacturer was using an outdated version of OpenOCD. I needed to add "overrideGDBServerStartedRegex": "Info\s:\s([^\n\.])\.cpu([^\n])" in the launch.json. For more details, see issue #166 . Finally, I can debug successfully in VSCode. Thank you for your response. Hoep u have a good day.😄😄😄