Marus/cortex-debug #843: Error when using "symbolFiles"

oyvindronningstad, 2023-03-10:
I have a launch.json with the following configurations
```
			"request": "attach",
			"type": "cortex-debug",
			"servertype": "jlink",
			"device": "nRF9160",
```
If I use "executable" with these configurations it works, but if I instead use "symbolFiles" instead (with the same path inside an array), it gives an error:
[image]
The error appears whether or not "executable" is also specified.

I am using Zephyr (at ed2c25effccc55e459e2b96f389d9b1a177daf7d) and the nRF9160 dev kit. I build and flash the hello_world sample independently before starting the debug session. The path is an absolute path to the zephyr elf file. By "it works", I mean that it hits my breakpoint in the hello_world main function after I hit restart.

haneefdm, 2023-03-10:
Can you enable debug traces? Instructions in Debug Console. Also, a bit more of the launch.json can be helpful. What you provided gives me zero insight.

oyvindronningstad, 2023-03-10:
>Can you enable debug traces? Instructions in Debug Console

Sorry how do I do this? The debug console doesn't output anything before the error happens.

> Also, a bit more of the launch.json

This is my entire launch.json to provoke the error:

```
{
	"configurations": [
		{
			"name": "Cortex Debug",
			"symbolFiles": ["<path to Zephyr>/build/zephyr/zephyr.elf"],
			"request": "attach",
			"type": "cortex-debug",
			"servertype": "jlink",
			"device": "nRF9160",
		}
	]
}

oyvindronningstad, 2023-03-10:
Oh, found it now. After adding 
```
			"showDevDebugOutput": "raw",
```
there is still no output in the Debug Console (or Terminal) before the error happens (or after for that matter).

haneefdm, 2023-03-10:
There is no output because it did not even start.

Your symbolFiles is wrong. VSCode probably gave you are warning that it is the wrong format. It should be an object.
```JSON
    "symbolFiles": [
        {
            "file": "<path to Zephyr>/build/zephyr/zephyr.elf"
        }
     ]
```
Unfortunately, you will still need an executable although, it will not actually be used or even read. Technically, it is not needed if you have valid symbolFiles but for now, you have to specify one and it needs to exist on the file system

You should use IntelliSense as much as possible. If the format is expecting an object, you need to enclose it in `{}` and inside there if you type a `"` VSCode will tell you what is possible.

oyvindronningstad, 2023-03-10:
Thanks, that worked!

I did realize that intellisense was trying to tell me something, but the docs AFAICT didn't say how to format "symbolFiles", so I didn't know. I didn't know all the intellisense tricks. Maybe I'll open a PR to improve the docs. Thanks anyway!

closed 2023-03-10 by oyvindronningstad