Marus/cortex-debug #846: Problem flashing STM32H7 (multiple variations)

jwbmwv, 2023-03-11:
I have several STM32H7 projects that used to flash from cortex-debug just fine. In the last couple of weeks I've been getting errors. My workaround has been to flash from either STM32CubeIDE, Sewgger JFlash, or IAR Workbench then switch to cortex-debug.

VS Code 1.76.1
cortex-debug  1.11.0
Segger J-link v7.86c

I am having this issue on different projects built using gcc (STM32CubeIDE and/or MS Embedded tools conversion) or IAR.

I was also having the issue with Ozone project files before I regenerated them. The regenerated files deleted the check for SP != 0xFFFFFFFF as follows in the _SetupoTarget() function

  if (SP != 0xFFFFFFFF) {
    Target.SetReg("SP", SP);
  } else {
    Util.Error("Project script error: failed to set initial SP", 1);
  }
  
lost it's else and  became

if (SP != 0xFFFFFFFF) {
    Target.SetReg("SP", SP);
  }

This leads me to believe Segger changed something in their J-link software recently that is incompatible with cortex-debug on STM32H7's. I've submitted tickets/inquiries to segger with no response.

Based on the output below and the observed changes in Ozone, does my conclusion sound plausible? Is there a way to accommodate this in cortex-debug?

Thanks

Debug Console
Cortex-Debug: VSCode debugger extension version 1.11.0 git(0ff776a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:/ST/STM32CubeIDE_1.11.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954/tools/bin/arm-none-eabi-objdump.exe --syms -C -h -w C:/projects/fourtalker_nucleo-h723zg/build/debug/build/fourtalker_nucleo-h723zg.elf
Reading symbols from c:/st/stm32cubeide_1.11.2/stm32cubeide/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954/tools/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/projects/fourtalker_nucleo-h723zg/build/debug/build/fourtalker_nucleo-h723zg.elf
Launching GDB: "C:\\ST\\STM32CubeIDE_1.11.2\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954\\tools\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2
1-gdb-version
Launching gdb-server: "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32H723ZG -rtttelnetport 19021 -rtos GDBServer/RTOSPlugin_FreeRTOS.dll -gui -speed 50000
    Please check TERMINAL tab (gdb-server) for output from C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe
Finished reading symbols from objdump: Time: 114 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Tools for STM32 10.3-2021.10.20211105-1100) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
Finished reading symbols from nm: Time: 168 ms
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/jbaldwin/.vscode/extensions/marus25.cortex-debug-1.11.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/jbaldwin/.vscode/extensions/marus25.cortex-debug-1.11.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
[... 393 lines cut ...]
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFF4
Reading 64 bytes @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFF0
Reading 64 bytes @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFF0
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFFE
WARNING: Failed to read memory @ address 0xFFFFFFF4
Reading 64 bytes @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFF0
Reading 64 bytes @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFC0
WARNING: Failed to read memory @ address 0xFFFFFFF0

jwbmwv, 2023-03-11:
New revelation.

If I remove the line for FreeRTOS,
"rtos":             "FreeRTOS",

I can now flash and debug.

haneefdm, 2023-03-11:
The property `rtos` is not something we use or care. We simply pass it along to JLink/OpenOCD software. There is no compatibility issue that is possible unless the command-line format changed. We don't even communicate with JLink beyond the command-line argument. Ever. GDB does all the communication and it doesn't care about RTOSes either. See documentation

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

If we see threads reported by the gdb-server we automatically use them. We don't need to be told, really. Neither does GDB

JLink expects a path to a DLL (plugin) and not just a name. Silly but true. OpenOCD simply expects a name and it does the rest. Please look at the JLink gdb-server documentation. OpenOCD goes even further if you specify `auto` and it will figure it out. Jlink does not. They should have an auto option, IMO at least for the RTOS plugins they ship with. You should submit a request for that with SEGGER.

Now, I have specified the wrong RTOS before to JLink and it did not affect flashing or debugging. Just that no RTOS was used and I could not see any threads. Not sure what went wrong with your setup. Always pay attention to how we launch the gdb-server (JLinkGDBServerCL.exee) to see if we are doing it right and make sure the plugin exists. You should report a problem to SEGGER if it is not doing the right thing and the rtos is the only difference.

Bottom line, this is a JLinkGDBServerCL issue.

haneefdm, 2023-03-11:
For the "rtos", try specifying the full path name, just for kicks

closed 2023-03-12 by haneefdm