Marus/cortex-debug #848: [Windows Only] Let gdb drop elf file lock when press Restart button

eZioPan, 2023-03-14:
Cortex-Debug version: 1.10.0 and 1.11.0

when I press "Restart" button, vscode will rebuild elf, but `gdb-multiarch.exe` (from msys2) seems acquire elf file lock, which prevent elf rebuild.

I try manually launch `gdb-multiarch` (with or without elf as parameter), `target remote`, then rebuild elf, it does say gdb lock the file. But I can use `file` gdb command (without any parameter) force gdb unload elf, thus let elf to be rewrite.

I try to configure `launch.json`

```json
"preRestartCommands": [
    "monitor echo \"\\ntry release file lock before rebuild\\n\"",
    "file"
]
```

but it seems not working when I click `Restart` button.

Is there anything suggest to force gdb unload elf after `Restart` button being clicked?

haneefdm, 2023-03-14:
I've never heard of gdb holding a lock on the elf-file. Also, why on a re-start. The restart does not use the elf file anymore. Only at the start. I have a feeling you are building for restart and it is the one holding the lock: No clue. Never heard of such a thing in my 20+ years of using gdb. Only difference between restart and reset is that VSCode attempts a build (preLaunchTask I think) ... otherwise, they are identical. Also, even you rebuild, we do not reload the elf file.

haneefdm, 2023-03-14:
I think you maybe unloading yourself by doing a `file`  command. We don't do that. If you truly want to restart the whole session, it is best to Stop and Launch again. The Restart logic, I never understood and it changed a couple of times in the history of VSCode lifetime. That is why we introduced Reset which has a predictable behavior.

You are on your own and you seem to know more than I do.

haneefdm, 2023-03-14:
Also, the 'preRestart' happens after the build. VSCode does even tell us a restart is about to happen. Just like the launch. The `pre` is for running commands before a reset. Not before a build which we do not even know of.

You can learn more about how everything works in our extension....

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood#restart-or-reset

haneefdm, 2023-03-14:
If gdb is truly holding a lock on the elf file, then it should be reported to the gdb folks. This is silly. You can't even do what they suggest here...

https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_10.html#:~:text=If%20you%20need%20to%20execute,just%20use%20the%20shell%20command.&text=Invoke%20a%20standard%20shell%20to,determines%20which%20shell%20to%20run.

eZioPan, 2023-03-14:
Thanks for detailed reply!

Now I understand it is not related to `Cortex-Debug`. And I will try to report it to `mingw64` (maybe `msys2`) team.

I can confirm that on linux, `gdb` doesn't hold elf file lock at all. But on windows, `gdb-multiarch.exe` behavior more wired: 

After I load elf, before `target remote`, `gdb-multiarch` seems "soft" lock elf file, windows doesn't think any hold elf file lock, but rebuild stuck on rewrite elf file. And after I attach gdb to openocd, then windows does tell me that elf file is lock by another program, which turns out `gdb-multiarch`.

And since VSCode doesn't expose anything before building step, I think for now I will stick to re-run entire debug session when I need to rebuild.

Thanks again!

closed 2023-03-14 by eZioPan