Marus/cortex-debug #852: Can

Joebeazelman, 2023-03-20:


haneefdm, 2023-03-22:
Step commands only work with code that has source line info available -- even if the source file does not exist. However, you can do instruction-level steps instead of the default VSCode step which is code-level.

You can do this in the disassembly window managed by VSCode (we supply the instructions). See

https://github.com/Marus/cortex-debug/wiki/Disassembly-Debugging

If you have issues with the UI, then that is not us, please file issues with the VSCode team. We only supply the raw data.

haneefdm, 2023-03-22:
I wish VSCode had `stepi/nexti` built into their Debug Toolbar buttons. For instance, Shift+Button could mean `stepi` instead of `step`. Those buttons do not belong to us, so we can't change their behavior.

haneefdm, 2023-03-22:
I am closing this as it is not something we can fix/address

closed 2023-03-22 by haneefdm