Marus/cortex-debug #854: Let swoFrequency be optional parameter for SWO

fitch22, 2023-03-22:
Using OpenOCD and SWO, apparently the -pin-freq option, if omitted during configuration allows the adapter driver to determine what the maximum supported rate is and it gets set accordingly.  I think in many cases it would be desirable to let the adapter driver determine this.  I tried to leave out the swoFrequency parameter in my SWO configuration and unfortunately OpenOCD still gets passed the string "-pin-freq" but with no corresponding value, and that results in this error:

c:/Users/Bill/.vscode/extensions/marus25.cortex-debug-1.11.1/support/openocd-helpers.tcl: STM32G431KBTx.tpiu configure -protocol uart -output :50003 -traceclk 170000000 -pin-freq undefined

Is there any chance that the swoFrequency can be made optional?  And if it is omitted from launch.json that cortex-debug would only configure the traceclk but not the pin-freq?

By the way, when I get the above error, everything goes into debug mode, and the terminal creates an SWO output screen, but as you can imagine the SWO just does not work.  It might also be nice if this simply did not allow the debug session to start.

Bill

fitch22, 2023-03-23:
A bit more info:

I tried to customize openocd-helpers.tcl so that it would not configure the SWO frequency.  I did this by changing the line in CDSWOConfigure from this:
_$mytpiu configure -protocol uart -output $CDSWOOutput -traceclk $CDCPUFreqHz -pin-freq $CDSWOFreqHz_
to this:
_$mytpiu configure -protocol uart -output $CDSWOOutput -traceclk $CDCPUFreqHz_

but for some reason, openocd still gets the SWO frequency configured.

If I comment out the line in my lauch.json for '"swoFrequency": 4000000,', I then get a message:
_SWO/RTT Initialization failed: No symbol "undefined" in current context. (from interpreter-exec console "set $swoFreq = undefined")_

Surprisingly, after all of this the SWO output still works.  So I guess this is a way to prevent the explicit configuration of SWO frequency.  Some of this behavior must be new in 1.11.1 because in previous versions the SWO output would not work.

Am I missing something in the customization of openocd-helpers.tcl?

Thanks,
Bill

haneefdm, 2023-03-23:
I am going to close this as there is nothing actionable here.

closed 2023-03-23 by haneefdm