Marus/cortex-debug #855: External STM32 GDB server issue

truongn1, 2023-03-28:
We can not use an external `STM32 GDB Server` as the launch command still using `monitor reset halt`.  Any idea to solve this problem? 

Config: 

> {
            "name": "Remote Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/Debug/debug.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "external",
            "gdbTarget": "ip:61234",
            "device": "STM32F777IIKx",
            "runToEntryPoint": "main",
            "preLaunchTask": "Debug", 
            "showDevDebugOutput": "both",
      }

GDB server 
`STMicroelectronics ST-LINK GDB server. Version 7.1.0`
started with `--halt` option. 

I can replace all the `monitor reset halt` commands by the `monitor reset` in the `launchCommands` and it worked, just wonder are there any way to do this by configuration.

haneefdm, 2023-03-28:
We have no idea what external servers do and how they do it. We use openocd as a template. You can use the `override` commands to completely customize how to configure your gdb-server.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

truongn1, 2023-03-29:
Thanks @haneefdm 

For those who got this issue, it has been mentioned here: [#688 ](https://github.com/Marus/cortex-debug/issues/688). Just need to customize your `overrideLaunchCommands`