Marus/cortex-debug #860: flash side binary before attach

imyaFamiliev, 2023-03-31:
Hello!
I have a project with bootloader and main firmware for stm mcu
After  firmware compile there are script that concatenate boot and firmware and add some addition information to it.
So when i want to debug main firmware it consist of two stages
 1 flashing  my target with these complex image using task in makefile or launch j-link flasher by hand
 2  attach to device with cortex debug
I cant debug in "launch" mode  because  cortex-debug try to flash file that located in "executable" attribute of launch.json 
If i try to set my complex  image  in this "executable" attribute but   nothing happens.

So, if there a way to combine these two operations on one cortex-debug task ?
i see "preAttachCommands" Attribute for launch.json   but it for GDB, not for shell

haneefdm, 2023-03-31:
GDB (and us) need to know which image (in elf format) you want to actually debug regardless of what the whole image looks like.

> i see "preAttachCommands" Attribute for launch.json but it for GDB, not for shell

There is no shell. We only talk to gdb and nothing else. We do launch a gdb-server when appropriate but we talk to it via gdb as well.

haneefdm, 2023-04-01:
Why aren't you using `preLaunchTask`. This is a Microsoft command for any kind of debugger. You should also read VSCode/Microsoft documentation. You have to setup tasks in tasks.json and we have nothing to do with it, btw

closed 2023-04-01 by haneefdm