Marus/cortex-debug #861: LiveWatch causes cortex-m4 processors to not start after loading with jlink and GDB server

uhrynste, 2023-03-31:
Tested on STM32F437VG and STM32F427ZG.

If i enable liveWatch on these processors without "runToEntryPoint" the device won't start after loading firmware.
Works fine if I attach to the processor. And also sometimes it does work, rarely though.

Tested some GD32F207 processors and it works every time. 

No breakpoints are set and no variables are in any watches.

Configuration:
`"configurations": [
        {
            "name": "load",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/project/.build/release/test.elf",
            "request": "launch",
            "type":"cortex-debug",
            "servertype":"jlink",
            "liveWatch":{
                "enabled": true,
                "samplesPerSecond": 1
            },
            "device": "STM32F437VG",
            "interface": "jtag",
            "showDevDebugOutput": "raw",
            "svdFile":"${workspaceRoot}/project/Library/STM_Lib/STM32F437x.svd",
            "preLaunchCommands": [
                "set mem inaccessible-by-default off",
                "set verbose on"
            ],
            "runToEntryPoint": ""
        },
        {
            "name": "attach",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/project/.build/release/test.elf",
            "request": "attach",
            "type":"cortex-debug",
            "servertype":"jlink",
            "liveWatch":{
                "enabled": true,
                "samplesPerSecond": 4
            },
            "device": "STM32F437VG",
            "interface": "jtag",
            //"showDevDebugOutput": "raw",
            "svdFile":"${workspaceRoot}/project/Library/STM_Lib/STM32F437x.svd"
        },
    ]`

haneefdm, 2023-03-31:
> "interface": "jtag"

This may be a problem. Most JTAG implementations do not allow bus snooping. AFAIK. Can you switch to SWD?

haneefdm, 2023-03-31:
LiveWatch: I can't tell why it works intermittently. I also found that with JLink. Every once in a while it fails and we cannot point as we can't reproduce it. It happened to be 2-3 times in the hundreds of times I tried. Never had issues with OpenOCD though,

I think we need to separate issues about live watch vs. runToEntryPoint vs, other stuff.

uhrynste, 2023-03-31:
I can try using SWD instead of JTAG(on monday). To expand a bit on the behaviour: liveGDB still updates but GDB server doesn't respond to commands when it happens. I can quit debugging and then attach and that seems to unstuck the execution.

The part of the code where attach breaks at the beginning is fine though. My last idea is that it might be stuck in loader somewhere. I'll try that.

Hermann-Core, 2023-04-01:
Hello. By the way what is live watch please ? Is this a new feature ?

haneefdm, 2023-04-01:
Yes. You have to enable/install the pre-release version to see liveWatch in action. You can do this from within VSCode in the extension manager.

https://github.com/Marus/cortex-debug/issues/810

uhrynste, 2023-04-03:
Okay i have tried it with SWD and it has identical behaviour.

haneefdm, 2023-04-03:
> Okay i have tried it with SWD and it has identical behaviour.

SWO has nothing to do with liveWatch. totally unrelated. Please submit a separate issue with all the details.

uhrynste, 2023-04-04:
After a lot of debugging it's probably not an issue with cortex-debug, it's probably an issue with cmake and corrupted instructions, which causes the debugging and the device to crash. So i'm sorry for wasting your time. I still need to figure out what's wrong with the compilation, as i have the same flags and parameters in keil and in cmake, but keil produces functioning executable while cmake does not. This issue can be closed.

uhrynste, 2023-05-19:
I'm back. The issue remains even after fixing the compilation and it only occurs in Launch configuration with liveWatch. Attach works fine. Debug console log is attached.

Configuration:
`"name": "JLink debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/ICTGD.elf",
            "request": "launch",
            "type":"cortex-debug",
            "servertype":"jlink",
            "liveWatch":{
                "enabled": true,
                "samplesPerSecond": 4
            },
            "device": "GD32F207VG",
            "interface": "jtag",
            "showDevDebugOutput": "raw",
            "svdFile":"${workspaceRoot}/GD32F20x_CL.svd",
            "preLaunchTask": ""`

[log.txt](https://github.com/Marus/cortex-debug/files/11516953/log.txt)

haneefdm, 2023-05-19:
I don't know how to debug this. I have seen issues myself and no way to reliably reproduce it. See 

https://github.com/Marus/cortex-debug/issues/810#issuecomment-1407742188

Maybe we are starting live watch too soon but it still should not happen.

haneefdm, 2023-05-19:
Does it always happen or is it intermittent?

haneefdm, 2023-05-19:
I don't think this is supposed to happen. When a second instance of gdb connects to JLink, it is halting the processor and only notifies one GDB, so now, they are out of sync.

Can you try adding the option [-nohalt](https://wiki.segger.com/J-Link_GDB_Server#-nohalt) to `serverArgs`?
```
LiveGDB: 12-target-select extended-remote localhost:50000
LiveGDB: -> =thread-group-started,id="i1",pid="42000"
LiveGDB: -> =thread-created,id="1",group-id="i1"
LiveGDB: -> ~"0x0801b052 in EI2C_Dly (dl=149) at D:/prace2/CMAKE_PORTING_VSCODE/ICTGD_port/Library/Commmon/EEPROM_I2C/eeprom_I2c.c:58\n"
LiveGDB: 0x0801b052 in EI2C_Dly (dl=149) at D:/prace2/CMAKE_PORTING_VSCODE/ICTGD_port/Library/Commmon/EEPROM_I2C/eeprom_I2c.c:58
LiveGDB: -> ~"58\t   while (--dl);\n"
LiveGDB: 58	   while (--dl);
LiveGDB: -> *stopped,frame={addr="0x0801b052",func="EI2C_Dly",args=[{name="dl",value="149"}],file="D:/prace2/CMAKE_PORTING_VSCODE/ICTGD_port/Library/Commmon/EEPROM_I2C/eeprom_I2c.c",fullname="D:\\prace2\\CMAKE_PORTING_VSCODE\\ICTGD_port\\Library\\Commmon\\EEPROM_I2C\\eeprom_I2c.c",line="58",arch="armv7"},thread-id="1",stopped-threads="all"
LiveGDB: mi2.status = stopped
LiveGDB: Program stopped, probably due to a reset and/or halt issued by debugger
LiveGDB: -> 12^connected
Started live-monitor-gdb session
```

haneefdm, 2023-05-19:
Unless I can duplicate it, I cannot fix it. This is something JLink users have to live with. I don't think we are doing something wrong. Or we can remove JLink support for LiveWatch.

haneefdm, 2023-05-22:
Can you please try out the following pre-release and give feedback

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre2

uhrynste, 2023-05-23:
Okay tested it. Attach still works perfectly. Launch hasn't worked once. Still the same configuration, it doesn't work at all now. Nothing changes with or without the -nohalt option.

[log_after_update2.txt](https://github.com/Marus/cortex-debug/files/11540622/log_after_update2.txt)
[log_after_update3.txt](https://github.com/Marus/cortex-debug/files/11540623/log_after_update3.txt)
[log_not_responding.txt](https://github.com/Marus/cortex-debug/files/11540624/log_not_responding.txt)
[log_after_update1.txt](https://github.com/Marus/cortex-debug/files/11540625/log_after_update1.txt)

haneefdm, 2023-06-06:
Lets give this another try...

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre5

This is a pre-release

uhrynste, 2023-06-07:
Tested. Livewatch in attach configration doesnt work now, and neither in launch configuration. But some good news are that the device doesn't get completely stuck now. it responds to resets and halts. The output data are a bit confusing though, especially for the attach configuration.

output:
[pre5-3-load-terminal-resetting-stopping.txt](https://github.com/Marus/cortex-debug/files/11675620/pre5-3-load-terminal-resetting-stopping.txt)
[pre5-1-load.txt](https://github.com/Marus/cortex-debug/files/11675621/pre5-1-load.txt)
[pre5-2-attach.txt](https://github.com/Marus/cortex-debug/files/11675622/pre5-2-attach.txt)
[pre5-3-load-resetting-stopping.txt](https://github.com/Marus/cortex-debug/files/11675623/pre5-3-load-resetting-stopping.txt)
[pre5-1-load-terminal.txt](https://github.com/Marus/cortex-debug/files/11675624/pre5-1-load-terminal.txt)
[pre5-2-attach-terminal.txt](https://github.com/Marus/cortex-debug/files/11675625/pre5-2-attach-terminal.txt)

haneefdm, 2023-06-07:
I have no clue what is going on and it works fine for me with ST devices and JLink. Both attach and launch.

The logs you attached are not matching either. For attach, the output from debug console is grossly incomplete. I see none of the commands sent to gdb. Just tiny bit of gdb output. The program stopped three times at the same location for some unknown reason.

The only differences I see is that you are using a different device and using JTAG interface vs SWD. I don't have the capability to test with JTAG due to my boards.

Also, the launch logs actually shows that liveWatch is working.

Next time, we can try with `"showDevDebugOutput": "vscode"`. It will be much more verbose but may give a bit more insight.

uhrynste, 2023-06-07:
Well, you are right. I just tested it on ST device with swd and it works well. No idea what is wrong with the GD devices, will have to test it further. This solves a lot of issues, thank you.

haneefdm, 2023-06-07:
I am going to close this. I don't think we are doing anything wrong. People have all the debug logs and our source code. Welcome for any help if you have the actual device...

closed 2023-06-07 by haneefdm