Marus/cortex-debug #862: Persistant values in the terminal using RTT

Hermann-Core, 2023-04-01:
I use cortex debug with SEGGER's RTT to display some information on the terminal and it works well but there is a slight problem. even when we end the debugging session, the content of the previous display persists in the new terminal. how to solve this please? I would like the terminal to be cleared before each debugging session
`Config : `
```
"configurations": [
    {
        "name": "Application",
        "request": "launch",
        "type": "cortex-debug",
        "cwd": "${workspaceFolder}",
        "executable": "${workspaceFolder}/bin/debug/main_app.out",
        "servertype": "jlink",
        "serverpath": "${env:ProgramFiles}/SEGGER/JLink/JLinkGDBServerCL.exe",
        "runToEntryPoint": "__Setup_app",
        "device": "STM32F303VC",
        "showDevDebugOutput": "raw",
        "rttConfig": {
          "enabled": true,
          "address": "auto",
          "decoders": [
              {
                  "label": "",
                  "port": 0,
                  "type": "console",
                  "iencoding": "utf8",
              }
          ]
        },
        "preLaunchTask": "CMake: build",
        "svdFile": "${workspaceFolder}/device/SVD_file/STM32F303.svd"
      },
```

haneefdm, 2023-04-01:
That behavior is consistent and intentional in all such windows. Take Debug Console for example. Also, the gdb-server. People want to look at what happened with the last session. And, please don't ask for adding a feature for this.

The content is cleared with the start of a new session if that RTT window is needed and if this is not happening, let us know.

Hermann-Core, 2023-04-01:
> That behavior is consistent and intentional in all such windows. Take Debug Console for example. Also, the gdb-server. People want to look at what happened with the last session. And, please don't [...]

Hermann-Core, 2023-04-01:
The content of the terminal is not cleaned when I open a new debug session.

haneefdm, 2023-04-01:
New debug session that uses the RTT console?

haneefdm, 2023-04-01:
Can you attach some screenshots. Btw, Unlike SWO the RTT output from your previous run gets buffered and when you re-open a new session a simple reset does not clear the SRAM. So, you can have some remnants in the chip. This is a common issue with RRT.

When you power cycle, the SRAM gets destroyed but not when just doing a reset.

haneefdm, 2023-04-01:
At the end of a debug session, you will a message like
```
RTT connection on TCP port XXXXXXX ended. Waiting for next connection...
```
When a new session starts the entire window is cleared including the message above. Are you seeing that message persist on a new session? If not then the window is being cleared.

But if JLink is sending data on a new connection, we have no idea if it is old or new. We do try to clear the previous RTT info but JLink works differently than openocd. We do that just to be nice and is not a requirement.

Are you draining the RTT buffer(s) before ending the session. If they are in non-blocking mode, then they may not drain at all the way you want it to especially if you end the session without stopping first.

Hermann-Core, 2023-04-01:
> New debug session that uses the RTT console?

yes exactly

Hermann-Core, 2023-04-01:
> Can you attach some screenshots. Btw, Unlike SWO the RTT output from your previous run gets buffered and when you re-open a new session a simple reset does not clear the SRAM. So, you can have some remnants in the chip. This is a common issue with RTT.
> 
> When you power cycle, the SRAM gets destroyed but not when just doing a reset.

the SRAM ? how can it access to the content of my SRAM to prints infos without any call of RTT_Printf and stuffs ?

Hermann-Core, 2023-04-01:
> When a new session starts the entire window is cleared including the message above. Are you seeing that message persist on a new session? If not then the window is being cleared.
I dont think but rather my last strings in the terminal

Hermann-Core, 2023-04-01:
> Are you draining the RTT buffer(s) before ending the session. If they are in non-blocking mode, then they may not drain at all the way you want it to especially if you end the session without stopping first.

Hm tell me more please. how I can drain the RTT buffers ?8 And how can I stop a session please ?

haneefdm, 2023-04-01:
> the SRAM ? how can it access to the content of my SRAM to prints infos without any call of RTT_Printf and stuffs ?
> I dont think but rather my last strings in the terminal

Contrary to popular believe, RTT is not realtime. JLink/OpenOCD poll your SRAM if something new was written once every 100ms or so. What you are seeing may be the remnant of your last session still in RAM.

Trust us on this. We don't remember what was printed to even preserve some your last output. We don't even know how to do that. The entire window is cleared. In fact, the old process exists and a new process starts so there is no carryover. It is JLink who is sending us that data because it found it in your SRAM. It doesn't make stuff up either.  There may be a reset option for JLink that also clears your SRAM (like hard reset). Check with SEGGER and your device manufacturer.

I recommend reading up on how RTT works. Lots of good info on the SEGGER website and also study the FW. It is not that much.

> Hm tell me more please. how I can drain the RTT buffers ?8 And how can I stop a session please ?

If you just first hit the pause button, then it should drain the buffers. But if you disconnect, then your program may continue to run (depends on if you Stop/Terminate vs Disconnect/Detach) and keep writing to the SRAM (which is where your buffers live). The red square button to stop/detach the session has two modes if you hover on it. You will know which is which.

But if you Stop/Disconnect without first pausing, then yes there is a possibility of some stuff not being drained.

You may want to output something at the beginning of your main to know where your really output starts.

Btw, you can also use the free SEGGER tools while using VSCode for your debugging. Pass the right command-line options and start a telnet/netcat/putty client yourself.

Hermann-Core, 2023-04-02:
Thanks for your explanation. When I use JlinkeRTT Viewer for debugging while using the one with Vs Code they are not synchronous. Sometime the output go to the segger viewer and sometime to Vs Code never both at the same time

haneefdm, 2023-04-02:
There are two ways.
1. You can let us enable RTT and you can use putty/nctcat to view the info. Just don't specify any decoders. This should work.
```
        "rttConfig": {
          "enabled": true,
          "address": "auto"
          ]
        },
```
2. You let JLink RTT Viewer do it all. Remove/disable "rttConfig"

> Also I dont know what happens here but when I try to step quickly the debug session crashes 

What crashes? gdb? jlinkgdbserver or session just ends. Please tell us details in a separate Issue with debug enabled and the contents of the Debug Console.

closed 2023-04-02 by haneefdm

Hermann-Core, 2023-04-03:
I think I will choose the first option.

> What crashes? gdb? jlinkgdbserver or session just ends.

I think the server
```
Performing single step...
...Breakpoint reached @ address 0x0800103E
Reading common registers: R0, R1, R2, R3, R4, R5, R6, R7, R8, R9, R10, R11, R12, SP, LR, PC, XPSR
Performing single step...
...Breakpoint reached @ address 0x08001040
Reading common registers: R0, R1, R2, R3, R4, R5, R6, R7, R8, R9, R10, R11, R12, SP, LR, PC, XPSR
Performing single step...
...Breakpoint reached @ address 0x080011B2
Reading common registers: R0, R1, R2, R3, R4, R5, R6, R7, R8, R9, R10, R11, R12, SP, LR, PC, XPSR
Read 4 bytes @ address 0x080011B2 (Data = 0xB084B580)
Read 4 bytes @ address 0x08001044 (Data = 0xF2419801)
Reading 64 bytes @ address 0x20001140
Read 2 bytes @ address 0x08001044 (Data = 0x9801)
Setting breakpoint @ address 0x08001044, Kind = 2, Type = THUMB, BPHandle = 0x001E
Starting target CPU...
...Breakpoint reached @ address 0x08001044
Reading common registers: R0, R1, R2, R3, R4, R5, R6, R7, R8, R9, R10, R11, R12, SP, LR, PC, XPSR
Read 4 bytes @ address 0x08001044 (Data = 0xF2419801)
Reading 64 bytes @ address 0x20001140
Removing breakpoint @ address 0x08001044, Size = 2
Removing breakpoint @ address 0x10000020, Size = 2
Read 4 bytes @ address 0x08001044 (Data = 0xF2419801)
Setting breakpoint @ address 0x10000020, Kind = 2, Type = THUMB, BPHandle = 0x001F
Performing single step...
[2023-04-03T00:12:35.354Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET
[2023-04-03T00:12:35.356Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

Hermann-Core, 2023-04-03:
By the way Im trying to jump from one application to another but in different binaries files. When I jump to any function in the same app it works well but when I try to jump to an address defined to another address in the different app it go nowhere. But I can see that the address is correct and is the same as in the map file. tIs it because I flash the 2 apps independtly using JLink and cortex-debug or the content of the flash is erased before flashing even with a specified en adddress in the linker file ? But in that case why in the 1st I can see the correct address addresses of the vector table of the 2nd app ? with the corrects addresses inside ? I'm confused. I dont know what's wrong

haneefdm, 2023-04-03:
You have all the info in the Debug Console and the JLink gdb-server windows. Enable debug and you can see every operation. Since you are flashing twice and we are flashing once, I am not sure what actually got written to where You would know more than me. If you don't want us to flash, then you can do `"loadFiles": []`

Please look at the Debug Console for ALL the commands being issued to gdb. We don't do any magic without you knowing about it. We even let gdb program your device. We don't do it ourselves.