Marus/cortex-debug #865: BlackMagic Probe, can't pause connected target after run.

wmaxfield, 2023-04-03:
Congratulations on such a good extension.   However, there is an issue which may/may not be your issue.

I created a BlackMagic probe and tested it on an STM32F103T6 clone.  I can load a program and run.  When it hits the 
'main' breakpoint, then it pauses, and I can set another breakpoint, run, and it will break. I can see variables, single step, step into and out of functions, etc.  With all debugger probes so far, pause execution works.

But, if I have BlackMagic run the target again without breakpoints in the execution path, pressing Pause does nothing.  I can quit the session, restart it and reload and restart the processor, and set a new breakpoint and have it hit that breakpoint.  If I connect to the GDB Server on the serial port via a terminal program, I can stop it; so it could come down to the right commands.  I don't know GDB command, so I really don't know how to go from that point on.  This may be a BlackMagic probe problem, I am just letting you know.

Thanks for such a good product!  I have used it minimally JLink, STLink.  I use it almost every day with OpenOCD (psoc5lp DAPLink).  It works very well in that configuration.

haneefdm, 2023-04-04:
Unfortunately, I don't have a blackmagic probe. Can you enable Debug (`"showDevDebugOutput": "raw"`) and attach the output of the `Debug Console`? This way we can see ALL the commands being sent to gdb. We never talk to the gdb-server directly.

A second run with `"showDevDebugOutput": "vscode"` is also helpful. This may tell us why the Pause button is not sending the command to our debugger ... if that is the case.

May I ask why you prefer BlackMagic Probe?

wmaxfield, 2023-04-04:
It appears that when I click on the "pause" button, nothing is sent at all, from observation.  The button appears to be ignored.  I created my BlackMagic probe from a pair of cheap stlink/blue-pill boards from amazon.  I documented that on my blog (see following)

Answer to your question:
I do not prefer BlackMagic Probe.  I have a blog ( socmaker . com ) where I discuss PSOC family, and other things.  I was trying to work with a Quicklogic QuickFeather and bricked it.  That lead me on a chase for cheap debug probes that could possibly work and I stumbled across several options.  I saw where your extension indicated support for BlackMagic, and tested it, just to see.  That is where I ran across this issue.

I prefer whatever probe will work, but for my readers, I wanted to point them to the least expensive probes possible.  

BlackMagic is just one of them, but is contractually ok for many processors.  JLink-OB's (On-Board) are contractually tied to a processor family, as are STLink OB's.  J-Links that are not tied are over 500usd.

BlackMagic was a possibility of one not contractually tied to any processor family.  That is the only reason for looking at it.

Thanks!

Output:

Note: the showDevDebugOutput is not shown in the intellisense suggestions:

showDevDebugOutput:raw

This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
=cmd-param-changed,param="pagination",value="off"
0x08000362 in ?? ()
1: (859) <-1015-thread-info
1: (860) ->1015^done,threads=[{id="1",target-id="Thread 1",frame={level="0",addr="0x08000c94",func="Reset_Handler",args=[],file="startup_stm32f103x6.s",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/startup_stm32f103x6.s",line="65",arch="armv7"},state="stopped"}],current-thread-id="1"
1: (860) ->(gdb)
1: (861) 1015: elapsed time 1
1: (868) <-1016-stack-list-frames 0 1000
1: (868) ->1016^done,stack=[frame={level="0",addr="0x08000c94",func="Reset_Handler",file="startup_stm32f103x6.s",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/startup_stm32f103x6.s",line="65",arch="armv7"}]
1: (868) ->(gdb)
1: (870) 1016: elapsed time 1
1: (872) Send Event AD7ProcessInfoUpdatedEvent
1: (878) Send Event AD7ThreadCreateEvent
1: (885) <--exec-continue
1: (885) ->^running
1: (885) ->*running,thread-id="all"
1: (887) ->(gdb)
1: (889) ->=breakpoint-modified,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x0800022e",func="main",file="Core/Src/main.c",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/Core/Src/main.c",line="74",thread-groups=["i1"],times="1",original-location="main"}
1: (892) ->~"\n"
1: (892) ->~"Breakpoint 1, main () at Core/Src/main.c:74\n"
1: (892) ->~"74\t  HAL_Init();\n"

Breakpoint 1, main () at Core/Src/main.c:74
1: (892) ->*stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x0800022e",func="main",args=[],file="Core/Src/main.c",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/Core/Src/main.c",line="74",arch="armv7"},thread-id="1",stopped-threads="all"
74	  HAL_Init();
1: (892) <-1017-stack-list-frames 0 1000
1: (893) ->1017^done,stack=[frame={level="0",addr="0x0800022e",func="main",file="Core/Src/main.c",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/Core/Src/main.c",line="74",arch="armv7"}]
1: (893) ->(gdb)
1: (893) 1017: elapsed time 0
1: (896) Send Event AD7ExceptionEvent
Execute debugger commands using "-exec <command>", for example "-exec info registers" will list registers in use (when GDB is the debugger)
1: (952) <-1018-stack-list-arguments 0 0 0
1: (952) ->1018^done,stack-args=[frame={level="0",args=[]}]
1: (952) ->(gdb)
1: (953) 1018: elapsed time 1
1: (1131) <-1019-var-create - * "counter"
1: (1132) ->1019^done,name="var1",numchild="0",value="0",type="int16_t",has_more="0"
1: (1132) ->(gdb)
1: (1132) 1019: elapsed time 1
1: (1417) <-1020-stack-list-variables 0
[... 162 lines cut ...]
1: (867) ->~"74\t  HAL_Init();\n"
74	  HAL_Init();
1: (867) ->*stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x0800022e",func="main",args=[],file="Core/Src/main.c",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/Core/Src/main.c",line="74",arch="armv7"},thread-id="1",stopped-threads="all"
1: (867) <-1017-stack-list-frames 0 1000
1: (867) ->1017^done,stack=[frame={level="0",addr="0x0800022e",func="main",file="Core/Src/main.c",fullname="/Users/maxfield/Desktop/STM32_Projects/blink2/Core/Src/main.c",line="74",arch="armv7"}]
1: (867) ->(gdb)
1: (867) 1017: elapsed time 0
1: (871) Send Event AD7ExceptionEvent
Execute debugger commands using "-exec <command>", for example "-exec info registers" will list registers in use (when GDB is the debugger)
1: (928) <-1018-stack-list-arguments 0 0 0
1: (929) ->1018^done,stack-args=[frame={level="0",args=[]}]
1: (929) ->(gdb)
1: (929) 1018: elapsed time 1
1: (1049) <-1019-var-create - * "counter"
1: (1050) ->1019^done,name="var1",numchild="0",value="0",type="int16_t",has_more="0"
1: (1050) ->(gdb)
1: (1050) 1019: elapsed time 0
1: (1403) <-1020-stack-list-variables 0
1: (1403) ->1020^done,variables=[]
1: (1403) ->(gdb)
1: (1404) 1020: elapsed time 1
1: (4330) <--exec-continue
1: (4331) ->^running
1: (4331) ->*running,thread-id="all"
1: (4331) ->(gdb)

haneefdm, 2023-04-04:
> Note: the showDevDebugOutput is not shown in the intellisense suggestions:

Not sure why. The best way to know the left over attributes is to just type two double quotes and the Intellisense will pop up but you may have to scroll. I am not sure why whey are not sorted.

[image]

Everything is also documented in https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

haneefdm, 2023-04-04:
Could we try this. While we are in a state that the pause button has no effect, enter the following in the Debug Console
```
-exec-interrupt
```
It should pause the debugger almost immediately. I want to know this part actually works. I dont know why VSCode is not sending us a request to actually pause - perhaps it thinks that the session hasn't even started yet.

haneefdm, 2023-04-04:
Oh wait. You are not using our debugger at all. You are using Microsofts debugger. Our debugger type is `cortex-debug`. You probably have `cppdbg`. No wonder II was all confused. Our setup is TOTALLY different.

https://github.com/Marus/cortex-debug/wiki/Black-Magic-Probe-Specific-Configuration

wmaxfield, 2023-04-04:
I APOLOGIZE!!!!!

I TRULY thought I was using BMP!!!    I don't understand the launch.json and the script I was given is wrong.  I replaced it with yours AND IT WORKS!

I apologize Again. I  WAS WRONG!

Yours works well!

haneefdm, 2023-04-04:
You may be using BMP more than likely but you were using microsoft's debugger. Not ours.

haneefdm, 2023-04-04:
I am closing this as there is not much more to do..

Btw, I had a long relationship with PSoC5 (I was one of the architects).

closed 2023-04-04 by haneefdm