Marus/cortex-debug #867: Enable Users to disable the '--shared' option for stlink-gdbservers

Delzye, 2023-04-06:
**Is your feature request related to a problem? Please describe.**
For some reason, the gdbserver from stlink is unable to enumerate my st-link device when the --share option is enabled.
I have not found a reason for this behavior or a way to fix it.

**Describe the solution you'd like**
Add an option in launch.json to disable sharing the gdbserver

**Describe alternatives you've considered**
Switching the type to external, setting the port to 50000 and manually starting the server. That works, but seems unnecessarily complicated, as I don't need to share the server in the first place.

**Additional context**
-

haneefdm, 2023-04-06:
I believe you, but are you sure about this? That `--shared` would be the problem. In fact it should be the other way when you have exclusive access. Just wondering what the real cause is. What OS are you having trouble on?

* Have you tried to run the stlink gdb server with just a `-q` option to see if there is something else connected and if it able to enumerate? See https://www.st.com/resource/en/user_manual/um2576-stm32cubeide-stlink-gdb-server-stmicroelectronics.pdf
* Have you contacted ST to see why this would be happening?

History:
We added the `--shared` because that is the only way multi-core debug works and it did not make a difference in a single-core scenario. When multi-core fails, most users would not know why. We also don't/can't know that multiple cores are being (or about to be) debugged at the same time so we can't automatically add `--shared`.

haneefdm, 2023-04-06:
Btw, while I consider adding an option, you can also use a wrapper script that strips the option and using that script as the `serverpath`. I know, it is not simple especially on Windows to create such a script.

Delzye, 2023-04-06:
Jup, I'm sure. If I run the exact command the extension runs, I get the same error. If I remove the shared option and leave everything else, it works.

The OS is Windows 10.
Just the -q Option does not work, I need -cp aswell. Then my stlink shows up. If I add the shared option now (don't know if that makes any sense in combination with -q), my stlink does not show up.
I have not contacted ST yet.

I know it makes sense to enable the option, and I would be fine with it being on by default (since it should work with single-core aswell) but having the option to turn it off would be nice.

Delzye, 2023-04-06:
The problem could be that I did not install the tools, but instead copied them from an existing install (I need it to be portable), but I know that copying the tools from that exact install worked in the past, debugging and everything...

haneefdm, 2023-04-06:
A few things to try.

* Does the problem happen after a fresh reboot on the first run?
* After running once, if you look in the process viewer, is the stlink gdb server still running?
*  If you try multiple times, does the problem go away

My issue with adding a feature is that they are sticky and I am not sure what problem I am fixing. I have tons of users, and I would have heard about this a while ago.

To me, `--shared` means that there can be many many clients connecting and disconnecting and it should still work. This is the opposite of that.

Delzye, 2023-04-07:
Hi @haneefdm,
I will try your suggestions when I'm at work next Wednesday, thanks in advance!

Delzye, 2023-04-12:
Here are the results:
- It happens on first run
- The stlink gdb server is not running anymore, as far as I can see
- It does not go away by trying multiple times

schiltztool, 2023-05-22:
HI, I'm having the exact same issue on windows. Any time I add the --shared flag I get the error `Error: Unable to connect to the target device, Access port is not valid.` from STM32CubeProgrammer v2.13.0

I checked everything suggested, and have the same results as @Delzye
- Only my st-link shows up with the -q option
- The problem happens after the first start of the system.
- The stlink gdb server does not running
- The problem does not go away with multiple tries.

I've confirmed I am exclusively using the binaries from STM32CubeCLT and my ST-link firmware is up to date. I uninstalled everything, and only installed the [STM32CubeCLT](https://www.st.com/en/development-tools/stm32cubeclt.html?dl=UnueLgBTqMLXJRyvIRLctA%3D%3D%2Cod7BotphY0xZEsGgSdIT%2B1u84gkFGPEZbyO6ThoEYvA7PFTUZ%2BxWIfqpeaJb8x%2BL5CA2XlpKStfu2RNqkjMZfXVIqvNUjSRbUH0vob5FmTWnpcx6PrVMb%2BbYJ2xzVmZD2WP7TvOdyK5JgTgHo1fEctKSdZxWa28d%2BPU1EAtubr5ftFcXy8UQKf4vdaOcWSRIkYoo51vQvD1N8rYMkF0dYvyYu2TmUY%2FHhFlpz3lk40y%2FuS%2F4jqxl%2BbbMqVdvRVNfpKN%2FNly4fxNJhFokvRzVot4zr9Ljs%2BDPp7EpheqBVTDMBlukMX4pCQfEbd0BxaCN0X6pQC6cN31%2BSKSy3oi8mOoyDYH6WTSdDKVLoO8FtrHnNuZj5W7HZsgbIj6nsATQYjXt3%2F%2FfQPxin2Y3pWRcwHgFsGyAgNcRyNZff7peOnq26roAgii33No0PG706v%2Bq)

haneefdm, 2023-05-22:
@schiltztool  But isn't this an issue with the ST tools. We are using it as it is documented. I will try to reproduce this.

Also, STM32CubeCLT is not something I have tried. I always install STM32CubeIDE.

schiltztool, 2023-05-22:
Ok. I wasn't sure. I'll reach out to ST. 

I tried wrapping the ST-LINK program in a script to strip the variable off, but it's ballooned out to 70+ lines of powershell. The script works in my terminal, redirecting stdin, stdout, stderr, but it fails in the TERMINAL tab of vscode.
[image]

haneefdm, 2023-05-22:
Hang on, I am working on a patch..

haneefdm, 2023-05-22:
Can you please try out the following pre-release and give feedback

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre2

See if liveWatch still works...

closed 2023-05-23 by haneefdm

schiltztool, 2023-05-26:
Ok, So.
I had a meeting with ST and it is partially an issue with the system bootloader that is baked into the board during manufacturing on the alpha WAB52 chip.

They are also seeing the problem intermittently on their release boards, so they are reaching out to their bootloader and STM32CubeProgrammer teams

schiltztool, 2023-05-26:
Ok, So.
I had a meeting with ST and it is partially an issue with the system bootloader that is baked into the board during manufacturing on the alpha WAB52 chip.

They are also seeing the problem intermitantly on their release boards, so they are reaching out to their bootloader and STM32CubeProgrammer teams

haneefdm, 2023-05-26:
Thanks. All users benefit when things are reported to the right people.