Marus/cortex-debug #868: Source environment variables

JPHutchins, 2023-04-10:
**Is your feature request related to a problem? Please describe.**
I like to commit launch.json to repositories.  I would like to make it possible for users to set local environment variables that configure things like ARM_TOOLCHAIN_PATH.

**Describe the solution you'd like**
Add a property that allows executing a shell command before anything else.  Users would probably like to do something like:

```json
  "preLaunchShellCommands": [". env"],
 ```

**Describe alternatives you've considered**
What I have to do presently:
* source a script to set the environment variables
* launch vscode from that terminal, `code .`

Not sure I can get this to work with Remote SSH!  And it's not very elegant to be tied to launching vscode from a terminal.

**Additional context**
Add any other context or screenshots about the feature request here.

Thanks for this amazing work, it is indispensable and I hope that I can contribute in the future.

haneefdm, 2023-04-10:
VSCode interprets the entire launch.json and only gives it to us in its final expanded form ... and **only** the one configuration you launched. We don't have access to the whole launch.json or which launch.json (you can have multiple in a workspace). For instance, we never see the `"inputs"` section. Stuff like `"${workspaceRoot}/build/zephyr/zephyr.elf"` and `"${input:arm_path}"` is expanded by VSCode. See

https://code.visualstudio.com/docs/editor/debugging#_variable-substitution
https://code.visualstudio.com/docs/editor/variables-reference

There are many Unix-like processes involved that you could also look at to understand how everything works. We have to think about who the environment variable is for, vscode, this extension, gdb, gdb-server, etc.?

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

Also, why not create a script/shortcut that sets all the needed env. vars and then launch that script/shortcut?

Perhaps, this rightfully should be submitted as a request for VSCode itself.

haneefdm, 2023-04-10:
>> Note that some debuggers support this. For example, the node.js debugger has "env" and "envFile" options (envFile is the one I'm looking for, env would be useless to me): https://code.visualstudio.com/docs/nodejs/nodejs-debugging#_load-environment-variables-from-external-file

Those env vars are intended for the final target. Not for VSCode itself or any of the intermediate programs. For embedded that does not make sense and since we are gdb based, you can already do that with gdb commands/.gdbinit file.

Sorry, but I have to close this. As a team, we all set certain environment variables in our company or configure out PATH properly. This is what we do in our company as most env. variables are actually machine dependent as well.

closed 2023-04-10 by haneefdm

JPHutchins, 2023-04-10:
Sounds good, I will pursue with MS.  All of my projects support MacOS/Linux/Windows environments and I can provide instructions that VSCode must be launched from a terminal with the correct environment variables set in order to use the launch.json / cortex-debug.