Marus/cortex-debug #870: GDB not handling the MI commands: Undefined command: "1-gdb-version".  Try "help".

sammyj85, 2023-04-13:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
It could well be that I'm doing something wrong. But I can't see this has been an issue that has already been reported.

I am using the Arm GNU Toolchain from the README, but the latest 12.2 version.
I had to manually build Python 3.8, which I found to be the only way to have that gdb version run.

I had to set Cortex-debug settings:

    "cortex-debug.objdumpPath.linux": "arm-none-eabi-objdump",
    "cortex-debug.gdbPath.linux": "arm-none-eabi-gdb.sh"

Where `arm-none-eabi-gdb.sh` is just a mini script pointing PYTHONHOME to the Python 3.8 directory when it runs gdb.

**Cortex-debug is passing GDB an MI command, and GDB is saying the command is undefined.**

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Toolchain set up as specified
2. settings.json as specified
3. launch.json as specified
4. Start debug session
5. Debug console shows error

**Expected behavior**

GDB to launch and accept MI commands as it should.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension) 1.6.7
 - OS: Fedora Linux 37 x86_64
 - GDB Version: 12.1.90.20221210-git
 - Compiler Toolchain Version: Arm GNU Toolchain arm-none-eabi 12.2.Rel1

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [
    {
      "name": "Controller DEBUG",
      "device": "STM32F429",
      "cwd": "${workspaceFolder}",
      "executable": "./build/controller_nucleof429_hw1_debug.elf",
      "request": "launch",
      "runToEntryPoint": "main",
      "showDevDebugOutput": "raw",
      "type": "cortex-debug",
      "servertype": "openocd",
      "configFiles": [
        "${workspaceFolder}/scripts/stlink.cfg",
        "${workspaceFolder}/scripts/stm32f4x.cfg"
      ],
      "svdFile": "${workspaceFolder}/scripts/STM32F429.svd",
      "swoConfig": {
        "enabled": true,
        "cpuFrequency": 180000000, // The booloader and the application should run at 180MHz for this SWO to work.
        "swoFrequency": 4000000,
        "source": "probe",
        "decoders": [{ "type": "console", "label": "ITM", "port": 0 }]
      }
    }
  ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version 1.6.7 git(b0f5563). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w ~/work/controller/build/controller_nucleof429_hw1_debug.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p ~/work/controller/build/controller_nucleof429_hw1_debug.elf
Launching GDB: arm-none-eabi-gdb.sh -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50001" -c "tcl_port 50002" -c "telnet_port 50003" -s ~/work/controller -f ~/.vscode-oss/extensions/marus25.cortex-debug-1.6.7-universal/support/openocd-helpers.tcl -f ~/work/controller/scripts/stlink.cfg -f ~/work/controller/scripts/stm32f4x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 23 ms
Finished reading symbols from nm: Time: 40 ms
GNU gdb (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.1.90.20221210-git
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.linaro.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) (gdb)
Undefined command: "1-gdb-version".  Try "help".
Error: Unable to start GDB even after 5 seconds or it couldn't even start Make sure you can start gdb from the command-line and run any command like "echo hello".
    If you cannot, it is most likely because "libncurses5" or "python" is not installed. Some GDBs require these
Failed to generate gdb commands: Error: Could not start gdb, no response from gdb
Error: Could not start gdb, no response from gdb
    at Timeout._onTimeout (~/.vscode-oss/extensions/marus25.cortex-debug-1.6.7-universal/dist/debugadapter.js:2:135797)
    at listOnTimeout (node:internal/timers:559:17)
    at process.processTimers (node:internal/timers:502:7)
```

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-04-13:
I need to see your script. Did you pass all the arguments we pass to GDB if not, gdb will not be in machine interface mode. What we pass as gdb arguments can change depending on many things in launch.json. So, it is not always the same

Your final line should probably look like
```
exec <path-to-gdb> "$@"
```

The exec will do a nice job of inheriting stdin/stdout/stderr. All communication with GDB happens over these channels. And the "$@" should pass all the arguments your script received.

https://www.computerhope.com/unix/bash/exec.htm
https://stackoverflow.com/questions/3811345/how-to-pass-all-arguments-passed-to-my-bash-script-to-a-function-of-mine

Please close this if that fixes your issue.

sammyj85, 2023-04-13:
I feel silly - no wonder it wasn't a common problem (like the Python 3.8 issue)!

Thanks for that reminder - that was it. Closed.

closed 2023-04-13 by sammyj85

haneefdm, 2023-04-14:
At least, you knew enough to create a wrapper script :-) Most people blame us for an incomplete gdb setup.

sammyj85, 2023-04-14:
For completeness, the GDB - Python 3.8 manual build issue is detailed here:

https://community.arm.com/support-forums/f/compilers-and-libraries-forum/53996/arm-gnu-toolchain-12-2-rel1-x86_64-arm-none-eabi-gdb-fails-on-ubuntu-22-04/180229#180229