Marus/cortex-debug #872: Does coretex-debug support armclang?

DavidA2014, 2023-04-14:
Hi, Does coretex-debug support the armclang toolset, or does it only support the gnu arm toolchain?

Just a suggestion: could you enable 'Discussions' for this project so that Q&A could be separated from Issues?

haneefdm, 2023-04-14:
You can use any compiler you want. I know we have had IAR users here. Ultimately, they all produce an elf file conforming to the ARM specification -- that can be debugged by GDB.

So, you only need some tools gdb/nm/objdump from the ARM GNU toolchain to debug 

> Just a suggestion: could you enable 'Discussions' for this project so that Q&A could be separated from Issues?

Good idea. Not sure I can do that. I don't own the repo but I will look into it. I maintain this project.

Eplankton, 2023-04-16:
> Hi, Does coretex-debug support the armclang toolset, or does it only support the gnu arm toolchain?
> 
> Just a suggestion: could you enable 'Discussions' for this project so that Q&A could be separated from Issues?

For armclang (or armcc) toolchain, you can choose to use both `axf` and `elf` binary.

haneefdm, 2023-04-16:
Both are ELF formats I believe. You can use either. I don't think there is a difference. If you are in Unix, you can do `file blah.axf` and it will tell you what the format actually is.

closed 2023-04-16 by haneefdm