Marus/cortex-debug #873: Global hex/dec mode

Eplankton, 2023-04-16:
It's said in the change log that you can choose hex/dec mode in every individual expression, but I can only change them globally with the `n/0x` button, for example I'd like to red part show as hex and yellow part show as dec, what can i do?
[image]

Eplankton, 2023-04-16:
I know that i can use something like `tmp, x` to show variable in hex mode within `LiveWatch` part, but how can I control the display mode up there in `Watch` part?

haneefdm, 2023-04-16:
Isn't there a button on the top right of the Titel Bar for the Panel? That setting is global. Use can also do using the  Cortex-Debug Settings in VSCode. I hope

You will see those buttons only when your mouse is on the title bar. That is how VSCode does it.

haneefdm, 2023-04-16:
[image]

See the button on the top-right. If this doesn't work let us know. Closing this.

closed 2023-04-16 by haneefdm

Eplankton, 2023-04-16:
> [image]
> 
> See the button on the top-right. If this doesn't work let us know. Closing this.

I believe that you have misunderstood my problem exactly the opposite way, I'd like to have a non-global way to set certain entry as hex/dec mode in the watch windows, which won't disturb the mode selected in other space.

haneefdm, 2023-04-16:
Yes, we had it that way for a while. Got too many complaints and then did it this way.

It is a lot of work to go back and our code base has moved on. And, we don't own the Variables and Watch windows they belong to VSCode and changes over there are difficult to impossible.

It is going to stay this way for a while until I get sufficient opposite feedback.

haneefdm, 2023-04-16:
And, sorry for misreading your original request.

PhilippHaefele, 2023-04-16:
> I believe that you have misunderstood my problem exactly the opposite way, I'd like to have a non-global way to set certain entry as hex/dec mode in the watch windows, which won't disturb the mode [...]

@Eplankton If i do remember right (haven't tested it recently), the exact same thing (like shown in the screenshot above) from LiveWatch is also working in Watch window.

Was that your request?
In Variables window this is not possible and only global switching is possible.