Marus/cortex-debug #874: Could not launch any debug session with last versions

ANauzet, 2023-04-27:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
Starting the Debug session. The small "popup" with the shortcut to RESET/STOP/CONTINUE... display for few seconds, and than disappear. And nothing else happens.

Bug is seen with versions from 1.9.3 to 1.11.2 (all version still available on VSC Marketplace)
Bug is NOT seen with version 1.6.3

**To Reproduce**
Steps to reproduce the behavior:
1. Start the debug session
2. Seems to crash after few seconds, but no error displayed on Debug Console

**Expected behavior**

Should start like when using version 1.6.3 or should display a configuration error to correct behavior.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension) 1.9.3 to 1.11.2
 - OS: Windows 10
 - GDB Version: GNU gdb (GNU Tools for STM32 10.3-2021.10.20211105-1100)
 - Compiler Toolchain Version: arm-none-eabi V10.3.1

**Please include `launch.json`**
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "executable": "./build/STM32F407.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32F407",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f4x.cfg"
            ],
            "gdbPath": "arm-none-eabi-gdb.exe",
            "showDevDebugOutput": "raw"
        }
    ]
}

**Attach text from `Debug Console`**
Cortex-Debug: VSCode debugger extension version undefined git(014039f). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump.exe --syms -C -h -w C:\Work\Projets\20220405_STNLV2_SentinelleV2\05_MainBoard_v1.0\build\05_MainBoard_v1.0.elf
Reading symbols from arm-none-eabi-nm.exe --defined-only -S -l -C -p C:\Work\Projets\20220405_STNLV2_SentinelleV2\05_MainBoard_v1.0\build\05_MainBoard_v1.0.elf
Launching GDB: arm-none-eabi-gdb.exe -q --interpreter=mi2
1-gdb-version

And that is all :-(

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-04-27:
Sounds like your gdb installation is incomplete/bad. Can you run this command manually in a terminal/cmd window?

```
arm-none-eabi-gdb.exe -q --interpreter=mi2
```
Just like how we are attempting to do.

ANauzet, 2023-04-28:
Hello,

I'm using the gdb installed by STM32CubeIDE.

```
arm-none-eabi-gdb.exe -q --interpreter=mi2
=thread-group-added,id="i1"
(gdb)

&"\n"
^done
```
Each "return" display 
```
&"\n"
^done
```
And
```
arm-none-eabi-gdb.exe -q
(gdb)
```
It just launch gdb prompt with no other display

haneefdm, 2023-04-28:
> Cortex-Debug: VSCode debugger extension version undefined git(https://github.com/Marus/cortex-debug/commit/014039f7cabdd1dfd8abb06d8e55b4bef537d8d2). Usage info: https://github.com/Marus/cortex-debug#usage

Above line shows no version# That is strange. It should look like this

> Cortex-Debug: VSCode debugger extension version 1.11.2 git(441af06). Usage info: https://github.com/Marus/cortex-debug#usage

Also, what do you see in the TERMINAL tab for gdb-server? It is not clear to me which openocd it is and what happened. Doesn't even look like it got launched or something horrible happened while launching.

ANauzet, 2023-05-09:
Hello again. Sorry is was a few days off.

So, TERMINAL do not display anything when launching the debug.

For information, when launching with the 1.6.3 version, here is the `terminal` display:
```
Waiting for gdb server to start...[2023-05-09T13:03:01.316Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Work\\Projets\\Prj\\Main" -f "c:/Users/T467/.vscode/extensions/marus25.cortex-debug-1.6.3/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32f4x.cfg
Open On-Chip Debugger 0.11.0 (2021-07-29) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Warn : ignoring extra IDs in hla_vid_pid (maximum is 8 pairs)
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 4000 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK V2J41M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 2.876347
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10076413
Info : flash size = 1024 kbytes
Warn : GDB connection 1 on target stm32f4x.cpu not halted
undefined debug reason 8 - target needs reset
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08016384 msp: 0x20020000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08016384 msp: 0x20020000
configuring PLL
Info : Padding image section 0 at 0x08000188 with 8 bytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08016384 msp: 0x20020000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08016384 msp: 0x20020000
```

I do not know if it mater, I installed the extension using the .VSIX files. My computer is not connected to internet (and must stay unconnected)

In the cortex-debug-server-exiting.log file:
When using the last version:
```
[2023-05-09T12:55:31.063Z] ppid=2516 pid=2516 Cortex Debug: ******* Starting new session request type="launch"
[2023-05-09T12:55:31.171Z] ppid=2516 pid=9372 GDB started ppid=2516 pid=9372
[2023-05-09T12:55:56.333Z] ppid=2180 pid=2180 Cortex Debug: ******* Starting new session request type="launch"
[2023-05-09T12:55:56.390Z] ppid=2180 pid=12608 GDB started ppid=2180 pid=12608
```

When using the 1.6.3 version
```
[2023-05-09T13:03:01.243Z] ppid=10448 pid=10448 Cortex Debug: ******* Starting new session request type="launch"
[2023-05-09T13:03:01.300Z] ppid=10448 pid=15308 GDB started ppid=10448 pid=15308
[2023-05-09T13:04:58.032Z] ppid=10448 pid=15308 Cortex Debug: Begin disconnectRequest
[2023-05-09T13:04:58.187Z] ppid=10448 pid=15308 GDBServer(7136): exited code=0 signal=null
[2023-05-09T13:04:58.188Z] ppid=10448 pid=15308 Cortex Debug: disconnectRequest sendResponse 2
```

I hope this is not too horrible ;-)
Thank you

haneefdm, 2023-05-09:
> I do not know if it mater, I installed the extension using the .VSIX files. My computer is not connected to internet (and must stay unconnected)

Not a problem. No internet requirement from us. VSIX files are fine, AFAIK.

It appears that it is our code that is crashing right after gdb starts. We didn't even get a chance to start the gdb-server and finish initializing the rest of the system.

> I hope this is not too horrible ;-)

Yes, this will be horrible to debug :-) But, in the end it will be something very simple to fix. Hate stuff like this and this is the first time something like this happened. I need a day or so (I do have a day job) to figure out a strategy.

I don't know what kind of help I will need from you yet or what your availability is. It can go a lot faster if we can collaborate real-time but not sure how to do that.

One more request. If you start GDB yourself from a command line

```
arm-none-eabi-gdb.exe -q --interpreter=mi2
```

and enter the following after (on gdb's stdin)
```
1-gdb-version
```

What do you see?

Also, if you just run the command `arm-none-eabi-gdb.exe`, what do you see on the stdout/stderr?

ANauzet, 2023-05-09:
For 
`arm-none-eabi-gdb.exe -q --interpreter=mi2`

```
=thread-group-added,id="i1"
(gdb)
1-gdb-version
~"GNU gdb (GNU Tools for STM32 10.3-2021.10.20211105-1100) 10.2.90.20210621-git\n"
~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
~"\nType \"show copying\" and \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=x86_64-w64-mingw32 --target=arm-none-eabi\".\n"
~"Type \"show configuration\" for configuration details.\n"
~"For bug reporting instructions, please see:\n"
~"<https://www.gnu.org/software/gdb/bugs/>.\n"
~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
~"\n\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
1^done
(gdb)

&"\n"
^done
(gdb)
```

And for `arm-none-eabi-gdb.exe`
```
GNU gdb (GNU Tools for STM32 10.3-2021.10.20211105-1100) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb)
```

Is there a way to get older .VSIX file? I may install different versions and find out from which it start to crash. Maybe it can help?

haneefdm, 2023-05-09:
Could you install this experimental release. I just cannot figure out what is going on. I have more debug messages going to the cortex-debug-server-exiting.log file.

https://github.com/Marus/cortex-debug/releases/tag/v11.1.3

haneefdm, 2023-05-09:
Sorry, wrong release# and link. Here is the right one

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre1

ANauzet, 2023-05-10:
The log seems the same:

```
[2023-05-10T12:43:30.230Z] ppid=1940 pid=1940 Cortex Debug: ******* Starting new session request type="launch"
[2023-05-10T12:43:30.305Z] ppid=1940 pid=15988 GDB started ppid=1940 pid=15988
```

But I'm not sure the extension install correctly. The version displayed seems correct (v1.11.3-pre1) but there was a error message when installing it: "Cannot activate the 'Cortex-Debug' extention because it depends on an unknown 'mcu-debug.debug-tracker-vscode' extension

haneefdm, 2023-05-10:
We now depend on 2-3 other extensions. You have to install all of them manually (or use their VSIX files) as VSCode cannot do that for you if you are not connected to the internet. While we do not need internet at runtime, we do need it to finish the installation.

These are the four extensions we depend on (they are all on the marketplace)
```
    "extensionDependencies": [
        "mcu-debug.debug-tracker-vscode",
        "mcu-debug.memory-view",
        "mcu-debug.rtos-views",
        "mcu-debug.peripheral-viewer"
    ],
```
I am not sure how got that far to be able to start a debugger without getting any errors before. Or I don't know why VSCode did not automatically download and install the other extensions.

FYI: We had to break up our extension because it was getting too big and we were getting warnings from MS. Other reasons too.

ANauzet, 2023-05-10:
> Or I don't know why VSCode did not automatically download and install the other extensions.

That must be because the PC is not on the internet. But there was no error on installation...

Very good news, it is now working! I can debug step by step with last version you provide (and version 1.11.2). It seems that is was the missing extension. Really dont understand what was wrong with the installation

ANauzet, 2023-05-10:
Ok, I made a little test:

After removing all the extensions, cortex-debug and dependencies, I tried to install the last version of Cortex-debug, and there was the dependencies error. I was like "f**, i just miss the error message"

Buuuuuuut:
I remove the extension, install the 1.6.3, and then update to the 1.11.2 version, and NO error displayed!

Phew!

haneefdm, 2023-05-10:
So, closing this now.

closed 2023-05-10 by haneefdm

ANauzet, 2023-05-11:
Thank you for your support!