Marus/cortex-debug #875: Breakpoints cleared from VSCode BREAKPOINTS panel persist

GregTerrell, 2023-05-02:
If I clear a breakpoint from within the VSCode UI, I will frequently (not 100%) continue to hit and stop on the breakpoint line during execution. 

Upon examination in the DEBUG CONSOLE gdb "info b" will generally report multiple breakpoints set for the same line number. Using the gdb del # command deletes them so execution can resume, but the VSCode UI is not updated (that is not my issue here), which is understandable. 

Sooner or later the breakpoint manually deleted in gdb is deleted in VSCode, which then complains "not found". In this scenario, I must have deleted the VSCode matching break and not the orphan.

**Cortex Debug** v1.80 but reported at launch with "Cortex-Debug: VSCode debugger extension version 1.10.0 git(cf1c296). Usage info: https://github.com/Marus/cortex-debug#usage"

**Segger** J-Link GDB server V7.64 
**GDB** install: GNU Arm Embedded Toolchain 10 2021.10

haneefdm, 2023-05-02:
> Cortex Debug v1.80 but reported at launch with "Cortex-Debug: VSCode debugger extension version 1.10.0 git(https://github.com/Marus/cortex-debug/commit/cf1c2960a54ca1e0963422e065a0abcb586643c6). Usage info: https://github.com/Marus/cortex-debug#usage"

Hmmm. Not sure how this can happen!?!?! Will investigate.

> Upon examination in the DEBUG CONSOLE gdb "info b" will generally report multiple breakpoints set for the same line number

Is this a C++ project? Or breakpoint set in an inline function?

Can you give more details on your program?

And, thank you for reporting this issue.

GregTerrell, 2023-05-02:
The breakpoints are not inline functions. Both basic and conditional breakpoints have exhibited this behavior. 

The application being debugged is an Arduino (MS extension) test .ino application for a device (product) driver. The driver and test are public in GitHub: https://github.com/LooUQ/CircuitRiver-LTEmC.

I am working on it today and will post up a link to an demonstration image today.

GregTerrell, 2023-05-02:
I will attempt to capture the steps leading up to the issue today. Below is the build log.

[Starting] Verifying sketch 'ltemc-9-http.ino'
Please see the build logs in output path: c:\Users\GregTerrell\Documents\CodeDev\Arduino\libraries\LooUQ-LTEmC\tests\ltemc-9-http\build
Using board 'adafruit_feather_m0_express' from platform in folder: C:\Users\GregTerrell\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.5
Using core 'arduino' from platform in folder: C:\Users\GregTerrell\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.5
Detecting libraries used...
"C:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"ltemc-9-http.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_FEATHER_M0_EXPRESS -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0 Express\"" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -Os -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0 Express\"" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5/libraries/Adafruit_TinyUSB_Arduino/src/arduino" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/Core/Include/" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/DSP/Include/" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS-Atmel\\1.2.2/CMSIS/Device/ATMEL/" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5\\cores\\arduino" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5\\variants\\feather_m0_express" "C:\\Users\\GregTerrell\\Documents\\CodeDev\\Arduino\\libraries\\LooUQ-LTEmC\\tests\\ltemc-9-http\\build\\sketch\\ltemc-9-http.ino.cpp" -o nul
Alternatives for jlinkRtt.h: [SEGGER_JLINK_RTT@0.0.1]
ResolveLibrary(jlinkRtt.h)
  -> candidates: [SEGGER_JLINK_RTT@0.0.1]
"C:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"ltemc-9-http.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_FEATHER_M0_EXPRESS -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0 Express\"" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -Os -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0 Express\"" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5/libraries/Adafruit_TinyUSB_Arduino/src/arduino" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/Core/Include/" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/DSP/Include/" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS-Atmel\\1.2.2/CMSIS/Device/ATMEL/" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5\\cores\\arduino" "-IC:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.5\\variants\\feather_m0_express" "-IC:\\Users\\GregTerrell\\Documents\\Codedev\\Arduino\\libraries\\JLinkRTT\\src" "C:\\Users\\GregTerrell\\Documents\\CodeDev\\Arduino\\libraries\\LooUQ-LTEmC\\tests\\ltemc-9-http\\build\\sketch\\ltemc-9-http.ino.cpp" -o nul
Alternatives for lq-diagnostics.h: [LooUQ-EmbedDvc@1.0.1]
ResolveLibrary(lq-diagnostics.h)
  -> candidates: [LooUQ-EmbedDvc@1.0.1]
"C:\\Users\\GregTerrell\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"ltemc-9-http.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_FEATHER_M0_EXPRESS -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0  [... cut ...]
SPI                       1.0     C:\Users\GregTerrell\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.5\libraries\SPI             
Adafruit Zero DMA Library 1.1.0   C:\Users\GregTerrell\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.5\libraries\Adafruit_ZeroDMA

Used platform Version Path                                                                              
adafruit:samd 1.7.5   C:\Users\GregTerrell\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.5
IntelliSense auto-configuration disabled.
[Done] Verifying sketch 'ltemc-9-http.ino'

haneefdm, 2023-05-02:
All breakpoints are handled by VSCode. Not us. Even more complicated is that since it is built on top of Chrome, they are stuffed somewhere in the Chrome database. Chromium actually.

Note that the design says that you can set breakpoints can be set anywhere, anytime and even when our extension is not even installed. So, they are stored globally in some magic place.

haneefdm, 2023-05-02:
You are using `-Os` while compiling. That will likely trigger inlining. Could you try to use `-Og` which is more amenable for debugging? We will see if we can do anything about `-Os` later but I want to confirm first.

GregTerrell, 2023-05-02:
Sorry about the markdown, will do. The replacement of -Os with -Og had no effect.

I think I found the reproduction steps.

1. Hit a breakpoint
2. Clear the breakpoint within VSCode UI (F9, delete from BREAKPOINTS panel, doesn't matter)
3. Step or run
4. Pause or hit another break and the cleared breakpoint is still visible in GDB.

If the **execution steps off** of the line where the breakpoint is to be cleared, the subsequent clearing through the VSCode UI **is reflected** in GDB.

haneefdm, 2023-05-02:
Just so I am not chasing the wrong thing. Can you try `-O0` instead of `-0g` or `-Os`

GregTerrell, 2023-05-02:
My previous statement about -Og having no effect was incorrect. Upon further inspection the build was dropping in an additional -Os after my extra compiler switch with -Og. I had to go supplement the boards.txt to ensure that the -Og was the last optimization flag on the command line.

With -Og in effect, the problem goes away.