Marus/cortex-debug #878: Support for USB-Blaster or USB-Blaster2

sparkertim, 2023-05-08:
USB-Blaster is the debugger of intel's FPGA soft-core controller nios2
I have add support for these debugger
Can I push my code to this branch? it is not arm architecture

sparkertim, 2023-05-09:
> 
it is a support for nios2 gdb server.
I have added the  [ublast.ts](https://github.com/sparkertim/cortex-debug/blob/master/src/ublast.ts) to support nios2-gdb-server, based on the version two years ago.
- support use specified usb path to select jtag debugger.
- support select device in jatg chain by specified id.

haneefdm, 2023-05-09:
I will take a better look. But, first, you need to install ESLint ("dbaeumer.vscode-eslint"). We know that it has been deprecated but we still need it for this project. You will not be able to submit a PR either. Just visually, I see a lot of issues.

I noticed that post connect (launch and attach), the processor is not in a halted state. Many things will fail in VSCode. There is a fundamental assumption about this. If it worked for you, then it may be happenstance and you may  not have tested all kinds of features with cortex-debug and VSCode. Also, with a `launch` we expect the program to be reset and halted at the first instruction (generally at the reset vector and processor type does not matter). Does `target-select extended-remote ...` halt the processor?

How you do a `target-download` does not work with how all the other gdb-servers work and our own features like `loadFiles`, `symbolFiles`, etc. will not work. Note that these are more gdb features than server features.  Why did you depart from how all the other servers work?

When was this forked? It appears many years ago. That will not work.

Note that since I cannot test any of this, you will have to provide support for this server and I can only

sparkertim, 2023-05-09:
> I will take a better look. But, first, you need to install ESLint ("dbaeumer.vscode-eslint"). We know that it has been deprecated but we still need it for this project. You will not be able to [...]

yeah, i wrote these codes two years ago when i first use typescript, so there may be many issues in my code.

I prepare to modify these codes base on the latest version

nio2-gdb-server can not support some features when  the server started, 
i can not find any command to reset nios2 using nios2-elf-gdb. so i need to halt the CPU when calling the nios2-gdb-server。 at [line 201](https://github.com/sparkertim/cortex-debug/blob/master/src/ublast.ts#L201) i add '-r' to reset and halt the nios2 when nios2-gdb-server started,  the nios2 will reset everytime the  elf loaded

The following are the commands supported by nios-gdb-server
```
nios2-gdb-server --help
usage: nios2-gdb-server [-h/--help] [-c/--cable <cable name>]
       [-d/--device <device index>] [-i/--instance <instance>]
       [-s/--sidp <address>] [-I/--id <id>] [-t/--timestamp <timestamp>]
       [-w/--wait <seconds>] [--no-verify] [-g/--go] [--stop]
       [--tcpport=<port> | auto] [--tcpdebug] [--tcptimeout <seconds>]
       [--tcppersist] [--thread-vars <vars>] 
       [--write-gmon <gmonfile>] [--wait-target-exit]
       <filename.srec>*

 -h/--help                   Print this message

    --write-pid <filename>   Write the process ID to the file specified
    --signal-pid <filename>  Send a SIGUSR1 to the other process when ready

 -c/--cable <cable name>     Specifies which JTAG cable to use (not needed if
                             you only have one cable)
 -d/--device <device index>  Specifies in which device you want to look for the
                             Nios II OCI core (1 = device nearest TDI etc.)
 -i/--instance <instance>    Specifies the INSTANCE value of the JTAG debug core
                             (not needed if there is exactly one on the chain)

 -r/--reset-target           Always reset the target system before use

 -C/--init-cache             Always initialize target processor cache before use

 -s/--sidp <address>         Base-address of System ID peripheral on target
 -I/--id <system-id-value>   Unique ID code for target system
 -t/--timestamp <time-stamp> Timestamp for target-system (when last generated)
    --accept-bad-sysid       Continue even if the system ID comparison fails

 <filename.srec>*            S-Record file(s) to download to the target
    --no-verify              Don't verify after downloading

 -w/--wait <seconds>         Wait for time specified before starting processor
                             (signal USR1 will terminate the wait early)
 -g/--go                     Run processor from entry point after downloading
    --stop                   Stop processor (leave it in a paused state).

    --tcpport <port> | auto  Listen on specified TCP port for a connection from
                             GDB (with "auto" a free port will be used).
    --tcpdebug               Display the GDB remote protocol running over TCP
    --tcptimeout <seconds>   Timeout if GDB has not connected in time specified
    --tcppersist             Don't stop when GDB disconnects
    --thread-vars <vars>     Decode target thread list using specified string

    --write-gmon <gmonfile>  Wait for target to complete and write GMON data
                             (if available) to the file specified
    --wait-target-exit       Wait for target to complete and return target exit
                             code as our return code if no other errors

The processor state after execution is chosen as follows:
  1) If the GDB remote protocol is enabled then this chooses the state.
  2) If the --go option is specified then the processor will be started.
  3) If the --stop option is specified then the processor will be left paused.
  4) If a file is downloaded or the procesor is reset then it will be paused.
  5) Otherwise, the processor state will be restored.
```

haneefdm, 2023-05-09:
The options you show appear to be command-line options and runtime commands. Most gdb-servers support the gdb command `monitor help` once the server has started. This should give us a list of commands the server supports once it has started.

You can also do an `exec-interrupt` perhaps to halt but that does not do a reset. We need both reset and halt in most cases. So, `breakAfterReset` will never work. This is an important thing for embedded developers.

Have you tried looking at their Eclipse tools. You can enable (can't remember how) gdb traces and see ALL the commands going to gdb and any command that starts with `mon` or `monitor` are actually commands for the server.

sparkertim, 2023-05-09:
> The options you show appear to be command-line options and runtime commands. Most gdb-servers support the gdb command `monitor help` once the server has started. This should give us a list of [...]

it seems already reset and halt cpu after load again when start nios2-gdb-server with '-r', i will test it after i accomplished the new version
```
(gdb) load build/test.elf
Loading section .entry, size 0x20 lma 0x0
Loading section .exceptions, size 0x254 lma 0x20
Loading section .text, size 0x2a4c0 lma 0x274
Loading section .rodata, size 0x1bac lma 0x2a734
Loading section .rwdata, size 0x1db8 lma 0x2c2e0
Start address 0x00000274, load size 188568
Transfer rate: 559 KB/sec, 366 bytes/write.
(gdb) stepi
0x00000278 in ?? ()
(gdb) stepi
0x0000027c in ?? ()
(gdb) stepi
0x00000280 in ?? ()
(gdb) load build/test.elf
Loading section .entry, size 0x20 lma 0x0
Loading section .exceptions, size 0x254 lma 0x20
Loading section .text, size 0x2a4c0 lma 0x274
Loading section .rodata, size 0x1bac lma 0x2a734
Loading section .rwdata, size 0x1db8 lma 0x2c2e0
Start address 0x00000274, load size 188568
Transfer rate: 535 KB/sec, 366 bytes/write.
(gdb) stepi
0x00000278 in ?? ()
(gdb) stepi
0x0000027c in ?? ()
(gdb) 
```

haneefdm, 2023-05-09:
For launch, I guess you could use some combination of --go and --stop on the command-line?

There are sometimes equivalent gdb commands to try as well. For a restart for instance. Not being able to do a proper reset if a big flaw and hindrance. There is a fundamental assumption that we should be able to do a reset/restart in all of VSCode (not just for embedded). I see what you did for a restart and hope that works :-)

You probably should not use the `-r` if the request is for an attach. But then restart is broken. Please try `monitor halt` once a session stars and see what the server prints.

I am not liking supporting a gdb-server that has zero documentation or support. Like how we do a pause, we use the `exec-interrupt` gdb command.

I hope you get comfortable with the concepts and various gdb commands. If not, I cannot support this as I don't have any way/time to test various things.

haneefdm, 2023-05-09:
Did you approach Altera/Intel asking for documentation on the server beyond the command-line options.

sparkertim, 2023-05-09:
> For launch, I guess you could use some combination of --go and --stop on the command-line?
>
> There are sometimes equivalent gdb commands to try as well. For a restart for instance. Not being able to [...]

i can not find monitor reset halt nor monitor halt by using nios2-elf-gdb directly, with '-r', and use ```interpreter-exec console "load"```  can reset and halt the cpu.
maybe i can provide you the full commands supported by nios2 gdb client  later. it will take some time to use 'help xxx' to obtain them

sparkertim, 2023-05-09:
like this? 
```
GNU gdb (GDB) 9.2.90.20200608-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=nios2-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.altera.com/mysupport>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) apropos reset
directory -- Add directory DIR to beginning of search path for source files.
mem -- Define attributes for memory region or reset memory region handling to target-based.
set auto-load safe-path -- Set the list of files and directories that are safe for auto-loading.
set remote environment-reset-packet -- Set use of remote protocol `QEnvironmentReset' (environment-reset) packet.
show auto-load safe-path -- Show the list of files and directories that are safe for auto-loading.
show remote environment-reset-packet -- Show current use of remote protocol `QEnvironmentReset' (environment-reset) packet.
(gdb) help
List of classes of commands:

aliases -- Aliases of other commands.
breakpoints -- Making program stop at certain points.
data -- Examining data.
files -- Specifying and examining files.
internals -- Maintenance commands.
obscure -- Obscure features.
running -- Running the program.
stack -- Examining the stack.
status -- Status inquiries.
support -- Support facilities.
tracepoints -- Tracing of program execution without stopping the program.
user-defined -- User-defined commands.
```

sparkertim, 2023-05-09:
> Nope. Not `help`. It is `monitor help`
> 
> After a connection has been made with the gdb-server

not support, but i can debug with it. reset, restart, step works fine
```
monitor help
From client: evaluate({"expression":"monitor help","frameId":4096,"context":"repl"})
61-interpreter-exec console "monitor help"
-> &"Target does not support this command.\n"
Target does not support this command.
-> 61^error,msg="Target does not support this command."
To client: {"seq":0,"type":"response","request_seq":22,"command":"evaluate","success":false,"message":"Target does not support this command. (from interpreter-exec console \"monitor help\")","body":{"error":{"id":8,"format":"Target does not support this command. (from interpreter-exec console \"monitor help\")","showUser":true}}}
Target does not support this command. (from interpreter-exec console "monitor help")
From client: scopes({"frameId":4096})
```

closed 2023-05-09 by sparkertim