Marus/cortex-debug #882: debugger stops unexpectedly

tdjastrzebski, 2023-05-21:
Although recent improvements are very helpful, there is one issue I occasionally notice which did not seem to happen before.
Debugger quits while stepping through even very simple code like simple int variable assignments.
The problem seems to happen randomly. Tested with the most recent version 1.11.2, OpenOCD 0.12.0 and arm-none-eabi-gcc 12.2.1

```
From client: next({"threadId":1})
290-exec-next --thread 1
-> 290^running
-> *running,thread-id="all"
mi2.status = running
To client: {"seq":0,"type":"event","event":"continued","body":{"threadId":1,"allThreadsContinued":true}}
To client: {"seq":0,"type":"event","event":"custom-continued","body":{"threadID":1,"allThreads":true}}
To client: {"seq":0,"type":"response","request_seq":143,"command":"next","success":true}
From client: scopes({"frameId":4096})
291-stack-info-frame --thread 1 --frame 0
-> 291^error,msg="Selected thread is running."
```

haneefdm, 2023-05-21:
> Debugger quits while stepping through even very basic code like simple int variable assignments.

Can you describe what "debugger quits" means? Debug session stops? VSCode exits?

The sequence of commands is strange though.

> 291-stack-info-frame --thread 1 --frame 0

Not sure who is requesting a stack frame while program is running. Any frame requests come from VSCode (or another extension) and happen right after a halt. It may be a request from a previous halt and that is totally out of order. Even then, it should not cause the session to end.

Can I see a bit more of the output from the debug console?

tdjastrzebski, 2023-05-22:
@haneefdm What happens is that VS Code debug tab (img below) disappears and debug session stops.
I will sent more output once the problem happens again.
[image]

ANauzet, 2023-05-22:
Hello,
I have the same behavior as you describe. In debug session, the "debug tab" disappears. It seems random (not always crashing at same code location)
I'm using last cortex-debug version. Here is the contain of the cortex-debug-server-exiting.log file:

```
[2023-05-22T13:20:55.081Z] ppid=14724 pid=14724 Cortex Debug: ******* Starting new session request type="launch"
[2023-05-22T13:20:55.111Z] ppid=14724 pid=14780 GDB started ppid=14724 pid=14780
[2023-05-22T14:31:54.885Z] ppid=14724 pid=14780 Debug Adapter crashed or killed by VSCode? code=1 signal=undefined
[2023-05-22T14:31:54.886Z] ppid=14724 pid=14780 GDBServer(4088): forcing an exit with kill()

```

Debug Console last display is `halted PC: 0x....` and no other trace

haneefdm, 2023-05-22:
@ANauzet  I can't work with just one line from 'Debug Console'. I need some history...at least a few dozen lines

Not every problem is the same issue/cause.

haneefdm, 2023-05-22:
Can you please try out the following pre-release and give feedback

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre2

ANauzet, 2023-05-23:
Hello,

From last tests made with the pre-release I can confirm that:
- The crash occur randomly. Sometime I can make many step by step on specific function. Sometime the crash occur really fast
- The crash only occur in step by step mode. When running, the session stay alive
- Even with a very simple program, -ie- a loop with a printf and a delay, the crash occur

The 'Debug Console' only show a long list of "halted...":
```
Output radix now set to decimal 10, hex a, octal 12.
Input radix now set to decimal 10, hex a, octal 12.
0x0800f78c in prvUARTCommandConsoleTask (pvParameters=<optimized out>) at Common/FreeRTOS_Plus_CLI_Demos/UARTCommandConsole.c:259
259	            else if ( ( cRxedChar == '\b' ) || ( cRxedChar == cmdASCII_DEL ) )
Program stopped, probably due to a reset and/or halt issued by debugger
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0801a3dc msp: 0x20020000
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0801a3dc msp: 0x20020000
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, prvUARTCommandConsoleTask (pvParameters=<optimized out>) at Common/FreeRTOS_Plus_CLI_Demos/UARTCommandConsole.c:179
179	        vOutputString( pcWelcomeMessage );
halted: PC: 0x0800f6e4
halted: PC: 0x0800f680
halted: PC: 0x0800f6de
halted: PC: 0x0800f6e0
halted: PC: 0x0800f75c
halted: PC: 0x0800f760
halted: PC: 0x0800f764
halted: PC: 0x0800f766
halted: PC: 0x080014b4
halted: PC: 0x0800f76c
halted: PC: 0x0800f76e
halted: PC: 0x0800f772
halted: PC: 0x0800f776
halted: PC: 0x0800f694
halted: PC: 0x0800f77e
halted: PC: 0x0800f780
halted: PC: 0x0800f782
halted: PC: 0x0800f784
halted: PC: 0x0800f786
halted: PC: 0x0800f788
halted: PC: 0x0800f78a
halted: PC: 0x0800f78c
halted: PC: 0x0800f78e
halted: PC: 0x0800f790
halted: PC: 0x0800f792
halted: PC: 0x0800f796
halted: PC: 0x0800f798
halted: PC: 0x0800f79a
halted: PC: 0x0800f79c
halted: PC: 0x0800f79e
halted: PC: 0x0800f7a0
halted: PC: 0x0800f7a2
halted: PC: 0x0800f7a4
halted: PC: 0x0800f7a6
halted: PC: 0x0800f7a8
halted: PC: 0x0800f75c
halted: PC: 0x0800f760
halted: PC: 0x0800f764
halted: PC: 0x0800f766
halted: PC: 0x080014b4
halted: PC: 0x0800f76c
halted: PC: 0x0800f76e
halted: PC: 0x0800f772
halted: PC: 0x0800f776
halted: PC: 0x0800f694
halted: PC: 0x0800f77e
halted: PC: 0x0800f780
halted: PC: 0x0800f782
[... 148 lines cut ...]
Info : halted: PC: 0x0800f772
Info : halted: PC: 0x0800f776
Info : halted: PC: 0x0800f694
Info : halted: PC: 0x0800f77e
Info : halted: PC: 0x0800f780
Info : halted: PC: 0x0800f782
Info : halted: PC: 0x0800f784
Info : halted: PC: 0x0800f786
Info : halted: PC: 0x0800f788
Info : halted: PC: 0x0800f78a
Info : halted: PC: 0x0800f78c
Info : halted: PC: 0x0800f78e
Info : halted: PC: 0x0800f790
[2023-05-23T09:05:57.319Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET
[2023-05-23T09:05:57.321Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

The cortex-debug-server-exiting.log file do not contain more information than previously:
```
[2023-05-23T09:16:11.813Z] ppid=12988 pid=12988 Cortex Debug: ******* Starting new session request type="launch"
[2023-05-23T09:16:11.868Z] ppid=12988 pid=15652 GDB started ppid=12988 pid=15652
[2023-05-23T09:16:32.504Z] ppid=12988 pid=15652 Debug Adapter crashed or killed by VSCode? code=1 signal=undefined
[2023-05-23T09:16:32.505Z] ppid=12988 pid=15652 GDBServer(13760): forcing an exit with kill()
```

tdjastrzebski, 2023-05-23:
[dbg_console_output-2.txt](https://github.com/Marus/cortex-debug/files/11545064/dbg_console_output-2.txt)
Checked with V1.11.3-pre2, the problem persists.
In some ways it is deterministic, debug stops usually in the same place.
After some change in code it stops completely somewhere else.

haneefdm, 2023-05-23:
Could you set `"showDevDebugOutput": "vscode"` and attach the output of Debug Console, so I can see what VSCode is doing because there are out-of-band requests coming from somewhere that don't make sense?

ANauzet, 2023-05-24:
Hello,
I have done the same test with Debug Output. Here is my Debug Console file.
[bug.txt](https://github.com/Marus/cortex-debug/files/11552059/bug.txt)

haneefdm, 2023-05-26:
Thanks @ANauzet It helped me see what the issue is. In VSCode, everything is asynchronous and there are many processes involved (VSCode, gdb, gdb-server), and ourselves in the middle. This is caused by an unexpected sequence of events.

I am assuming that this is happening when you rapidly hit the step/next button. VSCode is still querying for stack info from a previous stop while it also asked for a step before that. If it has already asked us to step/next it should not be asking for information about the previous stop. But it does. Button events are not synchronized with the GUI updates. But, this can happen in other ways as well.

Not sure how to handle this but at least will try to work around it. Please stay tuned.

Btw, I cannot duplicate this on my laptop(s). So, I am working part blind here.

tdjastrzebski, 2023-05-28:
@haneefdm I would not say it happens only in case of quick step-through break points. The problem happens also when the step-through is very slow. That is, I wait at least several seconds at each break point. Another debug console attached.
[dbg_console_output-3.txt](https://github.com/Marus/cortex-debug/files/11585532/dbg_console_output-3.txt)

haneefdm, 2023-06-06:
Can we give this another try...

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre5

This is a pre-release

tdjastrzebski, 2023-06-07:
@haneefdm Thank you, I will give it a try.

haneefdm, 2023-06-08:
My changes are now in the VSCode Marketplace as a pre-release version 1.11.3.

Please let me if I need to reopen this. Maybe it is fixed now, but I need to push out other changes.