Marus/cortex-debug #883: Peripheral register access not working right

drobinson55, 2023-05-22:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
I have the SVD file set up, and two windows with which to view the registers of an LPC845 (Cortex m0+).  One window shows 0xffffffff for every register, one window shows 0x0.  If I go to the command line and type x/10xw 0x40064000, the address of USART0 for that chip, I see Cannot Access memory at address 0x40064000.  I have some logging in the code where I go out and read the register from the peripheral and the correct value seems to be there.

When I look at the registers, the value 245 is in r0, having been read from the peripheral CFG register at 0x40064000.

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Click on '....'
3. Scroll down to '....'
4. See issue

**Expected behavior**

I can seem to read and write the register at 0x40064000, which is the CFG register of USART0 on an LPC845.  I expect to be able to read those registers back via the debugger.

**Screenshots**
[image]

[image]

[image]
[comment]: <> If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version 1.10.0
 - OS: Windows 11
 - GDB Version: NXP LinkServer 1.0.9
 - Compiler Toolchain Version: arm-none-eabi 12.2 mpacbti.34
 - 
**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
Paste launch.json contents here
```

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/lpc845.elf",
            "name": "Debug with OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "external",
            "gdbTarget": "localhost:3333",
            "serverpath": "${env:BUILD_TOOLS_ROOT}/openocd/bin/openocd.exe",
            #"configFiles": [ "${env:BUILD_TOOLS_ROOT}/openocd/share/openocd/scripts/board/nxp_lpcxpresso845max.cfg"],
            "searchDir": [],
            "runToEntryPoint": "main",
            "showDevDebugOutput": "raw",
            "armToolchainPath": "${env:BUILD_TOOLS_ROOT}/gcc/bin",
            "device": "LPC84x",
            "interface": "swd",
            "serialNumber": "", // add J-Link serial number if having multiple attached the same time.
            "runToMain": false,
            "svdFile": "${workspaceRoot}/L4-Hal/lpc845/vendor/LPC845.svd"
        } ,
        {
            "cwd": "${workspaceRoot}",
[... 438 lines cut ...]
146-stack-list-variables --thread 1 --frame 0 --simple-values
-> 146^done,variables=[{name="pBase",type="volatile USART_Type *",value="0x40064000"},{name="cfgRegVal",type="uint32_t",value="245"},{name="stopBitsMask",type="uint32_t",value="64"},{name="dataBitsMask",type="uint32_t",value="4"},{name="parityMask",type="uint32_t",value="176"},{name="baudRateSetting",type="uint32_t",value="16"},{name="pConfigCtxt",arg="1",type="const HalUsartConfigContext_t *",value="0x1930 <s_consoleUsartConfigCtxt>"},{name="pDataCtxt",arg="1",type="HalUsartDataContext_t *",value="0x10000018 <s_consoleUsartDataCtxt>"},{name="platStatus",type="PlatStatus_t",value="ePLAT_STATUS_INVALID_ARG"}]
147-var-update --thread 1 --frame 0 --all-values var_pBase_256
-> 147^error,msg="Variable object not found"
148-var-create --thread 1 --frame 0 var_pBase_256 * "pBase"
-> 148^done,name="var_pBase_256",numchild="12",value="0x40064000",type="volatile USART_Type *",thread-id="1",has_more="0"
149-var-update --thread 1 --frame 0 --all-values var_cfgRegVal_256
-> 149^done,changelist=[{name="var_cfgRegVal_256",value="245",in_scope="true",type_changed="false",has_more="0"}]
150-var-update --thread 1 --frame 0 --all-values var_stopBitsMask_256
-> 150^done,changelist=[{name="var_stopBitsMask_256",value="64",in_scope="true",type_changed="false",has_more="0"}]
151-var-update --thread 1 --frame 0 --all-values var_dataBitsMask_256
-> 151^done,changelist=[{name="var_dataBitsMask_256",value="4",in_scope="true",type_changed="false",has_more="0"}]
152-var-update --thread 1 --frame 0 --all-values var_parityMask_256
-> 152^done,changelist=[]
153-var-update --thread 1 --frame 0 --all-values var_baudRateSetting_256
-> 153^done,changelist=[{name="var_baudRateSetting_256",value="16",in_scope="true",type_changed="false",has_more="0"}]
154-var-update --thread 1 --frame 0 --all-values var_pConfigCtxt_256
-> 154^done,changelist=[]
155-var-update --thread 1 --frame 0 --all-values var_pDataCtxt_256
-> 155^done,changelist=[]
156-var-update --thread 1 --frame 0 --all-values var_platStatus_256
-> 156^done,changelist=[]

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-05-22:
It appears you have multiple SVD (Peripheral) views, coming from various other extensions. The one "CORTEX REGISTERS" *was* from this extension. Others belong to other extensions and are potentially obsolete. The one where you are seeing all F's is from another developer and is probably now obsolete.

Please make your extensions current and remove obsolete ones. Not a good idea to have multiple views either as it affects debug performance.

Even our own is now obsolete has moved to [another repo](https://github.com/mcu-debug/peripheral-viewer).

When you install the pre-release of this extension, we now rely on 'XPERIPERALS' for SVD from the above repo and don't have owr own anymore.

If the issue still persists, then please update this extension to the latest pre-release in VSCode manager and try again.

drobinson55, 2023-05-22:
Hi Haneef,

Thanks for the quick response.  I switched to the Preview and it is not listed as Venus's Cortex-Debug.  Is this correct?

I was wondering about a conflict between different extensions.  I couldn't find the canonical list of extensions required.  I don't know how to get rid of the CORTEX PERIPERALS window.  I disabled an extension that dropped the XPERIPHERALS window.

[cid:d2732120-e3d9-4242-a0f0-6913886a78c0]

Do you have a canonical list of the required extensions?

I think this is a great tool, thanks for your efforts!

--Daniel

Daniel G. Robinson

Senior Firmware Engineer

***@***.***

Voice:      425-251-8777

Fax:          425-251-0536

[cid:27b37ebe-447f-4120-9891-30d8d14d925e]

Tri-Tec Manufacturing, LLC

6915 S 234th St

Kent, WA 98032

[cid:109d0b92-e885-48a7-9dbe-150ca98a9a1b]

CUI

CONFIDENTIALITY NOTICE:  This electronic message and its attachments may contain confidential information of Tri-Tec Manufacturing LLC and/or its vendors and clients.  If you are not the intended recipient, you may not read, disclose, print, copy, distribute or otherwise use in any way the information.  If you have received this electronic transmission in error, please notify the sender immediately and delete the electronic transmission.

haneefdm, 2023-05-22:
I have no clue who/what `Venus's Cortex-Debug` is. Our extension id is marus25.cortex-debug

https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug

When you install this, it will install any related extension it needs. We are open source and we cannot prevent people from forking and making copies. Please be careful what you install. Always visit their repos to see who is owning these extensions.

closed 2023-05-22 by haneefdm

drobinson55, 2023-05-22:
Hi Haneef,

> I have no clue who/what Venus's Cortex-Debug is. Our extension id is marus25.cortex-debug

LOL

I got that when I selected the preview version of Cortex-Debug.  I uninstalled it, and re-installed from the link you provided.

I think there are some other problems.  When I run arm-none-eabi-gdb from a command line, x/10xw 0x40054000, the address of the USART0 register block comes back with

(gdb) x/10xw 0x40064000
0x40064000:     Cannot access memory at address 0x40064000
(gdb) p pBase
$1 = (volatile USART_Type *) 0x40064000
(gdb) p *pBase
Cannot access memory at address 0x40064000

The tracing I have in place leads me to believe that the M0+ core can complete write and read cycles to those addresses, but GDB seems to not be able to.

I don't think that's your problem.

Thanks for your help,

Daniel

Daniel G. Robinson

Senior Firmware Engineer

***@***.***

Voice:      425-251-8777

Fax:          425-251-0536

[cid:ea28064e-495f-45e0-b9f9-ce6475d40e1a]

Tri-Tec Manufacturing, LLC

6915 S 234th St

Kent, WA 98032

[cid:adfefbbe-821e-4a2d-83f4-0ab0964baa0a]

CUI

CONFIDENTIALITY NOTICE:  This electronic message and its attachments may contain confidential information of Tri-Tec Manufacturing LLC and/or its vendors and clients.  If you are not the intended recipient, you may not read, disclose, print, copy, distribute or otherwise use in any way the information.  If you have received this electronic transmission in error, please notify the sender immediately and delete the electronic transmission.

https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug<https://usg02.safelinks.protection.office365.us/?url=https%3A%2F%2Fmarketplace.visualstudio.com%2Fitems%3FitemName%3Dmarus25.cortex-debug&data=05%7C01%7Cdrobinson%40tritecmfg.com%7Ccc79a75704c948e2ffe308db5aea0b8d%7C7ccabc122e2a4a618b94097458df88e8%7C0%7C0%7C638203733483062887%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=mNWdTNECzS9tt0ImGzXjoLkHcsqT0MsqEYI9UA4nrjg%3D&reserved=0>