Marus/cortex-debug #886: Failed to launch OpenOCD GDB Server: Timeout.

defyg, 2023-05-24:
Encounter the "Failed to launch OpenOCD GDB Server: Timeout" error dialog in VScode.

Enabled the "showDevDebugOutput": "raw" feature, found it stops at step 8,  the debug log is attached 

> Cortex-Debug: VSCode debugger extension version 1.10.0 git(cf1c296). Usage info: https://github.com/Marus/cortex-debug#usage
> Reading symbols from /usr/bin/objdump-multiarch --syms -C -h -w /home/alex/VS_code/Rust/cortex-m4-quick-start/target/thumbv7em-none-eabihf/debug/examples/hello
> Reading symbols from /usr/bin/nm-multiarch --defined-only -S -l -C -p /home/alex/VS_code/Rust/cortex-m4-quick-start/target/thumbv7em-none-eabihf/debug/examples/hello
> Launching GDB: /usr/bin/gdb-multiarch -q --interpreter=mi2
> 1-gdb-version
> Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/alex/VS_code/Rust/cortex-m4-quick-start -f /home/alex/.vscode/extensions/marus25.cortex-debug-1.10.0/support/openocd-helpers.tcl -f interface/stlink-v2-1.cfg -f target/stm32f4x.cfg
>     Please check TERMINAL tab (gdb-server) for output from openocd
> Finished reading symbols from objdump: Time: 31 ms
> Finished reading symbols from nm: Time: 39 ms
> -> =thread-group-added,id="i1"
> -> ~"GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2\n"
> -> ~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
> -> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
> -> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
> -> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
> -> ~"Type \"show configuration\" for configuration details.\n"
> -> ~"For bug reporting instructions, please see:\n"
> -> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
> -> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
> -> ~"\n\n"
> -> ~"For help, type \"help\".\n"
> -> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
> -> 1^done
> 2-gdb-set mi-async on
> -> 2^done
> 3-interpreter-exec console "set print demangle on"
> -> 3^done
> 4-interpreter-exec console "set print asm-demangle on"
> -> =cmd-param-changed,param="print asm-demangle",value="on"
> -> 4^done
> 5-enable-pretty-printing
> -> 5^done
> 6-interpreter-exec console "source /home/alex/.vscode/extensions/marus25.cortex-debug-1.10.0/support/gdbsupport.init"
> -> 6^done
> 7-interpreter-exec console "source /home/alex/.vscode/extensions/marus25.cortex-debug-1.10.0/support/gdb-swo.init"
> -> =cmd-param-changed,param="language",value="c"
> -> =cmd-param-changed,param="language",value="auto"
> -> 7^done
> 8-file-exec-and-symbols "/home/alex/VS_code/Rust/cortex-m4-quick-start/target/thumbv7em-none-eabihf/debug/examples/hello"
> -> 8^done
> Failed to launch OpenOCD GDB Server: Timeout.

The openocd was called by cortex-debug and ran in the terminal without issue until closed by the timeout
[image]

If I copied the openocd running command used by cortex-debug to a terminal and ran, I could connect to the openocd server successfully and do the none-gui debugging with running gdb in another terminal
> gdb-multiarch -q target/thumbv7em-none-eabihf/debug/examples/hello 

[image]

launch.json i am using
{
            "cwd": "${workspaceFolder}/cortex-m4-quick-start",
            "executable": "./target/thumbv7em-none-eabihf/debug/examples/hello",
            "name": "hello OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "configFiles": [
                "interface/stlink-v2-1.cfg",
                "target/stm32f4x.cfg"
            ],
            "searchDir": [],
            "runToEntryPoint": "main",
            "showDevDebugOutput": "raw",
            "device": "STM32F446x",
},

haneefdm, 2023-05-24:
I think you are using an ancient version of openocd.

defyg, 2023-05-25:
Thanks, mine is 0.10.0

haneefdm, 2023-05-25:
Yes, that is ancient...

defyg, 2023-05-25:
compiled 0.11.0 and installed, works. Thanks a lot

closed 2023-05-25 by defyg

haneefdm, 2023-05-25:
Use a minimum 0.11.x or use the one packaged with STM32CubeIDE which is even better.