Marus/cortex-debug #887: Can't load bin files

embcla, 2023-05-24:
Hello

I've been reading the previous bugs, and you always suggest to use elf or hex instead of pure binaries.
I found out this possible instruction
`arm-none-eabi-objcopy -I binary -O ihex STM32F429ZI_NUCLEO_144.bin STM32F429ZI_NUCLEO_144.hex`
from [this issue](https://github.com/Marus/cortex-debug/issues/726)
but for me it doesn't work.

So this is the situation: I start with an elf, but then I strip it, calculate the crc and add its value to a certain location.
This is used by the program to verify itself on start.

so I'm perfectly able to run with the elf, but that is missing the crc which is causing other issues later on which I'd like to avoid.
What I'd like to do, is flash the bin.

But when I use the following configuration
```
		{
			"name": "project debug/release (openocd)",
            "cwd": "${workspaceRoot}/.build/${input:boardName}/${input:buildType}/",
			"executable": "./project-bootloader.elf",
			// "executable": "./project-bootloader-with-crc.bin",
			"loadFiles": [
				"./project-bootloader-with-crc.bin",
			],
			"symbolFiles": [
				{
					"file": "./project-bootloader.elf"
				}
			],
			"request": "launch",  // Change to "attach" for Sample B
			"type": "cortex-debug",
            "servertype": "external",
            "gdbTarget": "172.16.17.24:3333",
			"device": "STM32G0B1KCT6",
			"configFiles": [
				"interface/stlink.cfg",
				"target/stm32g0x.cfg"
			],
			"svdFile": "../../../.vscode/STM32G0B1.svd",
			"showDevDebugOutput": "raw"
		},
```
it errors out with
[image]

our openocd configuration can flash the same file just fine
`openocd -f project.cfg -c "program ${1} 0x8000000 verify" -c "sleep 200" -c "reset run" -c "exit"`

what do you suggest in this case?

haneefdm, 2023-05-24:
If gdb cannot recognize the file, not much we cannot do. We simply pass it along to gdb to figure out what to do. Try hex format.

Can you post the output from the Debug Console? Why are you not following the bug submission template?

closed 2023-05-24 by haneefdm

embcla, 2023-05-24:
Because when I use the suggestion and convert my bin to hex, I get this
[image]

```
Cortex-Debug: VSCode debugger extension version 1.10.0 git(cf1c296). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Finished reading symbols from objdump: Time: 598 ms
Finished reading symbols from nm: Time: 873 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2\n"
-> ~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/claudio/.vscode/extensions/marus25.cortex-debug-1.10.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/claudio/.vscode/extensions/marus25.cortex-debug-1.10.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "add-symbol-file \"/home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf\""
-> ~"add symbol table from file \"/home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf\"\n"
add symbol table from file "/home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf"
-> ~"(y or n) [answered Y; input not from terminal]\n"
(y or n) [answered Y; input not from terminal]
-> ~"Reading symbols from /home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf...\n"
Reading symbols from /home/claudio/firmware/quercus/Eichrecht/bootloader/.build/sample_d/debug/eichrecht-bootloader.elf...
-> 8^done
9-target-select extended-remote 172.16.17.24:3333
-> =thread-group-started,id="i1",pid="42000"
-> &"warning: No executable has been specified and target does not support\ndetermining executable automatically.  Try using the \"file\" command."
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
-> &"\n"

-> =thread-created,id="1",group-id="i1"
-> ~"0x08002fb8 in Metadata_GetCertificateSize (certType=METADATA_CERT__L1) at ../../../source/fw_update/metadata.c:91\n"
0x08002fb8 in Metadata_GetCertificateSize (certType=METADATA_CERT__L1) at ../../../source/fw_update/metadata.c:91
-> ~"91\t\t\tcase METADATA_CERT__L2:   memcpy(&size, FIELD_ADDR(l2_size),   FIELD_SIZE(l2_size));   break;\n"
91			case METADATA_CERT__L2:   memcpy(&size, FIELD_ADDR(l2_size),   FIELD_SIZE(l2_size));   break;
-> *stopped,frame={addr="0x08002fb8",func="Metadata_GetCertificateSize",args=[{name="certType",value="METADATA_CERT__L1"}],file="../../../source/fw_update/metadata.c",fullname="/home/claudio/firmware/quercus/Eichrecht/bootloader/source/fw_update/metadata.c",line="91",arch="arm"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 9^connected
10-interpreter-exec console "monitor reset halt"
-> @"Unable to match requested speed 2000 kHz, using 1800 kHz\n"
Unable to match requested speed 2000 kHz, using 1800 kHz
-> @"Unable to match requested speed 2000 kHz, using 1800 kHz\n"
Unable to match requested speed 2000 kHz, using 1800 kHz
-> @"[stm32g0x.cpu] halted due to debug-request, current mode: Thread \n"
[... 49 lines cut ...]
-> 21^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x08002fb8",func="Metadata_GetCertificateSize",args=[{name="certType",value="METADATA_CERT__L1"}],file="../../../source/fw_update/metadata.c",fullname="/home/claudio/firmware/quercus/Eichrecht/bootloader/source/fw_update/metadata.c",line="91",arch="arm"},state="stopped"}]
22-data-list-register-values N
-> 22^done,register-values=[{number="0",value="-1"},{number="1",value="-1"},{number="2",value="-1"},{number="3",value="-1"},{number="4",value="-1"},{number="5",value="-1"},{number="6",value="-1"},{number="7",value="-1"},{number="8",value="-1"},{number="9",value="-1"},{number="10",value="-1"},{number="11",value="-1"},{number="12",value="-1"},{number="13",value="0x200237fc"},{number="14",value="-1"},{number="15",value="0x8002fb8 <Metadata_GetCert [... cut ...]
-> 29^error,msg="Variable object not found"
30-var-create --thread 1 --frame 0 var_certType_256 * "certType"
-> 30^done,name="var_certType_256",numchild="0",value="METADATA_CERT__L1",type="enum MetadataCertificateType",thread-id="1",has_more="0"
31-var-update --thread 1 --frame 0 --all-values var_size_256
-> 31^error,msg="Variable object not found"
32-var-create --thread 1 --frame 0 var_size_256 * "size"
-> 32^done,name="var_size_256",numchild="0",value="69",type="uint16_t",thread-id="1",has_more="0"

```

haneefdm, 2023-05-24:
that is a gdb issue if any. We don't do any of that.

It is our bug if your gdb works from command-line. You see every command we send to gdb and its responses. We are only a front end to gdb. All real debugging happens via gdb

embcla, 2023-05-24:
Thank you, yes I understand, I'll take my file format questions elsewhere.

Regarding the "executable" parameter, can you tell me why it's always needed?
I was expecting that if loadFiles and symbolFiles are set, then executable is not needed but the plugin fails to start.
Also if I may, I'd suggest making loadFiles and symbolFiles the same type (either strings or objects), it's quite confusing in the beginning.

haneefdm, 2023-05-24:
"executable", is technically not needed. No idea what version you are using. Try the latest pre-release and it might not be needed anymore.

Again, you did not use the issue report template so I get the details.

embcla, 2023-05-24:
Sorry, I clicked on "new issue" before logging in, and when you do so and then login, the form does not have the template.
I'll follow it tomorrow