Marus/cortex-debug #893: Setting runToEntryPoint when Symbol Name is not Unique

ckidder, 2023-06-02:
I'm debugging some firmware and I've got it loading the symbols for the bootloader as well, so I can see what's going on if there are issues.  The issue here is that there are two main() functions, one for the BL and one for the app.  I don't see how to specify which main() function it halts at after reboot.  For now it's always picking the one I don't want.  My current workaround is to put a breakpoint at the main() I want to make sure it stops there.  Here's my configuration from launch.json:

```
        {
            "name": "App:ATAK:BL:Launch",
            "executable": "${workspaceFolder}/build/ATAK_DEBUG_BL.elf",
            "cwd": "${workspaceFolder}",
            "request": "launch",
            "type": "cortex-debug",
            "serverpath": "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "interface": "swd",
            "armToolchainPath": "C:/arm/bin",
            "device": "STM32H753II",
            "svdFile":"${workspaceFolder}/STM32H7x3.svd",
            "rtos": "FreeRTOS",
            "showDevDebugOutput":"parsed",
            "runToEntryPoint": "main",
            "loadFiles": [],
            "symbolFiles": [
                            //Comment the next line out if you don't want to load the bootloader symbols
                            //Or update it to the correct path. 
                            {"file" : "${workspaceFolder}/../segger_emload/build/PITBULL_BL.elf"}   
                            {"file" : "${workspaceFolder}/build/ATAK_DEBUG_BL.elf"},
                        ],
        },
```

I'm not sure if this should be a feature request or bug, so I figured I'd start here.  Changing the order of the symbolFiles does not seem to change which main() is selected as the runToEntryPoint.  Thank you for your help.

haneefdm, 2023-06-05:
Nope. I wish there was a workaround. If gdb cannot support this neither can we. You have a few workarounds.

1. Set a breakpoint on an address (in the breakpoints panel managed by VSCode). If your mains are in separate files, the you can also set the breakpoint manually but this breakpoint will be a permanent breakpoint instead of a temporary one.
2. Change your main functions so they are trampolines to unique main's and then use that function name in `runToEntryPoint`
3. Other methods ... as main is not a required function per C language. It can be anything but you have to modify the pre-startup 

```C
int mainApp(...) {
   ...
}

int main(...) {
    return mainApp(...);
}
```

closed 2023-06-05 by haneefdm

ckidder, 2023-06-08:
Thanks for the answer.