Marus/cortex-debug #895: empty runToEntryPoint halts MCU if liveWatch is enabled

lijon, 2023-06-05:
**Describe the bug**

With openocd 0.12.0 and ST-Link/V2-1 debug probe, setting `"runToEntryPoint": ""` works fine (device is not halted when launching debug session), unless `liveWatch` is enabled - then it always halts.

It's very distracting to have it halt on launch, because it then opens some random source file (where it happened to be executing) and need me to click the Continue button.

**Environment**

 - Cortex-Debug Version 1.11.2
 - OS: macOS
 - GDB version: (GNU Arm Embedded Toolchain 10.3-2021.07) 10.2.90.20210621-git
 - Compiler Version: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.07) 10.3.1 20210621 (release)

**`launch.json`**

```
"version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/myProgram.elf",
            "name": "Debug with OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "interface": "swd",
            "device": "STM32G071",
            "configFiles": [
                "interface/stlink-v2-1.cfg",
                "target/stm32g0x.cfg"
            ],
            "svdFile": "../STM32G0_svd_V1.3/STM32G071.svd",
            "searchDir": [],
            "runToEntryPoint": "",
            "showDevDebugOutput": "none",
            "preLaunchTask": "make all",
            // "liveWatch": {
            //     "enabled": true,
            //     "samplerPerSecond": 2
            // },
            "postRestartCommands": [
                "load",
                "monitor reset run"
            ]
        }
    ]
}
```

**Attach text from `Debug Console`**

```
Cortex-Debug: VSCode debugger extension version 1.11.2 git(441af06). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/lijon/Coding/STM32/PolyVoiceBoardG071/build/PolyVoiceBoard.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/lijon/Coding/STM32/PolyVoiceBoardG071/build/PolyVoiceBoard.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /Users/lijon/Coding/STM32/PolyVoiceBoardG071 -f /Users/lijon/.vscode/extensions/marus25.cortex-debug-1.11.2/support/openocd-helpers.tcl -f interface/stlink-v2-1.cfg -f target/stm32g0x.cfg -c CDLiveWatchSetup
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 28 ms
Finished reading symbols from nm: Time: 32 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.07) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
[... 118 lines cut ...]
TIM1_BRK_UP_TRG_COM_IRQHandler () at Src/stm32g0xx_it.c:181
-> ~"181\t    if (LL_TIM_IsActiveFlag_UPDATE(TIM1)) {\n"
181	    if (LL_TIM_IsActiveFlag_UPDATE(TIM1)) {
-> *stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0x20001be0",func="TIM1_BRK_UP_TRG_COM_IRQHandler",args=[],file="Src/stm32g0xx_it.c",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Src/stm32g0xx_it.c",line="181",arch="armv6s-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
16-thread-list-ids
-> 16^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
17-thread-info 1
-> 17^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x20001be0",func="TIM1_BRK_UP_TRG_COM_IRQHandler",args=[],file="Src/stm32g0xx_it.c",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Src/stm32g0xx_it.c",line="181",arch="armv6s-m"},state="stopped"}]
18-stack-info-depth --thread 1 1000
-> 18^done,depth="5"
19-stack-list-frames --thread 1 0 4
-> 19^done,stack=[frame={level="0",addr="0x20001be0",func="TIM1_BRK_UP_TRG_COM_IRQHandler",file="Src/stm32g0xx_it.c",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Src/stm32g0xx_it.c",line="181",arch="armv6s-m"},frame={level="1",addr="0xfffffff9",func="<signal handler called>"},frame={level="2",addr="0x08002fb8",func="LL_GPIO_IsInputPinSet",file="Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_gpio.h",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_gpio.h",line="778",arch="armv6s-m"},frame={level="3",addr="0x08002fb8",func="main_loop",file="Src/main.c",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Src/main.c",line="2137",arch="armv6s-m"},frame={level="4",addr="0x080040b2",func="main",file="Src/main.c",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Src/main.c",line="2649",arch="armv6s-m"}]
20-data-read-memory-bytes "0x40020000" 144
-> 20^done,memory=[{begin="0x40020000",offset="0x00000000",end="0x40020090",contents="0700700000000000b72500007800000008740040102b002000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000a700000020000000244400406c300020000000009100000000000000284400400c3100200000000000000000000000000000000000000000"}]
LiveGDB: -> ~"0x20001be0 in LL_TIM_IsActiveFlag_UPDATE (TIMx=0x40012c00) at Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_tim.h:4278\n"
LiveGDB: 0x20001be0 in LL_TIM_IsActiveFlag_UPDATE (TIMx=0x40012c00) at Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_tim.h:4278
LiveGDB: -> ~"4278\t  return ((READ_BIT(TIMx->SR, TIM_SR_UIF) == (TIM_SR_UIF)) ? 1UL : 0UL);\n"
LiveGDB: 4278	  return ((READ_BIT(TIMx->SR, TIM_SR_UIF) == (TIM_SR_UIF)) ? 1UL : 0UL);
LiveGDB: -> *stopped,frame={addr="0x20001be0",func="LL_TIM_IsActiveFlag_UPDATE",args=[{name="TIMx",value="0x40012c00"}],file="Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_tim.h",fullname="/Users/lijon/Coding/STM32/PolyVoiceBoardG071/Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_tim.h",line="4278",arch="armv6s-m"},thread-id="1",stopped-threads="all"
LiveGDB: mi2.status = stopped
LiveGDB: Program stopped, probably due to a reset and/or halt issued by debugger
LiveGDB: -> 12^connected
Started live-monitor-gdb session
```

haneefdm, 2023-06-05:
Thanks for reporting this issue. Yes, this is an issue and is fixed in the next update.

Even though we ask for the program NOT halt on connect, almost all gdb-servers are halting the program on a connect. This happens with STLink gdb server as well and how most servers are written.

```
LiveGDB: 11-interpreter-exec console "set remote interrupt-on-connect off"
LiveGDB: -> 11^done
LiveGDB: 12-target-select extended-remote localhost:50000
-> @"accepting 'gdb' connection on tcp/50000\n"
accepting 'gdb' connection on tcp/50000
-> ~"\nProgram"

Program
-> ~" received signal SIGINT, Interrupt.\n"
 received signal SIGINT, Interrupt.
```

haneefdm, 2023-06-06:
If you wish to try out the fix, you can do so from here...

https://github.com/Marus/cortex-debug/releases/tag/v1.11.3-pre5

This is a pre-release

haneefdm, 2023-06-08:
The release is now also in the marketplace...

closed 2023-06-08 by haneefdm

lijon, 2023-06-08:
Thanks! Works fine now.