Marus/cortex-debug #905: Disassembly view not working when dragged to the left or the right

schwaerz, 2023-06-27:
**Describe the bug**
We are developing our own GDB server for simulating our HW.
When opening the disassembly window, everything works just fine. We can single-step, step over, etc. Just like what you would expect.

When running on Linux everythin works fine.

However, on Windows (11), when I  try to create a split view (source code and disassebly) in VSCode, it fails if I move the disassebly window to the left or the right.

The window will look as as follows:
[image]

There is an error message popping up, too:
[image]

I can however get it working if I keep the disassembly Window where it is, but move the source code Window to either the left or the right.
Reproducible on a different PC with different User (running Windows 10), too.

Expected of course would be that it does not matter where I move the disassebly Window.

[comment]:

 - Cortex-Debug Version: 1.12.0
 - OS: Windows 11 / Windows 10
 - GDB Version: GNU gdb (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.1.90.20221210-git
 - Compiler Toolchain Version: armclang 6.20

**Please include `launch.json`**

Please note that we are reading some of the values from the `settings.json` (see below. Please let me known whether / which additional values you need. I cannot provide the whole file as of security reasons.
```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "showDevDebugOutput": "none",
            "name": "Debug sim",
            "type": "cortex-debug",
            "request": "launch",
            "executable": "${config:project_folder}/build/${config:project_name}.axf",
            "loadFiles": [
                "${config:project_folder}/build/1st.hex", 
                "${config:project_folder}/build/2nd.hex"
            ],
            "symbolFiles": [
                {
                    "file": "${config:project_folder}/build/${config:project_name}.axf"
                }
            ],
            "servertype": "external",
            "cwd": "${workspaceRoot}",
            "device":"${config:cortexDevice}",
            "gdbTarget": "localhost:3333",
            "gdbPath": "arm-none-eabi-gdb",
            "preLaunchTask": "start gdb background",
            "runToEntryPoint": "main",
            "svdFile": "${config:svdFile}"
        }
    ]
}
```

settings.json (extract):
```json
{
    // project specific settings
    "project_folder":  "project_template",
    "project_name":    "<template name>",
    "derivative":  "<derivative name>",
    "svdFile":     "<path to svd>",
    // environment settings IG32
    "cortexMcpu":  "cortex-m33",
    "cortexDevice": "Cortex M33",
    // environment settings Lean
    //"cortexMcpu":  "cortex-m0",
    //"cortexDevice": "Cortex M0",
    // environment settings debug
    "memview.showMemoryPanel": true,
    "cortex-debug.registerUseNaturalFormat": false,
    "cortex-debug.variableUseNaturalFormat": false,
    "code-runner.customCommand": ". ",
    "files.associations": {
        "core_armv8mml.h": "c",
        "stdint.h": "c",
        "__config": "c",
        "arm_cmse.h": "c",
        "test.h": "c",
        "cstdint": "c"
    }
}
```

**Attach text from `Debug Console`**

The following I captured from the debug console after
1. clearing the console
2. dragging the Window

I used `"showDevDebugOutput": "none"`

After that I changed it to `"showDevDebugOutput": "raw"`. No more lines added in the output after dragging the window. Also after changing it back to "none" did not produce that output again?!
```
Debug-19: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x100061aa","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":19}
Debug-19: Dequeuing...
Bad instruction cache. Could not find address 268460356 that should have been found
```

As this error is purely related to where you drag the window in VS Code I strongly believe it is related to either the VSCode disassembly window implementation, or cortex-debug. I did not find anything about this issue wrt to the VSCode disassembly view. Therefore my first shot is the cortex-debug plugin.

jnz86, 2023-08-15:
I had a second so I tried to replicate this under Windows 11. Works fine for me, tried moving the window and stepping around, moving again, closing and re-opening. All works as expected.

haneefdm, 2023-08-29:
We do not manage ANY of the editor windows in VSCode -- not your source code or the disassembly view. It is done by VSCode proper. Please submit an issue with the vscode main repo.

The disassembly failure is specific to YOUR executable and symbol table. I cannot debug it without some help from you.

I have to close this issue if there is nothing further I can do. You can also set `"showDevDebugOutput": "vscode"` to see VSCode request/response sequences. It is possible that VSCode is asking for addresses that are invalid or we are doing something wrong. Disassembly is not perfect by any means and is a guessing game if you start with random addresses. It is more reliable when you start with known addresses (like start of functions, etc.)

closed 2025-01-17 by haneefdm