Marus/cortex-debug #906: GDB don't want to connect to OpenOCD server

Bricklou, 2023-07-03:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
I'm playing a bit with a Raspberry pico and a second Pico (running picoprobe) for debugging.
I tried 2 projets: 1 in C using the official examples, the second in rust.
In both cases, i ran into the same issue: gdb can't connect to the OpenOCD server (you can find all the errors below)

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare a projet like explained in the official [Getting started guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)
2. Start debug session
2. Wait a bit
4. See issue

**Expected behavior**

GDB should properly connect to the OpenOCD server and allow the debugging of the Pico board.

**Screenshots**
[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version: v1.12.0
 - OS: Archlinux 6.3.9-arch1-1
 - GDB Version: `arm-none-eabi-gdb` 13.1
 - Compiler Toolchain Version: `arm-none-eabi-gcc` 13.1
 - Rust compiler version: `rustc` 1.70.0

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```json
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [
    {
      "name": "Debug hello_world",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceFolder}/target/thumbv6m-none-eabi/debug/hello_world",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      // This may need to be arm-none-eabi-gdb depending on your system
      "gdbPath": "arm-none-eabi-gdb",
      "device": "RP2040",
      "configFiles": ["interface/cmsis-dap.cfg", "target/rp2040.cfg"],
      "svdFile": "${workspaceFolder}/.vscode/rp2040.svd",
      "runToEntryPoint": "main",
      // Work around for stopping at main on restart
      "postRestartCommands": ["break main", "continue"],

      "preLaunchTask": "Build binary",
      // If you have the Pico SDK installed, allows the
      // display of the RP2040 CPU registers in VS Code
      "showDevDebugOutput": "raw"
    }
  ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/louis/Documents/programmation/raspberry/pico/projects-rust/pico7segment/target/thumbv6m-none-eabi/debug/hello_world
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/louis/Documents/programmation/raspberry/pico/projects-rust/pico7segment/target/thumbv6m-none-eabi/debug/hello_world
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/louis/Documents/programmation/raspberry/pico/projects-rust/pico7segment -f /home/louis/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 139 ms
Finished reading symbols from nm: Time: 319 ms
-> =thread-group-added,id="i1"
-> =cmd-param-changed,param="mi-async",value="on"
Error: Unable to start GDB even after 5 seconds or it couldn't even start Make sure you can start gdb from the command-line and run any command like "echo hello".
    If you cannot, it is most likely because "libncurses" or "python" is not installed. Some GDBs require these
Failed to generate gdb commands: Error: Could not start gdb, no response from gdb
Error: Could not start gdb, no response from gdb
    at Timeout._onTimeout (/home/louis/.vscode/extensions/marus25.cortex-debug-1.12.0/dist/debugadapter.js:2:132433)
    at listOnTimeout (node:internal/timers:559:17)
    at process.processTimers (node:internal/timers:502:7)
```

haneefdm, 2023-07-03:
Well, does your gdb even work? We are trying to tell you it doesn't. Run gdb from the command line If you can't neither can we.

Bricklou, 2023-07-03:
I can assure you that gdb work fine from the command line using the commands below:

```sh
openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl

gdb path/to/my_program
# or
gdb -tui path/to/my_program
```

haneefdm, 2023-07-03:
We have shown the exact command line we have to use. Did you try that. OpenOCD does not seem to be the issue here.

The first thing we do is to try and determine the gdb-verson. Appears to get no response. So, we cannot proceed. We print  EVERY command sent to gdb and its response.

haneefdm, 2023-07-04:
You are not even running the right gdb. Please use the same gdb you asked us to use. As you can see it is `arm-none-eabi-gdb` and NOT just plain `gdb`.

Bricklou, 2023-07-04:
Oh yes, nevermind, I wrote the wrong command in my post above.

And I just saw that the openocd logs are missing, I'll send them as soon as possible.

Bricklou, 2023-07-04:
Here are my OpenOCD logs:
```
[2023-07-04T05:19:09.901Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/louis/Documents/programmation/raspberry/pico/projects-rust/pico7segment -f /home/louis/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at the low speed (100 kHz)
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6616408432F622A
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 50000
Info : Listening on port 50000 for gdb connections
Info : starting gdb server for rp2040.core1 on 50001
Error: couldn't bind gdb to socket on port 50001: Address already in use
[2023-07-04T05:19:17.017Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

I also checked the used ports, and there are none:
```sh
$ sudo netstat -pnltu | grep  -i '50001'

```

haneefdm, 2023-07-04:
Please run the commands manually as shown in the Debug Console. In this case the EXACT version of gdb. Not some random gdb. Why not copy-paste the command line shown into a terminal/shell and then post the output of that? 

I am closing this as you are not cooperating.

closed 2023-07-04 by haneefdm