Marus/cortex-debug #907: Support for custom path to `nm` and / or custom arguments to `objdump`

schwaerz, 2023-07-05:
**Is your feature request related to a problem? Please describe.**
We are running some debug sessions on Ubuntu 22.04, where we cannot simply use the arm-none-eabi-gdb as it would require Python 3.8, which is not available there using the package manager and also very old and should no longer be used.

Therefore we switched to using `gdb-multiarch`. We found out, however, that we need to also explicitly specify the path to the `objdump` tool in order to be able to start the debug session.

Now, disassembly did not work. That is because objdump was not able to detect the architecture by default. So we did...
```json
            "gdbPath": "gdb-multiarch",
            "objdumpPath": "objdump -marm",
```

Disassebly works now. But - I assume - as the `nm` does not support `-marm` the disassebly view will be lacking some details, i.e. the intermixed source code.

To be able to work around this, we'd need to be either
* Able to specify custom parameters for `objcopy`
* Or be able to specify a custom path to `nm`

Best would be to have both, of course :)

haneefdm, 2023-07-18:
Also, please use wrapper scripts if needed or install the right binaries. We are NOT going to add custom arguments for everything we do.

closed 2023-07-18 by haneefdm

schwaerz, 2023-07-18:
We switched to `binutils-multiarch`. This worked for us.