Marus/cortex-debug #908: Cannot locate Disassembly View

MucaGinger, 2023-07-06:
I'm trying to open disassembly view where it should appear as `Cortex-Debug: Disassembly View` after searching for it in the VS Code command palette. I get the following when I search for it:

[image]

I can only find _this_ disassembly view but I don't think this is provided as part of this extension.

[image]

haneefdm, 2023-07-06:
Yes, it is provided as part of this extension. But just like the Variable/Watch/Stack windows, all GUI elements are controlled by VSCode (Microsoft). If you are paused somewhere, you can right-click on the stack frames and get the disassembly view.

In short, we provide the disassembly data when VSCode asks for it and it asks us for the data when you ask for it. We provide the data for many of the debug panels. We don't manage them. How VSCode works so users across different debuggers have a consistent interface. The data you are seeing is in the disassembly view is coming from us -- only because VSCode requested it.

https://devzone.nordicsemi.com/f/nordic-q-a/99249/disassembly-mixed-view-in-vs-code-with-nrf-connect
https://devblogs.microsoft.com/cppblog/visual-studio-code-c-july-2021-update-disassembly-view-macro-expansion-and-windows-arm64-debugging/

Please google/read MS docs on this.

MucaGinger, 2023-07-06:
Okay, good to know.

I know this other question is out of the scope but can you also tell me what can I do about the Cortex Registers View? I can't get it to work, this is what I see:

[image]

As the hint suggests I have added the `liveWatch` property to `launch.json` file however without change:

`
"liveWatch": {
    "enabled": true,
    "samplesPerSecond": 4
}
`

haneefdm, 2023-07-06:
Yes, please search our documentation and issues list. It is even pinned to the top of the issues page.

https://github.com/Marus/cortex-debug/issues

Unless you add some variables/expressions to the Live Watch window, and it is enabled, of course, you won't see anything. Just like Microsoft's Watch window. If you don't add anything to watch, how would we know what to watch. Note that Live Watch only works with some gdb servers and some HW.

I am closing this issue as I haven't seen an issue yet.

closed 2023-07-06 by haneefdm