Marus/cortex-debug #912: Very slow debugging (stepping/breakpoints) on Linux with ST-Linkv3 + STM32

victorallume, 2023-07-20:
**Describe the bug**
I'm debugging a project targeting an STM32H7, using an STLink V3. Debugging via cortex-debug/VSCode is extremely slow i.e. stepping from one line to the next takes ~10-20 seconds (waiting for a breakpoint is similarly slow. Doing the same things in GDB or STM32CubeIDE is almost instant. I'm running on PopOS 22.04.

Any pointers to where to start working out how to resolve this?

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Wait until debugger has stopped execution at entry point.
3. step to next line of code.
4. Takes ~20s to get to next line.

**Expected behavior**
Gets to next line in less than 1s.

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) 1.12.0
 - OS: PopOS 22.04 LTS
 - GDB Version: 12.1
 - Compiler Toolchain Version: [e.g. arn-none-eabi V 11.1]

**Please include `launch.json`**

```
{
            "cwd": "${workspaceFolder}",
            "executable": ".../firmware.elf",
            "name": "Debug with ST-Link",
            "request": "launch",
            "type": "cortex-debug",
            // "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
            "servertype": "stutil",
            "serverArgs": ["--connect-under-reset", "--freq=24M"],
            "v1": false,
            "preLaunchCommands": [
                "set substitute-path /code ${workspaceFolder}"
            ]
        },
    ```

**Attach text from `Debug Console`**

haneefdm, 2023-07-21:
* You did not enable/attach any kind of logging. See debug console for details `"showDevDebugOutput": "raw"`. You can also enable timestamps to see what operation is taking so long. See docs
* Why use `stutil`? I don't support that. Maybe someone from the community can. Use either ST's own server or latest OpenOCD. If you are using STM32CubeIDE you are certainly not using `stutil`

victorallume, 2023-07-25:
Thanks; I switched to using `stlink` and now it works much more responsively. Changes to my `launch.json`:
```
            "servertype": "stlink",
            "serverpath": "/opt/st/stm32cubeide_1.12.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.500.202301161003/tools/bin/ST-LINK_gdbserver",
            "stm32cubeprogrammer": "PATH/TO/bin/STM32CubeProgrammer/bin",
```

closed 2023-07-25 by victorallume