Marus/cortex-debug #914: breakpoints not hit

MS-CZ, 2023-07-25:
Type: <b>Bug</b>

Hi,
when I run Cortex Debug and debug app for STM32L412 Nucleo breakpoints that are set before and even during debug session are always greyed out. Step over and Step into works flawlesly but breakpoints are not being hit at all. I do work with OpenOCD and all seems to be updated to the latest version.
Does this configuration works or I have to use ST link instead of OpenOCD?

Extension version: 1.12.0
VS Code version: Code 1.80.1 (74f6148eb9ea00507ec113ec51c489d6ffb4b771, 2023-07-12T17:22:07.651Z)
OS version: Windows_NT x64 10.0.19044
Modes:

MS-CZ, 2023-07-25:
here is debug console output and launch.json

Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump.exe --syms -C -h -w C:/Users/all/Desktop/datasheet, docs/new part/Nucleo STM32-L412-RB-P (STM32L412RBT6PU)/NUCLEO-L412RB-P_test/build/NucleoL412.elf
Reading symbols from arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users/all/Desktop/TACO_data/datasheet, docs/new part/Nucleo STM32-L412-RB-P (STM32L412RBT6PU)/NUCLEO-L412RB-P_test/build/NucleoL412.elf
Launching GDB: arm-none-eabi-gdb.exe -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Users\\all\\Desktop\\datasheet, docs\\new part\\Nucleo STM32-L412-RB-P (STM32L412RBT6PU)\\NUCLEO-L412RB-P_test" -f "c:/Users/all/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32l4x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd.exe
Finished reading symbols from objdump: Time: 124 ms
Finished reading symbols from nm: Time: 129 ms
Output radix now set to decimal 10, hex a, octal 12.
Input radix now set to decimal 10, hex a, octal 12.
0x08000b22 in HAL_Delay (Delay=Delay@entry=500) at Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal.c:414
414	  while ((HAL_GetTick() - tickstart) < wait)
Program stopped, probably due to a reset and/or halt issued by debugger
Unable to match requested speed 500 kHz, using 480 kHz
Unable to match requested speed 500 kHz, using 480 kHz
[stm32l4x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08001ad0 msp: 0x2000a000
Unable to match requested speed 500 kHz, using 480 kHz
Unable to match requested speed 500 kHz, using 480 kHz
[stm32l4x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08001ad0 msp: 0x2000a000
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.

Note: automatically using hardware breakpoints for read-only addresses.

Temporary breakpoint 1, main () at Core/Src/main.c:75
75	  HAL_Init();

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug test",
            "cwd": "${workspaceFolder}", 
			"executable": "${workspaceFolder}/build/NucleoL412.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "openocd",
            "device": "STM32L412RB",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32l4x.cfg"
            ],
            "svdFile": ".vscode/STM32L4x2.svd"
        }
    ],
    "logging": {
        "engineLogging": true,
    }
}

haneefdm, 2023-08-29:
First of all, you need to enable our debug output to see what is going on. See instructions in Debug Console.

Second, you can observe the Debug Console and see what happens when you set a breakpoint. If GDB cannot set the breakpoint like you asked, then VSCode will grey it out.

haneefdm, 2023-11-21:
No activity in a long time.

closed 2023-11-21 by haneefdm