Marus/cortex-debug #915: overrideResetCommands and/or overrideRestartCommands do not alter the reset calls in the launch commands when server type is "external"

mmerlo, 2023-07-26:
**Describe the bug**
overrideResetCommands and/or overrideRestartCommands do not alter the reset calls in the launch commands when server type is "external". Instead the reset calls when launching an external server are hardcoded as "monitor reset halt". 

**To Reproduce**
Steps to reproduce the behavior:
1. Make an `external` server `launch.json` configuration
2. override the reset and/or restart commands in launch.json
3. Start debug session
4. Review logs: `monitor reset halt` will be called instead of overridden commands.

**Expected behavior**
Expect overrideResetCommands and/or overrideRestartCommands to override all reset calls including those called during launch regardless of server type.

**`launch.json`**
```
    "configurations": [
        {
            "name": "Debug: cortex-debug + ST-LINK Launch",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/build/program.elf",
            "request": "launch",
            "preLaunchTask": "Build",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "vscode",
            "showDevDebugTimestamps": false,
            "servertype": "external",
            "gdbTarget": "host.docker.internal:61234",
            "svdFile": "${workspaceFolder}/STM32F0x2.svd",
            "overrideRestartCommands": [
                "monitor reset"
            ],
        },
```

**`Code snippet from external.ts`**
```
public launchCommands(): string[] {
        const commands = [
            ...genDownloadCommands(this.args, ['interpreter-exec console "monitor reset halt"']),
            'interpreter-exec console "monitor reset halt"'
        ];
```

**Workaround**
Override launch commands in launch.json with:
```
            "overrideLaunchCommands": [
                "monitor reset",
                "-target-download",
                "monitor reset"
            ]
```
This works, but is not the expected behavior of the overrideRestartCommand and requires the user to know the entire launch sequence just to override the reset commands. 

**Related issues**
#855 
#688

haneefdm, 2023-07-26:
Wrong expectation and we cannot change that. Not just for external but for any gdb-server. You do need to specify all your overrides. You are also wrong about repeating the Launch command for other overrides. We don't do that because the launch typically does a bit more than the reset/restart.

closed 2023-07-26 by haneefdm

mmerlo, 2023-07-27:
You're right, in retrospect my (wrong) expectation was for all gdb-servers (not just external). I expected the overrideRestartCommand to override all attempts to restart/reset even the restarts/resets that are part of a launch sequence. I suspect there must be use-cases where one would want to override the general reset/restart commands but leave the reset/restart commands in launch unchanged.