Marus/cortex-debug #917: JLinkGDBServer fails to launch due to a timeout

8bitter, 2023-08-03:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
The JLinkGDBServer can fail to launch due to a hardwired timeout. Increasing the timeout time eliminates the problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Wait for the JLinkGDBServer to launch
3. Launch fails to due to a timeout when it is waiting for JLinkGDBServer to acknowledge that it is waiting for commands

**Expected behavior**

[comment]: <> JLinkGDBServer launches successfully.

**Screenshots**

[image]

[comment]: <> If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension) 1.12.0
 - OS: Mac Ventura 13.4.1
 - GDB Version: SEGGER J-Link GDB Server V7.90 Command Line Version
 - Compiler Toolchain Version: Arm GNU Toolchain 12.2.MPACBTI-Rel1 (Build arm-12-mpacbti.34)
 - 
**Please include `launch.json`**

}

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Unit Tests",
            "type": "lldb",
            "request": "launch",
            "program": "${workspaceFolder}/fw/build_vsc/tests",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "lldb"
        },
        {
            "name": "fw Debug",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/fw/build/fw.axf",
            "request": "launch",
            "type": "cortex-debug",
            // don't pause "runToEntryPoint": "main",
            "servertype": "jlink",
            "device": "AMA4B2KP-KXR",
            "showDevDebugOutput": "raw",
            "rtos": "FreeRTOS",
            "svdFile": "${workspaceFolder}/ambiq_sdk/pack/SVD/apollo4p.svd",
            "serverArgs": [
                "-notimeout"
            ],
        }
    ]
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/xxx/mojo/repos/pym_lens/fw/build/fw.axf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/xxx/mojo/repos/pym_lens/fw/build/fw.axf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device AMA4B2KP-KXR -rtos GDBServer/RTOSPlugin_FreeRTOS.dylib -notimeout
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
Finished reading symbols from objdump: Time: 103 ms
Finished reading symbols from nm: Time: 1775 ms
Error: Unable to start GDB even after 5 seconds or it couldn't even start Make sure you can start gdb from the command-line and run any command like "echo hello".
    If you cannot, it is most likely because "libncurses" or "python" is not installed. Some GDBs require these
Failed to generate gdb commands: Error: Could not start gdb, no response from gdb
Error: Could not start gdb, no response from gdb
    at Timeout._onTimeout (/Users/xxx/.vscode/extensions/marus25.cortex-debug-1.12.0/dist/debugadapter.js:2:132433)
    at listOnTimeout (node:internal/timers:559:17)
    at process.processTimers (node:internal/timers:502:7)
```

**Additional context**
Add any other context about the problem here.

The hardwired timeout value is in src/backend/mi2/mi2.ts near line 90:
```
                let timeout = setTimeout(() => {
                    this.gdbStartError();
                    setTimeout(() => {
                        reject(new Error('Could not start gdb, no response from gdb'));
                    }, 10);
                    timeout = undefined;
                }, 5000);
```

Changing 5000 to a bigger value fixes the problem. Ideally, this could be a VSCode configurable parameter.

haneefdm, 2023-08-03:
5 seconds is an eternity. Are you able to run GDB the way you set it up for us from the command line? Please run the command manually in a terminal and report back what you see. We print every command exactly the way we run it in debug console.

> arm-none-eabi-gdb -q --interpreter=mi2

If we are not getting responses from gdb in a timely fashion, it will be a very poor debug experience.

haneefdm, 2023-08-03:
That timeout didn't used to exist. Not even zero. We added that because of stupid Linux gdb distributions with missing python or curses or some other library and it would just hang without any messages. So, no we are not increasing the timeout and rather we need to find your root cause.

haneefdm, 2023-08-03:
Please edit this title. JLink was not the problem as you yourself say.

haneefdm, 2023-08-29:
No response so closing this...

closed 2023-08-29 by haneefdm