Marus/cortex-debug #918: Failed to launch OpenOCD GDB Server: Timeout

Tracker647, 2023-08-06:
I'm using stm32f103c8t6 and st-linkv2, and encounter this.
the executable axf file is from keil5, I switch to stm32cube to generate makefile excutable elf file to debug and still prompt it.
[image]

 - Cortex-Debug Version : v1.12.0
 - OS: Windows 11
 - GDB Version:GNU gdb (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 13.2.90.20230627-git
 - Compiler Toolchain Version: Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)

my launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "openocd",
            "configFiles": [
                "interface/stlink-v2.cfg",
                "target/stm32f1x.cfg"
            ],
            "executable":"MDK-ARM/mdk_arm_test/mdk_arm_test.axf",
            "runToMain": true,
            "svdFile": "D:/Arm/Packs/Keil/STM32F1xx_DFP/2.2.0/SVD/STM32F103xx.svd",
            "showDevDebugOutput": "raw"
        }
    ]
}
```

Debug Console
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from D:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/bin/arm-none-eabi-objdump.exe --syms -C -h -w D:/A_EdiskOffline/myProgramming/MCU/STM32/mdk_arm_test/MDK-ARM/mdk_arm_test/mdk_arm_test.axf
Reading symbols from d:/program files (x86)/arm gnu toolchain arm-none-eabi/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p D:/A_EdiskOffline/myProgramming/MCU/STM32/mdk_arm_test/MDK-ARM/mdk_arm_test/mdk_arm_test.axf
Launching GDB: "D:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2
1-gdb-version
Launching gdb-server: "D:\\openocd-0.10.0\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:\\A_EdiskOffline\\myProgramming\\MCU\\STM32\\mdk_arm_test" -f "c:/Users/aaron/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl" -f interface/stlink-v2.cfg -f target/stm32f1x.cfg
    Please check TERMINAL tab (gdb-server) for output from D:\openocd-0.10.0\bin\openocd.exe
Finished reading symbols from objdump: Time: 118 ms
Finished reading symbols from nm: Time: 109 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 13.2.90.20230627-git\n"
-> ~"Copyright (C) 2023 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://bugs.linaro.org/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <"
-> ~"http://www.gnu.org/software/gdb/documentation/>.\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/aaron/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/aaron/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "D:/A_EdiskOffline/myProgramming/MCU/STM32/mdk_arm_test/MDK-ARM/mdk_arm_test/mdk_arm_test.axf"
-> &"warning: Loadable section \"RW_IRAM1\" outside of ELF segments\n  in "
warning: Loadable section "RW_IRAM1" outside of ELF segments
  in
-> &"D:\\A_EdiskOffline\\myProgramming\\MCU\\STM32\\mdk_arm_test\\MDK-ARM\\mdk_arm_test\\mdk_arm_test.axf\n"
D:\A_EdiskOffline\myProgramming\MCU\STM32\mdk_arm_test\MDK-ARM\mdk_arm_test\mdk_arm_test.axf
Failed to launch OpenOCD GDB Server: Timeout.

```

Terminal output:
```
"D:\\openocd-0.10.0\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:\\A_EdiskOffline\\myProgramming\\MCU\\STM32\\mdk_arm_test" -f "c:/Users/aaron/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl" -f interface/stlink-v2.cfg -f target/stm32f1x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v40 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.230587
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
[2023-08-06T02:02:22.270Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

haneefdm, 2023-08-06:
First, use a more uptodate version of openocd. I thin 0.10.0 is super old (2017)

You can see the exact command line for OpenOCD. Please run that after updating openocd and make sure it runs without crashing by executing the same command we are using. You can do this in a Windows command prompt.

Tracker647, 2023-08-06:
> First, use a more uptodate version of openocd. I thin 0.10.0 is super old (2017)
> 
> You can see the exact command line for OpenOCD. Please run that after updating openocd and make sure it runs without crashing by executing the same command we are using. You can do this in a Windows command prompt.

I switch to xpack-openocd0.12.0 and redebugging my code, it works, thanks.
[image]

closed 2023-08-06 by Tracker647