Marus/cortex-debug #919: GDB 13.2 supplied with ARM GNU Toolchain 12.3 dont work correctly

dsabala, 2023-08-06:
GDB 13.2 supplied with ARM GNU Toolchain 12.3  does not work. 

I am using DISC1 STM32F407 eval board.
VSCode GUI does not respond on debug buttons like start/step. 
Changing `runToEntryPoint` from `Reset_Handler` to `main.c` makes part of debugging available. 
Program is flashed correctly, after clicking debug stop button it does start to execute.
Fallback to ARM GNU Toolchain 11.3 or 10.3 solves the problem.
I tried both running debug in Dev Containers and natively, the result is same.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile example code for Cortex M4 with the newest ARM GNU Toolchain downloaded from ARM website.
2. Run debug
3. Click start / step / step through, nothing happen Reset_Handler line is highlighted.

**Expected behavior**
Debugging works.

**Logs**
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/dsabala/github/stm32f407-example/build/stm32f407-example
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/dsabala/github/stm32f407-example/build/stm32f407-example
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F407VG -rtttelnetport 19021 -noif
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServerCLExe
Finished reading symbols from objdump: Time: 64 ms
Finished reading symbols from nm: Time: 75 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 13.2.90.20230627-git\n"
-> ~"Copyright (C) 2023 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-pc-linux-gnu --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://bugs.linaro.org/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <"
-> ~"http://www.gnu.org/software/gdb/documentation/>.\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/dsabala/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/dsabala/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "/home/dsabala/github/stm32f407-example/build/stm32f407-example"
-> 10^done
11-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
[... 287 lines cut ...]
Read register 'r1' (4 bytes) from hardware: 0xD8FF0120
Read register 'r2' (4 bytes) from hardware: 0x20000000
Read register 'r3' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r4' (4 bytes) from hardware: 0xE8030000
Read register 'r5' (4 bytes) from hardware: 0x00000000
Read register 'r6' (4 bytes) from hardware: 0x00000000
Read register 'r7' (4 bytes) from hardware: 0xF8FF0120
Read register 'r8' (4 bytes) from hardware: 0x00000000
Read register 'r9' (4 bytes) from hardware: 0x00000000
Read register 'r10' (4 bytes) from hardware: 0x00000000
Read register 'r11' (4 bytes) from hardware: 0x00000000
Read register 'r12' (4 bytes) from hardware: 0x00000000
Read register 'sp' (4 bytes) from hardware: 0xE8FF0120
Read register 'lr' (4 bytes) from hardware: 0x355B0008
Read register 'pc' (4 bytes) from hardware: 0xEC290008
Read register 'xpsr' (4 bytes) from hardware: 0x00000041
Removing breakpoint @ address 0x080029EC, Size = 2
Read 4 bytes @ address 0x080029EC (Data = 0xFFD8F7FF)
Reading register 'msp' = 0x2001FFE8
Reading register 'psp' = 0x00000000
Reading 64 bytes @ address 0x2001FFC0
Read 4 bytes @ address 0x08005B34 (Data = 0x681B4B03)
Reading 64 bytes @ address 0x08005B00

```

haneefdm, 2023-08-06:
40-break-insert "/home/dsabala/github/stm32f407-example/src/main.c:13"
-> 40^error,msg="No source file named /home/dsabala/github/stm32f407-example/src/main.c."

That message is from GDB. You have to figure out why that is. We have nothing to do with that. You can launch everything command-line and see why you can't set the breakpoint there.

haneefdm, 2023-08-06:
Btw, that was a break point you set, perhaps via the VSCode editor. Not us. "runToEntryPoint" breakpoints look different.

dsabala, 2023-08-06:
Thank you for this information. Closes this issue. If I remember, I will add information here that helped to solve this problem

closed 2023-08-06 by dsabala