Marus/cortex-debug #920: Disassembly can not show source code

lee8871, 2023-08-08:
Disassembly can not show source code .

## Phenomenon
I am using `xpack-riscv-none-elf-gcc-12.2.0-3` and do some embedding developing.
Start debuging and open disassembly:
[image]
Than, using this commend `@command:debug.action.toggleDisassemblyViewSourceCode`, nothing happened, but terminal show more information:
```
Debug-24: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x0000132e","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":24}
Debug-24: Dequeuing...
Debug-24: Elapsed time for Disassembly Request: 002 ms
Debug-25: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x0000132e","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":25}
Debug-25: Dequeuing...
Debug-25: Elapsed time for Disassembly Request: 002 ms
Debug-26: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x0000132e","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":26}
Debug-26: Dequeuing...
Debug-26: Elapsed time for Disassembly Request: 002 ms
Debug-27: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x0000132e","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":27}
Debug-27: Dequeuing...
Debug-27: Elapsed time for Disassembly Request: 002 ms

## Old Version
```
I tried old `version Cortex-Debug v1.6.0`. Source code is display like this:
[image]
It worked for C code well, but for CPP code :
[image]

That is to say:
Version| C Source | C++ Source
---|---|--- 
1.6.0| good | bad
1.12.0| bad| bad
* I just change cortex-debug version and using same project and same configuration.

launch.json:
```json
        {
            "name": "UsbToAmmc6",
            "type": "cortex-debug",
            "cwd": "${workspaceFolder}",
            "executable": "./build/UsbToAmmc6/UsbToAmmc6.elf",
            "device": "ch32v203",
            "request": "launch",
            //"runToEntryPoint": "_start", 
            "runToEntryPoint": "main", 
            "servertype": "openocd", 
            "armToolchainPath": "C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin",
            "toolchainPrefix": "riscv-none-elf",
            "svdFile": "C:/MounRiver/MounRiver_Studio/template/wizard/WCH/RISC-V/CH32V203/NoneOS/CH32V203xx.svd",
            "configFiles": [
                "C:/MounRiver/MounRiver_Studio/toolchain/OpenOCD/bin/wch-riscv.cfg",
            ],
            "showDevDebugOutput": "raw"
        }
```

haneefdm, 2023-08-08:
Works fine for me with v1.12.0. This is not particularly a feature of this extension. The GUI is controlled by VSCode and they have a setting (that is passed to us) that controls viewing source code. Verify it is enabled for your Workspace and/or User. Also check for the Folder to make sure it is not being overridden. Use the Settings Gui to make sure

[image]

FYI: NOTHING changed between 1.6.0 and 1.12.0 for disassembly. We have not touched our code.

You can also set your launch.json as follows to see actual requests from VSCode.

```
"showDevDebugOutput": "vscode"
```

lee8871, 2023-08-08:
I have checked this setting is enabled. 
Strangely source code is showed only here:
[image]
And other source code is no.
[image]

When using different versions of extension, there are indeed differences in phenomenon. I think it may be necessary to add GDB configuration or modify GCC compilation options. The source code should have been written to elf and successfully parsed by GDB. Because breakpoints is working well. 
[image]
Or could this be an adaptation issue between GDB and Cortex debug? I am using risc-v gcc:`https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack` 

The following is the output of the debug console:
```
    This information is used to adjust bounds only when normal disassembly fails.
================================================================================
  Using following memory regions for disassembly
================================================================================
      Size   VMA Beg   VMA End   LMA Beg   LMA End  Attributes
================================================================================
  e4008000  00000000  e4008000  --------  --------  flash blocksize 0x400 nocache 
  1bff8000  e4008000  00000000  --------  --------  rw nocache 
  ------------------------------------------------------------------------------
  00000038  00000000  00000038  --------  --------  (.init) contents alloc load readonly code
  00000108  00000038  00000140  --------  --------  (.vector) contents alloc load readonly code
  000033c8  00000140  00003508  --------  --------  (.text) contents alloc load readonly code
  00000000  00003508  00003508  --------  --------  (.fini) contents alloc load code
  000000fc  20000000  200000fc  00003508  00003604  (.data) contents alloc load data
  00000074  200000fc  20000170  00003604  00003678  (.bss) alloc
  00000000  20000170  20000170  00003678  00003678  (.noinit) alloc
  00000800  20002000  20002800  --------  --------  (.stack) alloc
================================================================================
Debug-36: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00000140 -e 0x000002f4 -- 5      436 bytes  (memset)
Suppressing output for '54-data-disassemble -s 0x00000140 -e 0x000002f4 -- 5'
Debug: Gdb command: -data-disassemble -s 0x000002f4 -e 0x00002962 -- 5     9838 bytes  (rs485::init())
Suppressing output for '55-data-disassemble -s 0x000002f4 -e 0x00002962 -- 5'
Debug: data-disassemble -s 0x00000140 -e 0x000002f4 -- 5 => Found 154 instructions. 0 with source code, 154 without
From client: disassemble({"memoryReference":"0x00001346","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true})
Debug-37: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00001346","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":37}
Debug-37: ******** Waiting for previous request to complete
Debug: data-disassemble -s 0x000002f4 -e 0x00002962 -- 5 => Found 3562 instructions. 0 with source code, 3562 without
Debug-36: Elapsed time for Disassembly Request: 4979 ms
To client: {"seq":0,"type":"response","request_seq":36,"command":"disassemble","success":true,"body":{"instructions":[{"address":"0x00001120","pvtAddress":4384,"instruction":"2d83                                                  srli\ta4,a4,0xb","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"2d 83","instructionBytes":"<RCC_GetClocksFreq+138>"},{"address":"0x00001122","pvtAddress":4386,"instruction":"1d8b                                                  andi\ta4,a4,7","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"1d 8b","instructionBytes":"<RCC_GetClocksFreq+140>"},{"address":"0x00001124","pvtAddress":4388,"instruction":"ba96                                                  add\ta3,a3,a4","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"ba 96","instructionBytes":"<RCC_GetClocksFreq+142>"},{"address":"0x00001126","pvtAddress":4390,"instruction":"03c70600                                              lbu\ta4,0(a3)","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"03 c7 06 00","instructionBytes":"<RCC_GetClocksFreq+144>"},{"address":"0x0000112a","pvtAddress":4394,"instruction":"97f6ff1f                                              auipc\ta3,0x1ffff","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"97 f6 ff 1f","instructionBytes":"<RCC_GetClocksFreq+148>"},{"address":"0x0000112e","pvtAddress":4398,"instruction":"9386a6f7                                              addi\ta3,a3,-134 # 0x200000a4 <ADCPrescTable>","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":[...]
Debug-37: Dequeuing...
Debug-37: Elapsed time for Disassembly Request: 004 ms
To client: {"seq":0,"type":"response","request_seq":37,"command":"disassemble","success":true,"body":{"instructions":[{"address":"0x00001120","pvtAddress":4384,"instruction":"2d83                                                  srli\ta4,a4,0xb","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"2d 83","instructionBytes":"<RCC_GetClocksFreq+138>"},{"address":"0x00001122","pvtAddress":4386,"instruction":"1d8b                                                  andi\ta4,a4,7","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"1d 8b","instructionBytes":"<RCC_GetClocksFreq+140>"},{"address":"0x00001124","pvtAddress":4388,"instruction":"ba96                                                  add\ta3,a3,a4","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"ba 96","instructionBytes":"<RCC_GetClocksFreq+142>" [... cut ...] ructionBytes":"<RCC_GetClocksFreq+144>"},{"address":"0x0000112a","pvtAddress":4394,"instruction":"97f6ff1f                                              auipc\ta3,0x1ffff","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"97 f6 ff 1f","instructionBytes":"<RCC_GetClocksFreq+148>"},{"address":"0x0000112e","pvtAddress":4398,"instruction":"9386a6f7                                              addi\ta3,a3,-134 # 0x200000a4 <ADCPrescTable>","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":[...]
```

lee8871, 2023-08-08:
ALL DEBUG CONSOLE:
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "UsbToAmmc6",
    "type": "cortex-debug",
    "cwd": "D:\\SRB2\\cmake_ws",
    "executable": "D:/SRB2/cmake_ws/build/UsbToAmmc6/UsbToAmmc6.elf",
    "device": "ch32v203",
    "request": "launch",
    "runToEntryPoint": "main",
    "servertype": "openocd",
    "armToolchainPath": "C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin",
    "toolchainPrefix": "riscv-none-elf",
    "svdFile": "C:/MounRiver/MounRiver_Studio/template/wizard/WCH/RISC-V/CH32V203/NoneOS/CH32V203xx.svd",
    "configFiles": [
        "C:/MounRiver/MounRiver_Studio/toolchain/OpenOCD/bin/wch-riscv.cfg"
    ],
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "searchDir": [],
    "toolchainPath": "C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin",
    "extensionPath": "c:/Users/lee88/.vscode/extensions/marus25.cortex-debug-1.12.0",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.0",
    "__sessionId": "f897430a-2f22-4c66-a7a8-cf5a85026455"
}
Reading symbols from C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin/riscv-none-elf-objdump.exe --syms -C -h -w D:/SRB2/cmake_ws/build/UsbToAmmc6/UsbToAmmc6.elf
Reading symbols from c:/tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin/riscv-none-elf-nm.exe --defined-only -S -l -C -p D:/SRB2/cmake_ws/build/UsbToAmmc6/UsbToAmmc6.elf
Launching GDB: "C:\\Tool\\xpack-riscv-none-elf-gcc-12.2.0-3\\bin\\riscv-none-elf-gdb.exe" -q --interpreter=mi2
1-gdb-version
From client: initialize({"clientID":"vscode","clientName":"Visual Studio Code","adapterID":"cortex-debug","pathFormat":"path","linesStartAt1":true,"columnsStartAt1":true,"supportsVariableType":true,"supportsVariablePaging":true,"supportsRunInTerminalRequest":true,"locale":"en","supportsProgressReporting":true,"supportsInvalidatedEvent":true,"supportsMemoryReferences":true,"supportsArgsCanBeInterpretedByShell":true,"supportsMemoryEvent":true,"supportsStartDebuggingRequest":true})
To client: {"seq":0,"type":"response","request_seq":1,"command":"initialize","success":true,"body":{"supportsHitConditionalBreakpoints":true,"supportsConfigurationDoneRequest":true,"supportsConditionalBreakpoints":true,"supportsLogPoints":true,"supportsFunctionBreakpoints":true,"supportsEvaluateForHovers":true,"supportsSetVariable":true,"supportsRestartRequest":true,"supportsGotoTargetsRequest":true,"supportSuspendDebuggee":true,"supportTerminateDebuggee":true,"supportsDataBreakpoints":true,"supportsDisassembleRequest":true,"supportsSteppingGranularity":true,"supportsInstructionBreakpoints":true,"supportsReadMemoryRequest":true,"supportsWriteMemoryRequest":true}}
From client: launch({"name":"UsbToAmmc6","type":"cortex-debug","cwd":"D:\\SRB2\\cmake_ws","executable":"D:/SRB2/cmake_ws/build/UsbToAmmc6/UsbToAmmc6.elf","device":"ch32v203","request":"launch","runToEntryPoint":"main","servertype":"openocd","armToolchainPath":"C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin","toolchainPrefix":"riscv-none-elf","svdFile":"C:/MounRiver/MounRiver_Studio/template/wizard/WCH/RISC-V/CH32V203/NoneOS/CH32V203xx.svd","configFiles":["C:/MounRiver/MounRiver_Studio/toolchain/OpenOCD/bin/wch-riscv.cfg"],"showDevDebugOutput":"vscode","__configurationTarget":6,"gdbServerConsolePort":55878,"pvtAvoidPorts":[],"chainedConfigurations":{"enabled":false},"debuggerArgs":[],"swoConfig":{"enabled":false,"decoders":[],"cpuFrequency":0,"swoFrequency":0,"source":"probe"},"rttConfig":{"enabled":false,"decoders":[]},"graphConfig":[],"preLaunchCommands":[],"postLaunchCommands":[],"preAttachCommands":[],"postAttachCommands":[],"preRestartCommands":[],"postRestartCommands":[],"preResetCommands":[],"postResetCommands":[],"searchDir":[],"toolchainPath":"C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin","extensionPath":"c:/Users/lee88/.vscode/extensions/marus25.cortex-debug-1.12.0","registerUseNaturalFormat":true,"variableUseNaturalFormat":true,"pvtVersion":"1.12.0","__sessionId":"f897430a-2f22-4c66-a7a8-cf5a85026455"})
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stdout","output":"Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage"}}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"\"configuration\": {\n    \"name\": \"UsbToAmmc6\",\n    \"type\": \"cortex-debug\",\n    \"cwd\": \"D:\\\\SRB2\\\\cmake_ws\",\n    \"executable\": \"D:/SRB2/cmake_ws/build/UsbToAmmc6/UsbToAmmc6.elf\",\n    \"device\": \"ch32v203\",\n    \"request\": \"launch\",\n    \"runToEntryPoint\": \"main\",\n    \"servertype\": \"openocd\",\n    \"armToolchainPath\": \"C:/Tool/xpack-riscv-none-elf-gcc-12.2.0-3/bin\",\n    \"toolchainPrefix\": \"riscv-none- [... cut ...] ructionBytes":"<RCC_GetClocksFreq+144>"},{"address":"0x0000112a","pvtAddress":4394,"instruction":"97f6ff1f                                              auipc\ta3,0x1ffff","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":"97 f6 ff 1f","instructionBytes":"<RCC_GetClocksFreq+148>"},{"address":"0x0000112e","pvtAddress":4398,"instruction":"9386a6f7                                              addi\ta3,a3,-134 # 0x200000a4 <ADCPrescTable>","symbol":"RCC_GetClocksFreq","pvtInstructionBytes":[...]
```

haneefdm, 2023-08-08:
```
Debug: Gdb command: -data-disassemble -s 0x00000140 -e 0x000002f4 -- 5      436 bytes  (memset)
```
That gdb command says please give me disassembly for those addresses with inlined source code if available. See https://sourceware.org/gdb/onlinedocs/gdb/GDB_002fMI-Data-Manipulation.html#GDB_002fMI-Data-Manipulation

But source code is not guaranteed by GDB and depends on your version of GDB and how you compiled your code. You said if everything is the same, and just changing the version of Cortex-Debug, things fail. But something must be different. We don't filter out what is sent by GDB so, I have to assume that GDB is not finding the source code.

You can run the disassemble command from the Debug Console. Just use the command we used and you will see the raw output from GDB.

```
-data-disassemble -s 0x00000140 -e 0x000002f4 -- 5
```

lee8871, 2023-08-08:
With your help, I have chosen this piece of code for testing.
[image]
I run this code `-data-disassemble -s 0x00000d94 -e 0x00000db0 -- 5`, and get following response:
[image]
It is a json string.
[image]
From this perspective, GDB knows the line number and file name of the disassembly.

lee8871, 2023-08-08:
[image]

lee8871, 2023-08-08:
[image]
Based on this information, the software should be able to locate the corresponding line and obtain the source code within it.

This is GDB response：
```json
{"output":"","token":100,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["asm_insns",[["src_and_asm_line",[["line","89"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","90"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","91"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","92"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","93"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","94"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[[["address","0x00000d94"],["func-name","USART_Printf_Init"],["offset","0"],["opcodes","79 71"],["inst","addi\tsp,sp,-48"]],[["address","0x00000d96"],["func-name","USART_Printf_Init"],["offset","2"],["opcodes","22 d4"],["inst","sw\ts0,40(sp)"]],[["address","0x00000d98"],["func-name","USART_Printf_Init"],["offset","4"],["opcodes","2a 84"],["inst","mv\ts0,a0"]],[["address","0x00000d9a"],["func-name","USART_Printf_Init"],["offset","6"],["opcodes","11 65"],["inst","lui\ta0,0x4"]],[["address","0x00000d9c"],["func-name","USART_Printf_Init"],["offset","8"],["opcodes","85 45"],["inst","li\ta1,1"]],[["address","0x00000d9e"],["func-name","USART_Printf_Init"],["offset","10"],["opcodes","11 05"],["inst","addi\ta0,a0,4 # 0x4004"]]]]]],["src_and_asm_line",[["line","89"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[[["address","0x00000da0"],["func-name","USART_Printf_Init"],["offset","12"],["opcodes","06 d6"],["inst","sw\tra,44(sp)"]]]]]],["src_and_asm_line",[["line","90"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","91"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","92"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","93"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","94"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[[["address","0x00000da2"],["func-name","USART_Printf_Init"],["offset","14"],["opcodes","55 2e"],["inst","jal\t0x1156 <RCC_APB2PeriphClockCmd>"]]]]]],["src_and_asm_line",[["line","95"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","96"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[[["address","0x00000da4"],["func-name","USART_Printf_Init"],["offset","16"],["opcodes","93 07 00 20"],["inst","li\ta5,512"]],[["address","0x00000da8"],["func-name","USART_Printf_Init"],["offset","20"],["opcodes","23 12 f1 00"],["inst","sh\ta5,4(sp)"]]]]]],["src_and_asm_line",[["line","97"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[[["address","0x00000dac"],["func-name","USART_Printf_Init"],["offset","24"],["opcodes","37 15 01 40"],["inst","lui\ta0,0x40011"]]]]]]]]]}}
```

haneefdm, 2023-08-08:
Does this file `D:/SRB2/cmake_ws/old/Debug/debug.c` actually exist? I mean the exact path on disk. Note that actual source lines are not part of the GDB response. Just file and line numbers. The file needs to exist for VSCode to find the contents. GDB only reports. Instead of machine instructions lime `-data-disassemble` where all responses are made for machine readable format, you can use normal gdb commands for human readable purposes. Enter the following in the Debug Console

```
disassemble /s <start>,><end>
```
Or one of the other methods like `disassemble /s <start>,>+<length>`. See full documentation

https://sourceware.org/gdb/onlinedocs/gdb/Machine-Code.html

I notice you did not enable the proper debug mode either. It should be
```
    "showDevDebugOutput": "vscode",
```
This shows us what VSCode is asking us and what we are sending back as a response. I don't see any of that here. Hopefully, we can see what is going on there...

lee8871, 2023-08-08:
[image]

lee8871, 2023-08-08:
This file do exist.
[image]

I have confirmed to write  ` "showDevDebugOutput": "vscode",`，but in the output, this setting is still 'raw'
[image]
I tried change "runToEntryPoint" to  "initVTF", and I can see that this command is indeed taking effect.
[image]
This indicates that the file used for debugging is the one I just modified. But 'showDevDebugOutput' is still 'raw',

lee8871, 2023-08-08:
About disassemble conmmend. I did see successful execution of these instructions, as well as disassembly and source code. But when executed again. It is impossible to reproduce the previous situation.
[image]

haneefdm, 2023-08-08:
Wait, when you run the command to disassemble, you don't see any output from GDB to stdout? What we print as debug JSON stuff doesn't count. I see zero output from GDB when you ran the disassemble command. It executed but since this not a machine interface, we don't get to see the stdout of gdb. You should have seen like the examples shown here

https://sourceware.org/gdb/onlinedocs/gdb/Machine-Code.html

along with our junk.

Yes, showDevDebugOutput will start out as "vscode", but once VSCode is notified, we revert back to normal. You should see a LOT more output in your Debug Console and I don't see any. You should see stuff like

```
From client: ...
To client: ...
```

VSCode is the client and we are the adapter/server in this case. These while verbose, are important.

lee8871, 2023-08-08:
When I just executed these disassembly commands, I did see the correct output, as well as the source code and disassembly.
Like this：
[image] I didn't take any screenshots.

Later, I tried the issue with 'showDevDebugOutput', and of course, I restarted GDB. But I didn't see the correct output again, which is really strange.

I tried many configurations but couldn't see the output of the disassembly command again. I am continuing to search for the reason. 
I have no clue about this issue that cannot be reproduced, except for GDB ran for a relatively long time, about 30 minutes.

haneefdm, 2023-08-08:
To me actual text from the Debug Console is much more helpful than partial screenshots. Screenshots are fine for stuff outside the Debug Console.

In your last response, I have trouble telling what you actually did.

lee8871, 2023-08-08:
I have successfully used  `disassemble main`  command once.  GDB returns readable source code and disassemble 
When I run GDB again, this command returns nothing.
I may not have made any modifications to the configuration. But I'm not quite sure yet.
For now:
```
disassemble main
{"output":"","token":26,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
disassemble /s 0x1330,0x1340
{"output":"","token":30,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
-data-disassemble -s 0x00000d94 -e 0x00000da0 -- 5
{"output":"","token":34,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["asm_insns",[["src_and_asm_line",[["line","89"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","90"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","91"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","92"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","93"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[]]]],["src_and_asm_line",[["line","94"],["file","D:/SRB2/cmake_ws/old/Debug/debug.c"],["fullname","D:\\SRB2\\cmake_ws\\old\\Debug\\debug.c"],["line_asm_insn",[[["address","0x00000d94"],["func-name","USART_Printf_Init"],["offset","0"],["opcodes","79 71"],["inst","addi\tsp,sp,-48"]],[["address","0x00000d96"],["func-name","USART_Printf_Init"],["offset","2"],["opcodes","22 d4"],["inst","sw\ts0,40(sp)"]],[["address","0x00000d98"],["func-name","USART_Printf_Init"],["offset","4"],["opcodes","2a 84"],["inst","mv\ts0,a0"]],[["address","0x00000d9a"],["func-name","USART_Printf_Init"],["offset","6"],["opcodes","11 65"],["inst","lui\ta0,0x4"]],[["address","0x00000d9c"],["func-name","USART_Printf_Init"],["offset","8"],["opcodes","85 45"],["inst","li\ta1,1"]],[["address","0x00000d9e"],["func-name","USART_Printf_Init"],["offset","10"],["opcodes","11 05"],["inst","addi\ta0,a0,4 # 0x4004"]]]]]]]]]}}
xxxxxxx 
Undefined command: "xxxxxxx".  Try "help". (from interpreter-exec console "xxxxxxx ")
```

haneefdm, 2023-08-08:
You have a lot of information with various levels of debug output. I am only seeing bits and pieces. I have to sign off now and maybe you can make some progress on your own until tomorrow. I can't really help with such limited visibility.

lee8871, 2023-08-08:
Thank you very much for your help today.I will try to reproduce the phenomenon.

lee8871, 2023-08-11:
After this period of testing, I have not found any further issues. There was even a test where I saw the corresponding source code for disassembly, but I couldn't succeed next time.

Deeper testing requires manual execution of GDB, attempting non cross compilation. I don't have time to handle it at the moment.

So I closed the issue and made the following summary:
1. The source code cannot be displayed in disassembly, and the disassembly command does not return.
2. It is uncertain whether the problem is GDB or Cortex debugger.
3. The problem is unstable and occasionally work well.
4. Different versions of Cortex-debugger have different phenomena.

closed 2023-08-11 by lee8871